qcr/benchbot #2: TypeError: 'module' object is not callable

liucsg, 2020-04-15:
Hi, btalb
After installing benchbot, i run the command of benchbot_run as this, "benchbot_run --env miniroom:1 --task semantic_slam:active:ground_truth -f". There is a error as following,

################################################################################
################# CHECKING FOR BENCHBOT SOFTWARE STACK UPDATES #################
################################################################################

Skipping ...

################################################################################
##################### STARTING THE BENCHBOT SOFTWARE STACK #####################
################################################################################

Running the BenchBot system with the following settings:

    Selected environment: miniroom:1
    Selected task:        semantic_slam:active:ground_truth
    Actions set:          active
    Observations set:     ground_truth
    Maps:                 "/Game/AI_vol3_03_base/Maps/AI_vol3_scene_03"
                          (starting @ [0.7,0,0,-0.7,1.2,1.5,0.3])
a825ecb8729beabb70cbe2ee715030ff7c769e1574f00a35c8662cac32695f9a
Created benchbot_network.

non-network local connections being added to access control list
089e4f7b7a74ea670a21cd23eaa0b6bcdb3b12ae5caa6022ac9c986300a2938f
3e31dc23b4deec436f60c0963b3fde2ec9d3b281c036b0ebc020c25518d03087
9623b25144d5ce4c618049e6ea9e5a091101385a33f7fc8c3ddc782758d267ac
58f7fd8f431424c902762112df5c5877deeae7a2d05be2c98926244296c7dd1c
non-network local connections being removed from access control list

################################################################################
################### BENCHBOT IS RUNNING (Ctrl^C to exit) ... ###################
################################################################################

Initialising supervisor
Configuring the supervisor...
Starting a supervisor with the following configuration:

{'actions': ['move_distance', 'move_angle'],
 'environment_names': ['miniroom_1'],
 'observations': ['image_depth',
                  'image_depth_info',
                  'image_rgb',
                  'image_rgb_info',
                  'laser',
                  'poses'],
 'robot': {'image_depth': {'callback_api': 'api_callbacks.decode_jsonpickle',
                           'callback_supervisor': 'supervisor_callbacks.encode_depth_image',
                           'connection': 'ros_to_api',
                           'ros_topic': '/camera/depth/image_raw',
                           'ros_type': 'sensor_msgs/Image'},
           'image_depth_info': {'callback_api': 'api_callbacks.decode_jsonpickle',
                                'callback_supervisor': 'supervisor_callbacks.encode_camera_info',
                                'connection': 'ros_to_api',
                                'ros_topic': '/camera/depth/camera_info',
                                'ros_type': 'sensor_msgs/CameraInfo'},
           'image_rgb': {'callback_api': 'api_callbacks.decode_color_image',
                         'callback_supervisor': 'supervisor_callbacks.encode_color_image',
                         'connection': 'ros_to_api',
                         'ros_topic': '/camera/color/image_raw',
                         'ros_type': 'sensor_msgs/Image'},
           'image_rgb_info': {'callback_api': 'api_callbacks.decode_jsonpickle',
                              'callback_supervisor': 'supervisor_callbacks.encode_camera_info',
                              'connection': 'ros_to_api',
                              'ros_topic': '/camera/color/camera_info',
                              'ros_type': 'sensor_msgs/CameraInfo'},
           'laser': {'callback_api': 'api_callbacks.decode_jsonpickle',
                     'callback_supervisor': 'supervisor_callbacks.encode_laserscan',
                     'connection': 'ros_to_api',
                     'ros_topic': '/scan_laser',
                     'ros_type': 'sensor_msgs/LaserScan'},
           'move_angle': {'callback_supervisor': 'supervisor_callbacks.move_angle',
                          'connection': 'api_to_ros',
                          'ros_topic': '/cmd_vel',
                          'ros_type': 'geometry_msgs/Twist'},
           'move_distance': {'callback_supervisor': 'supervisor_callbacks.move_distance',
                             'connection': 'api_to_ros',
                             'ros_topic': '/cmd_vel',
                             'ros_type': 'geometry_msgs/Twist'},
           'move_next': {'callback_supervisor': 'supervisor_callbacks.move_next',
                         'connection': 'api_to_ros',
                         'ros_topic': '/cmd_vel',
                         'ros_type': 'geometry_msgs/Twist'},
           'poses': {'callback_api': 'api_callbacks.decode_jsonpickle',
                     'callback_supervisor': 'supervisor_callbacks.create_pose_list',
                     'connection': 'ros_to_api'}},
 'task_name': 'semantic_slam:active:ground_truth'}
Traceback (most recent call last):
  File "/benchbot/ros_ws/src/benchbot_supervisor/nodes/benchbot_supervisor", line 10, in <module>
    s.run()
  File "<string>", line 374, in run
TypeError: 'module' object is not callable

################################################################################
###################### CLEANING UP ALL BENCHBOT REMNANTS #######################
################################################################################

Stopped the following containers:
58f7fd8f4314
089e4f7b7a74

Deleted the following containers:
Deleted Containers:
58f7fd8f431424c902762112df5c5877deeae7a2d05be2c98926244296c7dd1c
9623b25144d5ce4c618049e6ea9e5a091101385a33f7fc8c3ddc782758d267ac
3e31dc23b4deec436f60c0963b3fde2ec9d3b281c036b0ebc020c25518d03087
089e4f7b7a74ea670a21cd23eaa0b6bcdb3b12ae5caa6022ac9c986300a2938f
74d7f9453579d005a0cb3f47ee255ada11bd83512ab4131e06d697c0f0b987ac
030e5d4de5e9d61e1f513241b98e37771eeced8e75503162cae0c00239fffc1f
f7f1370cbd81b6b40faa13b10995bdfe19254ad17350772826f9a383a65cef02
97a1410d0d12c17d5a06ce3f89691c38ee9f943ae6f16d2753a0d20b50a92e40
f7da8869792111d43c3891dfe5d0a8bf44b5c0079e2c002c80def09ad897c104
9a9ac8bfbe590eafe00c395ad90375666d99401d3894effe63ab9623a5914370

Deleted Networks:
benchbot_network

Total reclaimed space: 228.5MB

Finished cleaning!

btalb, 2020-04-16:
The error suggests the `signal` sub-module from `gevent` is not correctly importing (it instead seems to be using the entirely different `signal` package). There is no way this should happen given the automated installation process, and that it's inside the simulator Docker image.

Can I get some information to try understand how things could have gotten into this state:

1. Was this installed using the `./install` script, or have any manual actions been taken to get things up & running?
2. Were there any issues during the install / does `benchbot_install` run through now without any issues (it should simply validate the installation)?

As to addressing the issue, can you run the following command to attach the `benchbot.log` file so I can see what is different to a normal installation:
```
docker run -it benchbot/simulator:base /bin/bash -x -c 'python --version; source ros_ws/devel/setup.bash; python --version; python2 -m pip freeze; python3 -m pip freeze; python2 -c "from gevent import signal; print(signal.__file__)"; python3 -c "from gevent import signal; print(signal.__file__)"' &> benchbot.log
```
(here's an example from a working install: [benchbot.log](https://github.com/RoboticVisionOrg/benchbot/files/4484474/benchbot.log))

liucsg, 2020-04-16:
Hi, btalb
Before using the ./install script, I have taken some manual actions to installing the requirements of simulator. Then, I run benchbot_install successfully. This is benchbot.log as following,

+ python --version
Python 2.7.17
+ source ros_ws/devel/setup.bash
++ CATKIN_SHELL=bash
++++ dirname ros_ws/devel/setup.bash
+++ builtin cd ros_ws/devel
+++ pwd
++ _CATKIN_SETUP_DIR=/benchbot/ros_ws/devel
++ . /benchbot/ros_ws/devel/setup.sh
+++ : /benchbot/ros_ws/devel
+++ _SETUP_UTIL=/benchbot/ros_ws/devel/_setup_util.py
+++ unset _CATKIN_SETUP_DIR
+++ '[' '!' -f /benchbot/ros_ws/devel/_setup_util.py ']'
++++ uname -s
+++ _UNAME=Linux
+++ _IS_DARWIN=0
+++ '[' Linux = Darwin ']'
+++ unset _UNAME
+++ export CMAKE_PREFIX_PATH
+++ '[' 0 -eq 0 ']'
+++ export LD_LIBRARY_PATH
+++ unset _IS_DARWIN
+++ export PATH
+++ export PKG_CONFIG_PATH
+++ export PYTHONPATH
+++ '[' -z bash ']'
+++ '[' -d '' ']'
+++ _TMPDIR=/tmp
++++ mktemp /tmp/setup.sh.XXXXXXXXXX
+++ _SETUP_TMP=/tmp/setup.sh.6BCLYUmaqX
+++ unset _TMPDIR
+++ '[' 0 -ne 0 -o '!' -f /tmp/setup.sh.6BCLYUmaqX ']'
+++ CATKIN_SHELL=bash
+++ /benchbot/ros_ws/devel/_setup_util.py
+++ _RC=0
+++ '[' 0 -ne 0 ']'
+++ unset _RC
+++ unset _SETUP_UTIL
+++ . /tmp/setup.sh.6BCLYUmaqX
++++ export CMAKE_PREFIX_PATH=/benchbot/ros_ws/devel:/opt/ros/melodic
++++ CMAKE_PREFIX_PATH=/benchbot/ros_ws/devel:/opt/ros/melodic
++++ export LD_LIBRARY_PATH=/benchbot/ros_ws/devel/lib:/opt/ros/melodic/lib
++++ LD_LIBRARY_PATH=/benchbot/ros_ws/devel/lib:/opt/ros/melodic/lib
++++ export PATH=/opt/ros/melodic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
++++ PATH=/opt/ros/melodic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
++++ export PKG_CONFIG_PATH=/benchbot/ros_ws/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig
++++ PKG_CONFIG_PATH=/benchbot/ros_ws/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig
++++ export PYTHONPATH=/benchbot/ros_ws/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages
++++ PYTHONPATH=/benchbot/ros_ws/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages
++++ export _CATKIN_ENVIRONMENT_HOOKS_COUNT=12
++++ _CATKIN_ENVIRONMENT_HOOKS_COUNT=12
++++ export _CATKIN_ENVIRONMENT_HOOKS_0=/opt/ros/melodic/etc/catkin/profile.d/1.ros_distro.sh
++++ _CATKIN_ENVIRONMENT_HOOKS_0=/opt/ros/melodic/etc/catkin/profile.d/1.ros_distro.sh
++++ export _CATKIN_ENVIRONMENT_HOOKS_0_WORKSPACE=/opt/ros/melodic
++++ _CATKIN_ENVIRONMENT_HOOKS_0_WORKSPACE=/opt/ros/melodic
++++ export _CATKIN_ENVIRONMENT_HOOKS_1=/opt/ros/melodic/etc/catkin/profile.d/1.ros_etc_dir.sh
++++ _CATKIN_ENVIRONMENT_HOOKS_1=/opt/ros/melodic/etc/catkin/profile.d/1.ros_etc_dir.sh
++++ export _CATKIN_ENVIRONMENT_HOOKS_1_WORKSPACE=/opt/ros/melodic
++++ _CATKIN_ENVIRONMENT_HOOKS_1_WORKSPACE=/opt/ros/melodic
++++ export _CATKIN_ENVIRONMENT_HOOKS_2=/opt/ros/melodic/etc/catkin/profile.d/1.ros_package_path.sh
++++ _CATKIN_ENVIRONMENT_HOOKS_2=/opt/ros/melodic/etc/catkin/profile.d/1.ros_package_path.sh
++++ export _CATKIN_ENVIRONMENT_HOOKS_2_WORKSPACE=/opt/ros/melodic
++++ _CATKIN_ENVIRONMENT_HOOKS_2_WORKSPACE=/opt/ros/melodic
++++ export _CATKIN_ENVIRONMENT_HOOKS_3=/opt/ros/melodic/etc/catkin/profile.d/1.ros_python_version.sh
++++ _CATKIN_ENVIRONMENT_HOOKS_3=/opt/ros/melodic/etc/catkin/profile.d/1.ros_python_version.sh
++++ export _CATKIN_ENVIRONMENT_HOOKS_3_WORKSPACE=/opt/ros/melodic
++++ _CATKIN_ENVIRONMENT_HOOKS_3_WORKSPACE=/opt/ros/melodic
[... 699 lines cut ...]
tensorflow-estimator==1.13.0
tensorflow-gpu==1.13.1
termcolor==1.1.0
testresources==2.0.1
tf==1.12.1
tf-conversions==1.12.1
tf2-geometry-msgs==0.6.5
tf2-kdl==0.6.5
tf2-py==0.6.5
tf2-ros==0.6.5
topic-tools==1.14.5
tqdm==4.45.0
typing-extensions==3.7.4.2
unattended-upgrades==0.1
urdfdom-py==0.4.2
urllib3==1.22
wadllib==1.3.3
Werkzeug==1.0.1
xacro==1.13.5
xkit==0.0.0
zipp==3.1.0
+ python2 -c 'from gevent import signal; print(signal.__file__)'
/home/benchbot/.local/lib/python2.7/site-packages/gevent/signal.pyc
+ python3 -c 'from gevent import signal; print(signal.__file__)'
/home/benchbot/.local/lib/python3.6/site-packages/gevent/signal.py

liucsg, 2020-04-16:
Hi, btalb
Before using the ./install script, I have taken some manual actions to installing the requirements of simulator. Then, I run benchbot_install successfully. This is benchbot.log as following,

[benchbot.log](https://github.com/RoboticVisionOrg/benchbot/files/4484699/benchbot.log)

liucsg, 2020-04-16:
Hi, btalb
The cause of error is gevent version. The Installed  version of gevent version is 1.5.0. 
Thank you!

btalb, 2020-04-16:
You're spot on @liucsg . The `gevent` python package has removed API methods in version 1.5.0 (see [here](http://www.gevent.org/whatsnew_1_5.html) for more details).

The changes affected both [benchbot_simulator](https://github.com/RoboticVisionOrg/benchbot_simulator)  & [benchbot_supervisor](https://github.com/RoboticVisionOrg/benchbot_simulator). I've hotfixed both of those in commits [8f9cff](https://github.com/RoboticVisionOrg/benchbot_simulator/commit/8f9cff4c00c95ddc8cbbe1d79cb90fd6b8114d14) & [a900c9](https://github.com/RoboticVisionOrg/benchbot_supervisor/commit/a900c97a0e3853eab12da0912e68d1993770940f) respectively.

To get the updates simply run:
```
benchbot_install
```
The install will only have to change those two packages as they are installed last; so it should be a much quicker install. Let me know if these fixes resolve the issue.

liucsg, 2020-04-17:
Hi btalb, I canot run simulator. It is always waiting to establish connection to a running simulator ...
I excute  two commands ,one is benchbot_run, the other is benchbot_submit,as following, 
1)  benchbot_run --env miniroom:1 --task semantic_slam:active:ground_truth -f
2) benchbot_submit --containerised ./examples/hello_active/
Meanwhile there are two log, one is benchbot_run.log, the other is benchbot_submit.log.
[benchbot_run.log](https://github.com/RoboticVisionOrg/benchbot/files/4491705/benchbot_run.log)
[benchbot_submit.log](https://github.com/RoboticVisionOrg/benchbot/files/4491707/benchbot_submit.log)

liucsg, 2020-04-17:
Hi btalb, I canot run simulator. It is always waiting to establish connection to a running simulator ...
I excute two commands ,one is benchbot_run, the other is benchbot_submit,as following,

   1) benchbot_run --env miniroom:1 --task semantic_slam:active:ground_truth -f
   2) benchbot_submit --containerised ./examples/hello_active/
    Meanwhile there are two log, one is benchbot_run.log, the other is benchbot_submit.log.
[benchbot_run.log](https://github.com/RoboticVisionOrg/benchbot/files/4491856/benchbot_run.log)
[benchbot_submit.log](https://github.com/RoboticVisionOrg/benchbot/files/4491857/benchbot_submit.log)

liucsg, 2020-04-20:
There is a errror  when executing this command of "./IsaacSimProject "
    /IsaacSimProject  <map_name> \
    -isaac_sim_config_json='<path_to_isaac>/apps/carter/carter_sim/bridge_config/carter_full.json' \
    -windowed -ResX=960 -ResY=540 -vulkan -game

  **unable to read vr path registry from openvrpaths.vrpath**

liucsg, 2020-04-20:
There is a errror when executing this command of "./IsaacSimProject "

./IsaacSimProject <map_name>
-isaac_sim_config_json='<path_to_isaac>/apps/carter/carter_sim/bridge_config/carter_full.json'
-windowed -ResX=960 -ResY=540 -vulkan -game

unable to read vr path registry from openvrpaths.vrpath

liucsg, 2020-04-20:
There is a errror when executing this command of "./IsaacSimProject "

./IsaacSimProject <map_name>
-isaac_sim_config_json='<path_to_isaac>/apps/carter/carter_sim/bridge_config/carter_full.json'
-windowed -ResX=960 -ResY=540 -vulkan -game

**unable to read vr path registry from openvrpaths.vrpath**

liucsg, 2020-04-21:
Hi, btalb,
There is collapse after running IsaacSimProject a short time . I don't know where the problem is.
Could you export the images of "benchbot/simulator:base" as a compressed file and share with me ?
Thank you!

btalb, 2020-04-22:
@liucsg I'm closing this issue (the `gevent` error), & have moved your comments to a new issue #3 , which we can work through there.

closed 2020-04-22 by btalb