qcr/benchbot #5: Evaluation results

liucsg, 2020-05-09:
There are some evaluation results after excuting benchbot_run and benchbot_submit, as following,
1) benchbot_run --env miniroom:1 --task semantic_slam:active:ground_truth -f
2) benchbot_submit --evaluate --containerised examples/semantic_slam_attempt

################################################################################
################# Running evaluation on '/tmp/benchbot_result' #################
################################################################################

Evaluating the performance from results in '/tmp/benchbot_result':

/home/kinco2/anaconda3/lib/python3.6/site-packages/benchbot_eval/evaluator.py:224: UserWarning: No 'class_list' field provided; assuming results have used our default class list

**Evaluation results:

{'environment_details': {'name': 'miniroom', 'numbers': ['1']},
 'scores': {'OMQ': 0.0,
            'avg_fp_quality': 1.0,
            'avg_label': 0.0,
            'avg_pairwise': 0.0,
            'avg_spatial': 0.0},
 'task_details': {'control_mode': 'passive',
                  'localisation_mode': 'ground_truth',
                  'type': 'semantic_slam'}}

Done.

**
[benchbot_result.txt](https://github.com/RoboticVisionOrg/benchbot/files/4602599/benchbot_result.txt)

I am not sure if the evalution result is normal.

btalb, 2020-05-11:
The issue here is that the results contain both an `"objects"` (which is empty) & `"detections"` (which is where the results are) field. The field name was changed from `"detections"` to `"objects"` many versions ago. The evaluation is returning 0 because it is correctly looking at the values in the `"objects"` key.

Can you please ensure all of your BenchBot components are up to date? The easiest way to do this is simply running `benchbot_install`.

If you can't update this way, then you will have to manually. This should involve (no guarantees as we don't have any official support available for manual installation setups):

1. Updating all local components: run `git pull` from inside each of the `./api`, `./eval`, `./examples`, & `./batches` directories
2. Update all containerised components (`benchbot_supervisor` & `benchbot_simulator` in image `benchbot/simulator:base, and `benchbot_api` in `benchbot/submission:base`): start a container `docker run --name updating -it IMAGE_NAME /bin/bash`, run `git pull` inside the container, perform any necessary re-compilation steps, close the container, update the image using `docker commit updating IMAGE_NAME`.

btalb, 2020-05-12:
@liucsg thank you for reporting this issue to us. I was able to reproduce it on my end, and fixes for it are part of the `v0.7.1` updates I have just released. This specific issue is addressed by commits [e37f92b](https://github.com/RoboticVisionOrg/benchbot_examples/commit/e37f92b6612b6032f97b0553a633350c5356fb19) & [2f6bb93](https://github.com/RoboticVisionOrg/benchbot_examples/commit/2f6bb9389af0e34d21f527c80b4316c692da7ea1) in the [benchbot_examples](https://github.com/RoboticVisionOrg/benchbot_examples) repository.

Please follow the advice above to update, and let me know if you still have any issues

closed 2020-05-12 by btalb