qcr/benchbot #6: 02_3rd_person_view.patch and 03_main_cam_height.patch in .isaac_patches of benchbot_simulator

liucsg, 2020-05-14:
Hi, could you tell me the reason about modification of parent_T_camera ？

"CarterLeftCamera": {
     "CameraComponent": {
       "parent_actor": "carter_1",
-      "parent_T_camera": [1, 0, 0, 0, 0.11, 0.06, 0.31],
+      "parent_T_camera": [1, 0, 0, 0, 0.11, 0.06, 0.81],
       "cols": 960,
       "rows": 540,
       "fov": 90.0,

 "CarterRightCamera": {
     "CameraComponent": {
       "parent_actor": "carter_1",
-      "parent_T_camera": [1, 0, 0, 0, 0.11, -0.06, 0.31],
+      "parent_T_camera": [1, 0, 0.2, 0, -1, 0.0, 1.0],
       "cols": 960,
       "rows": 540,
       "fov": 90.0,

btalb, 2020-05-14:
We use one camera to get the 3rd person view seen when running BenchBot, and the other camera is raised so that we can see obstacles on tables, etc.

@david2611 can give further details.

liucsg, 2020-05-15:
Does it correspond to the output of depth ?

david2611, 2020-05-18:
Hi @liucsg. @btalb is correct. The left camera is the primary camera used in the challenge, and the right camera is a 3rd person camera used in our visualization of what the robot is doing (purely superficial).

We raise the left camera though to a height of 0.81 rather than the original 0.31 as otherwise the robot cannot see on top of tables. The 0.81 should correspond to the height (in metres) the camera is from the wheel-base of the robot.

Because of how Isaac configures things, both the colour and depth images come from this same camera.

closed 2020-05-20 by btalb