qcr/benchbot #8: Installing Benchbot on a GPU-accelerated VM?

achiatti, 2020-07-20:
Hi,
I have tried to install Benchbot from a Google Cloud Platform VM with GPU (Nvidia Quadro virtual workstation). 
All the Nvidia drivers are mounted correctly and visible from nvidia-smi. However the Benchbot install script
fails because it cannot access the graphics card:
[image]

Is there any alternative way to install Benchbot without direct access to the physical card?

Thanks!

btalb, 2020-07-20:
Thanks for sharing the issue with us @achiatti.

The check I have there is too restrictive and has caused issues in other cases. I'm doing a minor version update at the moment to improve the check, and I'd like to make sure the fix addresses the issue you're having as well. Are you able to post the output of `lspci -nn` on that VM for me so I can ensure my update addresses your issue?

As a hack so you can get up & running immediately, you can disable the check by commenting it out in the `checks_list_pre` variable in the install script:

```diff
index c96b2f7..e4fa636 100755
--- a/bin/benchbot_install
+++ b/bin/benchbot_install
@@ -105,7 +105,7 @@ checks_list_pre=(
   "Core host system checks:"
   "ubuntu1804"
   "Running Nvidia related system checks:"
-  "nvidiacard"  # Is a valid graphics card available?
+  # "nvidiacard"  # Is a valid graphics card available?
   "nvidiadriver"  # Is the driver available?
   "nvidiaversion"  # Is the driver version valid?
   "nvidiappa"  # Does the driver come from a supported PPA?
```

Let me know if you have any further issues.

achiatti, 2020-07-21:
Thanks for your reply and suggested quick-fix @btalb, and here you go:
```
00:00.0 Host bridge [0600]: Intel Corporation 440FX - 82441FX PMC [Natoma] [8086:1237] (rev 02)  
00:01.0 ISA bridge [0601]: Intel Corporation 82371AB/EB/MB PIIX4 ISA [8086:7110] (rev 03)   
00:01.3 Bridge [0680]: Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113] (rev 03)
00:03.0 Non-VGA unclassified device [0000]: Red Hat, Inc. Virtio SCSI [1af4:1004]
00:04.0 3D controller [0302]: NVIDIA Corporation Device [10de:1eb8] (rev a1)
00:05.0 Ethernet controller [0200]: Red Hat, Inc. Virtio network device [1af4:1000]
00:06.0 Unclassified device [00ff]: Red Hat, Inc. Virtio RNG [1af4:1005]
```

btalb, 2020-07-21:
Thanks for the information @achiatti. This confirms it is the same issue (card not being called "VGA compatible controller") which I've addressed with commit 46af27cc25af429a604f2ce58e3e37802e5b379a.

The fix will be live for everyone when I merge those bug fixes into master / do the next release in a couple of days.

btalb, 2020-07-23:
Fix released in 258bd60 as part of [release v1.0.2](https://github.com/RoboticVisionOrg/benchbot/releases).

Simply re-run `benchbot_install` to update.

closed 2020-07-23 by btalb