qcr/benchbot #13: Unable to install benchbot

gmuraleekrishna, 2021-03-02:
> +-----------------------------------------------------------------------------+
> | NVIDIA-SMI 460.32.03    Driver Version: 460.32.03    CUDA Version: 11.2     |
> |-------------------------------+----------------------+----------------------+
> | GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
> | Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
> |                               |                      |               MIG M. |
> |===============================+======================+======================|
> |   0  Quadro RTX 8000     On   | 00000000:1A:00.0 Off |                  Off |
> | 33%   35C    P8    17W / 260W |      6MiB / 48601MiB |      0%      Default |
> |                               |                      |                  N/A |
> +-------------------------------+----------------------+----------------------+
> |   1  Quadro RTX 8000     On   | 00000000:67:00.0  On |                  Off |
> | 34%   45C    P8    22W / 260W |    360MiB / 48598MiB |      0%      Default |
> |                               |                      |                  N/A |
> +-------------------------------+----------------------+----------------------+
> 

When I run `benchbot_run --robot carter --env miniroom:1 --task semantic_slam:active:ground_truth
`
i get the following error

`docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].`

btalb, 2021-03-03:
Thanks for reporting this to us @gmuraleekrishna.

This error seems to be coming from the [NVIDIA Container Toolkit](https://github.com/NVIDIA/nvidia-docker), which passes the GPU from your host OS to Docker containers.

It may be being caused by the two GPUs, or simply a bad state. I'd recommend the following:

1. Restarting the service (and then the computer if the error still occurs):
    ```
    sudo systemctl restart docker.service
    ```
2. Checking the following command works:
    ```
    docker run --rm --gpus all -it benchbot/backend:base
    ```
3. Trying some of the examples [here](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/user-guide.html#gpu-enumeration) to confirm GPU sharing is working
4. Posting the full output of `benchbot_install` so I can try and get a deeper understanding of what's going on:
    ```
    benchbot_install
    ```

Let me know how you go.

gmuraleekrishna, 2021-03-03:
@btalb Thanks for the reply. The steps 1-3 didn't work. We still get the error. The ouput for Step 4 is as follows.

```
$ benchbot_install

################################################################################
###################### CHECKING BENCHBOT SCRIPTS VERSION #######################
################################################################################

Fetching latest hash for Benchbot scripts ... 
		25459b5a9ae2bf8d5372cce9d1c3802c5e80cb16.

BenchBot scripts are up-to-date.

################################################################################
######################## PART 1: EXAMINING SYSTEM STATE ########################
################################################################################

Core host system checks:
	Ubuntu version >= 18.04:                                  Passed (18.04)

Running Nvidia related system checks:
	Nvidia GPU available:                     Found card of type '10de:1e30'
	Nvidia driver is running:                                          Found
	Nvidia driver version valid:                           Valid (460.32.03)
	Nvidia driver from a standard PPA:                          PPA is valid
	CUDA drivers installed:                                    Drivers found
	CUDA drivers version valid:                          Valid (460.32.03-1)
	CUDA drivers from the Nvidia PPA:                           PPA is valid
	CUDA is installed:                                            CUDA found
	CUDA version valid:                                         Valid (11.2)
	CUDA is from the Nvidia PPA:                                PPA is valid
	Isaac SDK archive in 'isaac' folder:                       Found archive

Running Docker related system checks:
	Docker is available:                                               Found
	Docker version valid:                                    Valid (20.10.4)
	Nvidia Container Toolkit installed:                        Found (1.3.3)
	Docker runs without root:                                         Passed

Running checks of filesystem used for Docker:
	/var/lib/docker on ext4 filesystem:                 Yes (/dev/nvme0n1p6)
	/var/lib/docker supports suid:                                   Enabled
	/var/lib/docker driver space check:               Sufficient space (84G)

Miscellaneous requirements:
	Pip python package manager available:                     Found (21.0.1)
	Tkinter for Python installed:                                      Found

All requirements & dependencies fulfilled. Docker containers for the BenchBot
software stack will now be built (which may take anywhere from a few seconds
to many hours). Would you like to proceed (y/N)? y

Proceeding with install ... 

################################################################################
################ PART 2: FETCHING LATEST BENCHBOT VERSION INFO #################
################################################################################

Fetching latest hash for BenchBot Simulator ... 
		d98690360b0c55d1b25c367d8b7d6ede1c618b0f.
Fetching latest hash for BenchBot Robot Controller ... 
		41f176ead1f972d54939b2120720d2fba8a401a0.
Fetching latest hash for BenchBot Supervisor ... 
		9e1cd82a3951feb204af1bfb134704d779a295ab.
Fetching latest hash for BenchBot API ... 
		785740ae3fc5dcfab1ec69fd5bf7eb5b741f995d.
Fetching latest hash for BenchBot ROS Messages ... 
		54487de28fae52c0d54a207b0c776736550fbd93.

[... 300 lines cut ...]
     |################################| 1.0 MB 10.5 MB/s 
Collecting matplotlib
  Downloading matplotlib-3.3.4-cp36-cp36m-manylinux1_x86_64.whl (11.5 MB)
     |################################| 11.5 MB 27.9 MB/s 
Requirement already satisfied: scipy in /home/benchbot/.local/lib/python3.6/site-packages (1.5.4)
Requirement already satisfied: numpy in /home/benchbot/.local/lib/python3.6/site-packages (1.19.5)
Collecting cycler>=0.10
  Downloading cycler-0.10.0-py2.py3-none-any.whl (6.5 kB)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.3 in /home/benchbot/.local/lib/python3.6/site-packages (from matplotlib) (2.4.7)
Collecting python-dateutil>=2.1
  Using cached python_dateutil-2.8.1-py2.py3-none-any.whl (227 kB)
Collecting pillow>=6.2.0
  Downloading Pillow-8.1.1-cp36-cp36m-manylinux1_x86_64.whl (2.2 MB)
     |################################| 2.2 MB 29.4 MB/s 
Collecting kiwisolver>=1.0.1
  Downloading kiwisolver-1.3.1-cp36-cp36m-manylinux1_x86_64.whl (1.1 MB)
     |################################| 1.1 MB 47.1 MB/s 
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from cycler>=0.10->matplotlib) (1.11.0)
Installing collected packages: python-dateutil, pillow, kiwisolver, cycler, shapely, matplotlib
Successfully installed cycler-0.10.0 kiwisolver-1.3.1 matplotlib-3.3.4 pillow-8.1.1 python-dateutil-2.8.1 shapely-1.7.1
sha256:2a076daf715a30d3428864e13b51bd5ddced4b1896e313f263a162c11d8db4db
tmp

Finished!
```

btalb, 2021-03-03:
Thanks for providing the logs @gmuraleekrishna . Sorry I can't provide an immediate solution; your configuration of 2xQuadros is not something we have available to test on in the lab.

Next debugging step is to narrow down the scope as much as possible. The command below isolates whether it is an issue with BenchBot, or the [Nvidia Container Toolkit](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#setting-up-nvidia-container-toolkit):
```
docker run --gpus all nvidia/cuda:11.1-base nvidia-smi
```

Can you show me the results of that command please? If it returns something like below it is a BenchBot issue, otherwise we need to figure out what is going on with Nvidia's Container Toolkit

```
ben@pc:~$ docker run --gpus all nvidia/cuda:11.1-base nvidia-smi
Unable to find image 'nvidia/cuda:11.1-base' locally
11.1-base: Pulling from nvidia/cuda
da7391352a9b: Pull complete 
14428a6d4bcd: Pull complete 
2c2d948710f2: Pull complete 
0ebd322634c1: Pull complete 
36520dd466ac: Pull complete 
fe6ccac2e64b: Pull complete 
Digest: sha256:c6bb47a62ad020638aeaf66443de9c53c6dc8a0376e97b2d053ac774560bd0fa
Status: Downloaded newer image for nvidia/cuda:11.1-base
Wed Mar  3 23:56:19 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 455.45.01    Driver Version: 455.45.01    CUDA Version: 11.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce GTX 1080    On   | 00000000:01:00.0 Off |                  N/A |
| 27%   32C    P8     9W / 180W |    492MiB /  8111MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+
```

hagianga21, 2021-03-04:
Hi @btalb, I am in his team. Thanks for your help. When running the cmd, I got:
```
Unable to find image 'nvidia/cuda:11.1-base' locally
11.1-base: Pulling from nvidia/cuda
da7391352a9b: Pull complete 
14428a6d4bcd: Pull complete 
2c2d948710f2: Pull complete 
0ebd322634c1: Pull complete 
36520dd466ac: Pull complete 
fe6ccac2e64b: Pull complete 
Digest: sha256:c6bb47a62ad020638aeaf66443de9c53c6dc8a0376e97b2d053ac774560bd0fa
Status: Downloaded newer image for nvidia/cuda:11.1-base
docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].
ERRO[0010] error waiting for container: context canceled
```
BTW, the CUDA version we have in the machine is 11.2, does it cause any problem?

btalb, 2021-03-04:
Thanks @hagianga21 . That information is good, it confirms that the error is being caused by the NVIDIA Container Toolkit ([example bug](https://github.com/NVIDIA/nvidia-docker/issues/1034)).

I'm trying to piece through this, but can't reproduce errors on our side.

1. Double check you have restarted the machine since installing the toolkit?
2. Did you install everything through the BenchBot installer? Or were some things installed manually?
3. You can try the previous command again with `nvidia/cuda:11.2-base` if you like, but I suspect it will also fail.

Otherwise, let's start digging into the NVIDIA Container Toolkit. Can you post the output of:
```
ben@pc:~$ nvidia-container-cli info
NVRM version:   455.45.01
CUDA version:   11.1

Device Index:   0
Device Minor:   0
Model:          GeForce GTX 1080
Brand:          GeForce
GPU UUID:       GPU-1ec26850-dd95-3255-3b1d-b2a944d1e50e
Bus Location:   00000000:01:00.0
Architecture:   6.1
```
Then if that works:
```
ben@pc:~$ nvidia-container-cli -d nvidia.log list
/dev/nvidiactl
/dev/nvidia-uvm
/dev/nvidia-uvm-tools
/dev/nvidia-modeset
/dev/nvidia0
/usr/bin/nvidia-smi
/usr/bin/nvidia-debugdump
/usr/bin/nvidia-persistenced
/usr/bin/nvidia-cuda-mps-control
/usr/bin/nvidia-cuda-mps-server
/usr/lib/x86_64-linux-gnu/libnvidia-ml.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-cfg.so.455.45.01
/usr/lib/x86_64-linux-gnu/libcuda.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-ptxjitcompiler.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-allocator.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-compiler.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-ngx.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-encode.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-opticalflow.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvcuvid.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-eglcore.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-tls.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-glsi.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-fbc.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-ifr.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-rtcore.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvoptix.so.455.45.01
/usr/lib/x86_64-linux-gnu/libGLX_nvidia.so.455.45.01
/usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.455.45.01
/usr/lib/x86_64-linux-gnu/libGLESv2_nvidia.so.455.45.01
/usr/lib/x86_64-linux-gnu/libGLESv1_CM_nvidia.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-glvkspirv.so.455.45.01
/usr/lib/x86_64-linux-gnu/libnvidia-cbl.so.455.45.01
/run/nvidia-persistenced/socket
```

And lastly paste the contents of `nvidia.log` back here.

hagianga21, 2021-03-04:
1. Yes, we restart the machine several times
2. Yes, I installed everything (including docker) through the BenchBot
3. 
```
(base) ecu@lab18:~/Projects/SSCD/benchbot$ nvidia-container-cli info
NVRM version:   460.32.03
CUDA version:   11.2

Device Index:   0
Device Minor:   0
Model:          Quadro RTX 8000
Brand:          Quadro
GPU UUID:       GPU-ac3f4f1a-a44f-f914-2b53-a4d9520f060a
Bus Location:   00000000:1a:00.0
Architecture:   7.5

Device Index:   1
Device Minor:   1
Model:          Quadro RTX 8000
Brand:          Quadro
GPU UUID:       GPU-638d034f-175e-324d-d52b-20d48f304932
Bus Location:   00000000:67:00.0
Architecture:   7.5
```
```
(base) ecu@lab18:~/Projects/SSCD/benchbot$ nvidia-container-cli -d nvidia.log list
/dev/nvidiactl
/dev/nvidia-uvm
/dev/nvidia-uvm-tools
/dev/nvidia-modeset
/dev/nvidia0
/dev/nvidia1
/usr/bin/nvidia-smi
/usr/bin/nvidia-debugdump
/usr/bin/nvidia-persistenced
/usr/bin/nvidia-cuda-mps-control
/usr/bin/nvidia-cuda-mps-server
/usr/lib/x86_64-linux-gnu/libnvidia-ml.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-cfg.so.460.32.03
/usr/lib/x86_64-linux-gnu/libcuda.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-ptxjitcompiler.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-allocator.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-compiler.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-ngx.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-encode.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-opticalflow.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvcuvid.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-eglcore.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-tls.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-glsi.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-fbc.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-ifr.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-rtcore.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvoptix.so.460.32.03
/usr/lib/x86_64-linux-gnu/libGLX_nvidia.so.460.32.03
/usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.460.32.03
/usr/lib/x86_64-linux-gnu/libGLESv2_nvidia.so.460.32.03
/usr/lib/x86_64-linux-gnu/libGLESv1_CM_nvidia.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-glvkspirv.so.460.32.03
/usr/lib/x86_64-linux-gnu/libnvidia-cbl.so.460.32.03
/run/nvidia-persistenced/socket
```

btalb, 2021-03-05:
Sorry about the delay in sorting this. I'm having troubles finding a cause in any of the logs you've provided me. It definitely appears to be an issue with the Nvidia Container Toolkit. I've been scouring [their documentation](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/overview.html) trying to find some hints, but haven't had any luck.

I've got a few possible things to try, but they are more hoping than based on any intuition:

1. Can I confirm at least one of your Quadros is connected to the display server? Can I see the full output of `nvidia-smi` on your host?

2. Confirm the BenchBot image definitely works without passing a GPU:
```
docker run --rm -it benchbot/backend:base
```
3. Try messing with the GPU argument to see if you can get something to work:
```
docker run --rm --gpus 0 nvidia/cuda:11.1-base nvidia-smi
docker run --rm --gpus 1 nvidia/cuda:11.1-base nvidia-smi
docker run --rm --gpus 0,1 nvidia/cuda:11.1-base nvidia-smi
docker run --rm --gpus 2 nvidia/cuda:11.1-base nvidia-smi
docker run --rm --gpus 3 nvidia/cuda:11.1-base nvidia-smi
```
4. Turn on debugging in `/etc/nvidia-container-runtime/config.toml`, restart the Docker service, and see if you can get any meaningful errors to appear in the log (see https://github.com/NVIDIA/nvidia-docker/wiki/Troubleshooting),
5. Install the entire NVIDIA Docker stack even though it shouldn't be needed:
```
sudo apt install nvidia-docker2
```

Sorry your first experiences with BenchBot have been this troublesome. Unfortunately, it seems to be caused by a dependency we have no control over.

My next step is to file an issue over at [NVIDIA/nvidia-docker](https://github.com/NVIDIA/nvidia-docker), but I'll need those debugging logs from step 4 before I can.

hagianga21, 2021-03-05:
1.
```
nvidia-smi
Fri Mar  5 15:54:34 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.32.03    Driver Version: 460.32.03    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Quadro RTX 8000     On   | 00000000:1A:00.0 Off |                  Off |
| 33%   34C    P8    16W / 260W |      6MiB / 48601MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  Quadro RTX 8000     On   | 00000000:67:00.0  On |                  Off |
| 34%   45C    P8    33W / 260W |    226MiB / 48598MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1769      G   /usr/lib/xorg/Xorg                  4MiB |
|    1   N/A  N/A      1769      G   /usr/lib/xorg/Xorg                162MiB |
|    1   N/A  N/A      3107      G   /usr/bin/compiz                    31MiB |
|    1   N/A  N/A      3157      G   ...mviewer/tv_bin/TeamViewer       15MiB |
|    1   N/A  N/A      5073      G   /usr/lib/firefox/firefox            3MiB |
|    1   N/A  N/A      5311      G   /usr/lib/firefox/firefox            3MiB |
|    1   N/A  N/A     13389      G   /usr/lib/firefox/firefox            3MiB |
|    1   N/A  N/A     13672      G   /usr/lib/firefox/firefox            3MiB |
+-----------------------------------------------------------------------------+
```
2. Yes, it works without GPU
```
docker run --rm -it benchbot/backend:base
benchbot@8aa96d76dbd0:/benchbot$
```
3. All commands got the same error:
```
docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]].
```
4. I uncommented the debug line, but then when restarting docker and try commands in step 3, no log file appears. Do I have to restart the machine?
5. Nothing new happens after running this line.

btalb, 2021-03-08:
I've opened the issue above to try and get to the bottom of this. Sorry it's been anything but straightforward. 

I've struggled with finding clarity in what's openly available describing the NVIDIA Docker stack.

Feel free to participate / add any extra info you think may be helpful directly to that issue.

hagianga21, 2021-03-08:
Hi, thanks so much for your support, I appreciate it.

btalb, 2021-03-12:
Any further progress on this @hagianga21 ?

Someone in the lab had exactly the same error come up yesterday. They were using Docker for something completely unrelated to BenchBot. But they were able to fix it by installing `nvidia-container-toolkit`, and rebooting.

Can I just double check what packages you have installed and where they are installed from. We're having real trouble being able to reproduce this error:
```
apt list --installed | grep nvidia
```
```
apt policy nvidia-container-toolkit
```

Here's the output of those commands on one of our working machines for reference:
```
ben@pc:~$ apt list --installed | grep nvidia                       
                                                                                                         
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
                                                    
libnvidia-cfg1-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-common-455/unknown,unknown,now 455.45.01-0ubuntu1 all [installed,automatic]
libnvidia-compute-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-container-tools/bionic,now 1.3.3-1 amd64 [installed,automatic]
libnvidia-container1/bionic,now 1.3.3-1 amd64 [installed,automatic]                             
libnvidia-decode-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-encode-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-extra-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]          
libnvidia-fbc1-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-gl-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]             
libnvidia-ifr1-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]
nvidia-compute-utils-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]     
nvidia-container-toolkit/bionic,now 1.4.2-1 amd64 [installed]
nvidia-dkms-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]              
nvidia-driver-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed]
nvidia-kernel-common-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]     
nvidia-kernel-source-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]
nvidia-modprobe/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]                      
nvidia-prime/now 0.8.15.3~0.20.04.1 all [installed,upgradable to: 0.8.16~0.20.04.1]
nvidia-settings/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]                      
nvidia-utils-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]
xserver-xorg-video-nvidia-455/unknown,unknown,now 455.45.01-0ubuntu1 amd64 [installed,automatic]

ben@pc:~$ apt policy nvidia-container-toolkit 
nvidia-container-toolkit:
  Installed: 1.4.2-1
  Candidate: 1.4.2-1
  Version table:
 *** 1.4.2-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
        100 /var/lib/dpkg/status
     1.4.1-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.4.0-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.3.0-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.2.1-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.2.0-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.1.2-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.1.1-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.1.0-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.0.5-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.0.4-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.0.3-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.0.2-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
```

hagianga21, 2021-03-12:
Hi, sorry for the delay. 
We 've already solved that but now, another problem happen. The stimulation window appears and then disappears immediately. 

About the output of double check:
```
ecu@lab18:~$ apt list --installed | grep nvidia

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libnvidia-cfg1-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
libnvidia-common-455/unknown,unknown,now 455.45.01-0ubuntu1 all [installed,auto-removable]
libnvidia-common-460/unknown,now 460.32.03-0ubuntu1 all [installed,automatic]
libnvidia-compute-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
libnvidia-container-tools/bionic,now 1.3.3-1 amd64 [installed,automatic]
libnvidia-container1/bionic,now 1.3.3-1 amd64 [installed,automatic]
libnvidia-decode-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
libnvidia-encode-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
libnvidia-extra-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
libnvidia-fbc1-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
libnvidia-gl-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
libnvidia-ifr1-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
nvidia-compute-utils-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
nvidia-container-runtime/bionic,now 3.4.2-1 amd64 [installed]
nvidia-container-toolkit/bionic,now 1.4.2-1 amd64 [installed]
nvidia-dkms-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
nvidia-driver-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
nvidia-kernel-common-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
nvidia-kernel-source-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
nvidia-modprobe/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
nvidia-prime/bionic-updates,bionic-updates,now 0.8.8.2 all [installed,automatic]
nvidia-settings/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
nvidia-utils-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]
xserver-xorg-video-nvidia-460/unknown,now 460.32.03-0ubuntu1 amd64 [installed,automatic]

```

```
 ecu@lab18:~$ apt policy nvidia-container-toolkit
nvidia-container-toolkit:
  Installed: 1.4.2-1
  Candidate: 1.4.2-1
  Version table:
 *** 1.4.2-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
        100 /var/lib/dpkg/status
     1.4.1-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.4.0-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.3.0-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.2.1-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.2.0-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.1.2-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.1.1-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.1.0-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.0.5-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.0.4-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.0.3-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
     1.0.2-1 500
        500 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  Packages
```

btalb, 2021-03-15:
Thanks @hagianga21 , I'm going to close this issue.

But feel free to create another describing your setup if that issue continues.

closed 2021-03-15 by btalb