qcr/benchbot #14: OMQ evaluated to 0.0 for semantic slam tutorial example

gmuraleekrishna, 2021-03-08:
We are trying to evaluate the baseline votenet (with pre-trained SUNRGBD) detection performance for `semantic_slam:passive:ground_truth` task. The solution script is unmodified code from [Part IV: Creating an Object-based Semantic Map from Detections](https://github.com/qcr/benchbot/wiki/Tutorial:-Performing-Semantic-SLAM-with-Votenet) in this repo.

The command we used to generate results:
```
benchbot_batch  --robo carter --task semantic_slam:passive:ground_truth --envs miniroom:1,miniroom:3,house:5 --native python my_solution.py
```
And the command used for evalulation:

```
benchbot_eval --method omq -o my_json_scores batch_0.json batch_1.json batch_2.json`
```
The result JSONs and score are here [results.zip](https://github.com/qcr/benchbot/files/6099297/results.zip).

For all 3 envs, we are getting an OMQ of 0.0. We were expecting OMQ to be > 0 because there are overlaps between GT and the detections [Figure_1] (Light colors are GTs and dark colors are detection). 
[image]

Can you please explain why the OMQ is 0.0?

btalb, 2021-03-08:
Hi @gmuraleekrishna . That definitely looks like a bug, thanks for reporting. We'll get this fixed asap. 

As you say, you should be getting a score above 0 (something like 0.11-0.13 is expected).

I'll assign this to @david2611, who'll look into it for you tomorrow.

gmuraleekrishna, 2021-03-08:
I found the issue with the wiki. The `save_result` function stores the detections in the `empty_results['objects']` instead of `emptu_results['results']['objects']`. As the evaluation script is expecting the latter, OMQ becomes 0. I have [updated the wiki,](https://github.com/gmuraleekrishna/benchbot/wiki/Tutorial:-Performing-Semantic-SLAM-with-Votenet) please merge it to this repo if convenient.

btalb, 2021-03-08:
Excellent. I've merged in that documentation change. 

Thanks for identifying and fixing that for us!

closed 2021-03-08 by btalb