qcr/benchbot #16: ERROR: cannot download default sources list

AndynWhen, 2021-03-18:
When I run "sudo ./install"
I get the follow error in building benchbot backend docker images

BUILDING BENCHBOT BACKEND DOCKER IMAGE:
Sending build context to Docker daemon  110.6MB
Step 1/36 : FROM benchbot/core:base
 ---> 82f8cca1412a
Step 2/36 : ENV ROS_WS_PATH="/benchbot/ros_ws"
 ---> Using cache
 ---> 9da9cdb2ad3d
Step 3/36 : RUN echo "deb http://packages.ros.org/ros/ubuntu bionic main" >     /etc/apt/sources.list.d/ros-latest.list &&     apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key     C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654 &&     apt update && apt install -y ros-melodic-desktop-full python-rosdep     python-rosinstall python-rosinstall-generator python-wstool     python-catkin-tools python-pip build-essential
 ---> Using cache
 ---> ad477e4329e3
Step 4/36 : RUN wget -qO - http://packages.lunarg.com/lunarg-signing-key-pub.asc |     apt-key add - && wget -qO /etc/apt/sources.list.d/lunarg-vulkan-bionic.list     http://packages.lunarg.com/vulkan/lunarg-vulkan-bionic.list &&     apt update && DEBIAN_FRONTEND=noninteractive apt install -yq vulkan-sdk
 ---> Using cache
 ---> 8ad96ddad140
Step 5/36 : RUN useradd --create-home --password "" benchbot && passwd -d benchbot &&     apt update && apt install -yq sudo && usermod -aG sudo benchbot &&     usermod -aG root benchbot && mkdir /benchbot &&     chown benchbot:benchbot /benchbot
 ---> Using cache
 ---> a9c2c02eb71f
Step 6/36 : USER benchbot
 ---> Using cache
 ---> 8bd97047c382
Step 7/36 : WORKDIR /benchbot
 ---> Using cache
 ---> 8ad0a705edcf
Step 8/36 : RUN sudo rosdep init && rosdep update &&     mkdir -p ros_ws/src && source /opt/ros/melodic/setup.bash &&     pushd ros_ws && catkin_make && source devel/setup.bash && popd
 ---> Running in 2c2ce8c1ac5f
ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.
The command '/bin/bash -c sudo rosdep init && rosdep update &&     mkdir -p ros_ws/src && source /opt/ros/melodic/setup.bash &&     pushd ros_ws && catkin_make && source devel/setup.bash && popd' returned a non-zero code: 4

ERROR: Building BenchBot backend returned a non-zero error code: 1

Ensure that Docker has been installed correctly AND that you can run Docker 
WITHOUT root access (there is no need to ever run Docker with root). See 
https://docs.docker.com/install/linux/linux-postinstall/ for details on how to 
fix this. 

If the error is more generic, please contact us so that we can update our
pre-install host system checks.

-------------------------------------------------------------------------------------------------------------------------
I encounter the same problem in installing ROS. It work fine after changing the proxy config in my PC,but in Docker it seems to be still wrong.

How can I solve this problem, change proxy settings of docker or installation instructions?

btalb, 2021-03-18:
Thanks for reporting @AndynWhen; looks like an interesting issue. 

One thing I'd recommend first, is not using `sudo` when installing BenchBot. We only use root permissions in one place (installing convenience symlinks for you), and we ask for the permission at that point.

As you reported, the failed content is definitely live and accessible (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list). So it definitely sounds like an issue related to the proxy configuration. I need to understand more about your proxy configuration to help. 

In theory, a change of your proxy configuration at the system level should mean all traffic flows through that configuration (including Docker).

How are you changing your proxy settings / can you share them if they're not sensitive?

AndynWhen, 2021-03-19:
I'm using Qv2Ray to set system proxy with "socks5:127.0.0.1:1088" , "http://localhost:8888"
Other configuration :

➜  ~ printenv http_proxy https_proxy
http://localhost:8888/
http://localhost:8888/
➜  ~ systemctl show --property=Environment docker                   
Environment=HTTP_PROXY=http://localhost:8888/

When I run "sudo rosdep init", it seems to be OK:
 ~ sudo rosdep init
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
Recommended: please run

	rosdep update

Maybe I can try to change instructions in docker build such as changing "sudo rosdep init" to "sudo -E rosdep init"? It also make sense in the past.

btalb, 2021-03-22:
Thanks for the extra information @AndynWhen .

Given the proxy config you've described requires environment variables to be followed, I can understand why it isn't working inside the Docker image. We don't explicitly pass those variables through as they are conventions not standards.

There's a few options for how we can get this sorted though:

1. [Configure the Docker client to use your proxy settings](https://docs.docker.com/network/proxy/#configure-the-docker-client). If we can get this working, it is the best solution by far as it doesn't involve re-writing software for use specific settings.
2. I rework the BenchBot scripts to pass proxy settings through. It would mean editing [these sort of lines](https://github.com/qcr/benchbot/blob/783e9ca27866ae64c891262aa5f46a7a53ee37c7/bin/benchbot_install#L1152) all throughout the scripts to manually pass the environment variables through. This is do-able, but not a good solution in practice.
3.  Play with [Qv2ray's proxychain settings](https://qv2ray.net/getting-started/step4.html#using-proxychains) to use the proxy at a system level rather than 'environment variable convention' level.

Please try the first option though, and see if you can get that to work. I'd recommend using the following process:
1. Adjust the settings in your `~/.docker/config.json` file
2. Restart all your Docker services via `sudo systemctl restart docker`
3. Test accessibility in a small test container before worrying about installing BenchBot: 
```
docker run --rm -it ubuntu:bionic /bin/bash -c 'apt update -y && apt install -y curl && curl https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list'
```

When it's working the above command should end with something like:
```
...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
# os-specific listings first
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx

# generic
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
gbpdistro https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml fuerte

# newer distributions (Groovy, Hydro, ...) must not be listed anymore, they are being fetched from the rosdistro index.yaml instead
```

Let me know how you go!

closed 2021-04-20 by btalb