qcr/benchbot #17: Benchbot core installation fails with "invalid CUDA version: 11.2.2-1: unknown"

gmuraleekrishna, 2021-03-22:
Hello, 

We tried to reinstall the benchbot after updating the repo. We are getting an error at Benchbot core installation. 

```
OCI runtime create failed: container_linux.go:367: starting container process caused: process_linux.go:495: container init caused: 
Running hook #1:: error running hook: exit status 1, stdout: , stderr: invalid CUDA version: 11.2.2-1: unknown
```
We had uninstalled both nvidia and cuda drivers to let the installation script to install them.  Both versions are chosen by the `benchbot_install` script.

Full log is attached. [benchbot_install.txt](https://github.com/qcr/benchbot/files/6181940/benchbot_install.txt)

btalb, 2021-03-22:
Thanks for reporting @gmuraleekrishna.

Is the error you've quoted there from `benchbot_run` after the failed install?

I've just updated our test machine to CUDA 11.2 & NVIDIA driver 460. The only odd issue I had is that on upgrading the driver, `nvidia-smi` wasn't happy. I had to reboot before `benchbot_install` would work.

Everything proceeded smoothly from there though.

Can I confirm this issue still occurs post-reboot?

gmuraleekrishna, 2021-03-23:
We have rebooted the system multiple times. Here is the output for `nvidia-smi`
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.32.03    Driver Version: 460.32.03    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Quadro RTX 8000     On   | 00000000:1A:00.0 Off |                    0 |
| 33%   35C    P8    16W / 260W |      5MiB / 45553MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  Quadro RTX 8000     On   | 00000000:67:00.0 Off |                    0 |
| 34%   41C    P8    19W / 260W |     99MiB / 45550MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1629      G   /usr/lib/xorg/Xorg                  4MiB |
|    1   N/A  N/A      1629      G   /usr/lib/xorg/Xorg                 97MiB |
+-----------------------------------------------------------------------------+
```
```
$nvcc  --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Sun_Feb_14_21:12:58_PST_2021
Cuda compilation tools, release 11.2, V11.2.152
Build cuda_11.2.r11.2/compiler.29618528_0
```
Can you tell us how `core.Dockerfile` is finding all the apt packages to install? May be there are clues there?

btalb, 2021-03-23:
Interesting, we have exactly the same versions & outputs on our working machine.

We are following the official steps [here](https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&target_distro=Ubuntu&target_version=2004&target_type=debnetwork), which the installer would've used to upgrade your drivers / CUDA on your host system. Which makes the problem even more confusing.

The log isn't clearly showing me what part of the Docker RUN command is failing, we should debug that. An easy way is to incrementally build the line up and see where it goes from succeeding to failing.

So, above [this line](https://github.com/qcr/benchbot/blob/783e9ca27866ae64c891262aa5f46a7a53ee37c7/docker/core.Dockerfile#L21) in `Docker/core.Dockerfile`:
```
RUN add-apt-repository ppa:graphics-drivers && \
    wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/cuda-ubuntu1804.pin && \
    mv -v cuda-ubuntu1804.pin /etc/apt/preferences.d/cuda-repository-pin-600 && \
    apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/7fa2af80.pub && \
    add-apt-repository -n "deb http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/ /" && \
    apt update
```
Add something like the following:
```
RUN echo "TEST START"
RUN echo "TEST PASSED"; false
```
Your `benchbot_install` should now reliably fail but print `TEST PASSED`. Now in between those lines start building up the failing line until you don't see `TEST_PASSED`. For example:
```
RUN echo "TEST START"
RUN add-apt-repository ppa:graphics-drivers
RUN echo "TEST PASSED"; false
```
then 
```
RUN echo "TEST START"
RUN add-apt-repository ppa:graphics-drivers && \
    wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/cuda-ubuntu1804.pin 
RUN echo "TEST PASSED"; false
```
then
```
RUN echo "TEST START"
RUN add-apt-repository ppa:graphics-drivers && \
    wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/cuda-ubuntu1804.pin && \
    mv -v cuda-ubuntu1804.pin /etc/apt/preferences.d/cuda-repository-pin-600
RUN echo "TEST PASSED"; false
```
etc.

Let me know when you've pinned down what part of the command is failing and we can continue from there.

closed 2021-04-27 by btalb