qcr/benchbot #19: OMQ improved after uptading the repo

gmuraleekrishna, 2021-03-24:
Hi,

As we reported earlier we obtained 0.13 OMQ from benchbot evaluation of  miniroom:1 on default votenet. Now after updating the repo, the OMQ improved to 0.21. The eval.ai shows your baseline as 0.41.  What changed in the evaluation from before?

david2611, 2021-03-24:
Hey @gmuraleekrishna. The baseline on EvalAI with the score of is just a simple method one of us in the lab came up with which is not part of any tutorial. The methodology is simplistic but a step above what we present in the tutorial which is really just meant as a jumping off point to show how to do the entire pipeline from beginning to end, outputting some meaningful result.

Regarding the improved OMQ score you have, I require more information. 

Did you run an experiment, save the results.json and use `benchbot_eval` on the results file before and after the update, or, did you run your experiment from beginning to end, with evaluation included before and after the update?

Was the experiment only on miniroom:1 using the default votenet?

Did you notice any difference in traversal between the two runs?

In subsequent attempts have you gotten a consistent 0.21 score or 0.13 score?

Any extra info can help us sort out what might be going on so all info is appreciated.

david2611, 2021-04-15:
Just pinging this issue again. If there have been no developments and you can provide us with no further info, I will close this issue.

closed 2021-04-27 by btalb