qcr/benchbot #20: TypeError: argument of type 'NoneType' is not iterable

tyou1, 2021-04-09:
Hi,

I am coming from issue [18](https://github.com/qcr/benchbot/issues/18), now I am using nomachine to connect to the remote machine;

I encountered a new error when I run
`<benchbot_run --robot carter --env miniroom:1 --task semantic_slam:passive:ground_truth>`

There is no "robot controller error" & vulkan crash log like before but unfortunately now I have this new error :

>################################################################################\
################### BENCHBOT IS RUNNING (Ctrl^C to exit) ... ###################\
################################################################################\
\
Initialising supervisor...\
\
Configuring the supervisor...\
Traceback (most recent call last):\
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main\
    "__main__", mod_spec)\
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code\
    exec(code, run_globals)\
  File "/home/benchbot/.local/lib/python3.6/site-packages/benchbot_supervisor/__main__.py", line 27, in <module>\
    addons_path=args.addons_path)\
  File "/home/benchbot/.local/lib/python3.6/site-packages/benchbot_supervisor/benchbot_supervisor.py", line 61, in __init__\
    self.load()\
  File "/home/benchbot/.local/lib/python3.6/site-packages/benchbot_supervisor/benchbot_supervisor.py", line 82, in load\
    \{\} if 'results_format' not in self.config['task'] else\
TypeError: argument of type 'NoneType' is not iterable\
\
################################################################################\
###################### CLEANING UP ALL BENCHBOT REMNANTS #######################\
################################################################################\

I wonder if anyone has encountered the same problem as well?

Many Thanks!

btalb, 2021-04-12:
That's an interesting one @tyou1 ; I haven't encountered this error before.

There is a bit of a smell in the logs of something being wrong with the add-ons content, but I'm not sure at this stage. Can I please gather a bit more information to get a better understanding:

1. Information about the currently installed BenchBot add-ons:
```
benchbot_install --list-addons
```
2. Information about currently installed result formats:
```
benchbot_run --list-formats
```
3. A check that all of the BenchBot libraries can be found in your current Python environment:
```
pip3 list | grep benchbot
```
4. A full dump of the output above:
```
benchbot_run --robot carter --env miniroom:1 --task semantic_slam:passive:ground_truth 2>&1 | tee benchbot.log
```

Thanks!

tyou1, 2021-04-12:
Thanks for looking into this!

For the first 3 commands, it shows:

    tyou@benchbot:~$ benchbot_install --list-addons 

    Currently installed add-ons:
	benchbot-addons/batches_isaac (c4023d11)
	benchbot-addons/envs_isaac_challenge (db8c9a0f, with remote content)
	benchbot-addons/envs_isaac_develop (47fd00e8, with remote content)
	benchbot-addons/eval_omq (b88b7070)
	benchbot-addons/examples_base (ee8ada6f)
	benchbot-addons/examples_ssu (4a74f4cf)
	benchbot-addons/formats_object_map (85bc18de)
	benchbot-addons/ground_truths_isaac_develop (1beb7f54)
	benchbot-addons/robots_isaac (38403a0c)
	benchbot-addons/ssu (ca495363)
	benchbot-addons/tasks_ssu (74ff18bf)

    Our GitHub organisation (https://github.com/benchbot-addons) contains all of our official add-ons.
    The following additional official add-ons are available, with more details at the above URL:
	benchbot-addons/data_collect
	benchbot-addons/examples_data_collect
	benchbot-addons/tasks_data_collect

    If you would like to add your community-created add-on to the official list, please follow the instructions here:
	https://github.com/qcr/benchbot-addons

    tyou@benchbot:~$ benchbot_run --list-formats
    Formats are used by a task to declare the formats of results in a re-usable manner. You should ensure that tasks you use point to installed results formats. The following formats are supported in your BenchBot installation:
    
	object_map
	object_map_ground_truth
	object_map_with_states

    See the '--show-format FORMAT_NAME' command for specific details about each format, or check you have the appropriate add-on installed if you are missing a format.

    tyou@benchbot:~$ pip3 list | grep benchbot
    benchbot-addons         2.1.0                /home/tyou/benchbot/addons
    benchbot-api            2.1.0                /home/tyou/benchbot/api   
    benchbot-eval           2.0.0                /home/tyou/benchbot/eval  
 

The full benchbot.log is in the file attached;

Thanks!

[benchbot.log](https://github.com/qcr/benchbot/files/6296686/benchbot.log)

btalb, 2021-04-16:
Sorry about the delay @tyou1 .

I haven't been able to reproduce this error on any of our systems, and am a little stumped to be honest.

Can you make sure you upgrade to the latest  version  which was released yesterday via `benchbot_install`?

The log suggests the supervisor is receiving a blank task configuration, but I have no idea how that could happen. Does this error occur for all tasks, or just the one you showed?

And lastly can I just check your system can find the benchbot hosts:
```
cat /etc/hosts | grep benchbot | cut -d' ' -f2 | xargs getent hosts
```
Should output something something like:
```
172.20.0.100    benchbot_ros
172.20.0.101    benchbot_robot
172.20.0.102    benchbot_supervisor
172.20.0.200    benchbot_debug
```
If there's any other information you think may be relevant / helpful, please let me know.

tyou1, 2021-04-21:
Sorry about the delay @btalb.

I have upgraded BenchBot, I just want to check if v2.1.1 is the latest version?

    tyou@benchbot:~$ benchbot_run —v
    BenchBot Software Stack.
    Version 'HEAD -> master, v2.1.1', from branch 'tags/v2.1.1^0'
    (3d41cab9be053470e420f16549fbd8d5304556c8)

Also, for the command to find the benchbot hosts, it shows:

    tyou@benchbot:~$ cat /etc/hosts | grep benchbot | cut -d' ' -f2 | xargs getent hosts
    127.0.1.1       benchbot
    127.0.1.1       benchbot
    172.20.0.100    benchbot_ros
    172.20.0.101    benchbot_robot
    172.20.0.102    benchbot_supervisor
    172.20.0.200    benchbot_debug
    
I tried again with
    
    benchbot_run --robot carter --env miniroom:1 --task semantic_slam:passive:ground_truth 2>&1 | tee benchbot.log

this time the benchbot log suggests a robot controller error (the log is attached in the file),  I think this issue is similar to the Vulcan crush [18](https://github.com/qcr/benchbot/issues/18)?
 
After that I did as suggested in [18](https://github.com/qcr/benchbot/issues/18) to run `xhost local:root`, and commented out the line `xhost -local:root > /dev/null`,

and also tried with the vkcube command:

    tyou@benchbot:~$ docker run -e DISPLAY --volume /tmp/.X11-unix:/tmp/.X11-unix --rm --gpus all -it benchbot/backend:base /bin/bash -c 'vkcube'
    No protocol specified
    Cannot find a compatible Vulkan installable client driver (ICD).
    Exiting ...
    
    tyou@benchbot:~$ docker run -e DISPLAY --volume /tmp/.X11-unix:/tmp/.X11-unix --rm --gpus all -it benchbot/backend:base /bin/bash -c 'vulkaninfo'
    No protocol specified
    WARNING: [Loader Message] Code 0 : loader_icd_scan: Can not find 'ICD' object in ICD JSON file  /usr/share/vulkan/icd.d/nvidia_layers.json.  Skipping ICD JSON
    error: XDG_RUNTIME_DIR not set in the environment.
    No protocol specified
    XCB failed to connect to the X server due to error:1.
    ERROR at /build/vulkan-tools-1.2.162.1~rc1-1lunarg18.04/vulkaninfo/vulkaninfo.h:847: AppCreateXcbSurface failed to establish connection

So I figure maybe the failing vkcube command maybe the problem? What might be the possible way to fix this problem then?

Many thanks!

[benchbot.log](https://github.com/qcr/benchbot/files/6352827/benchbot.log)

btalb, 2021-04-27:
Sorry about the delay @tyou1 ; that's some really productive debugging.

I've run `vulkaninfo` on a working test machine & got the following:

<details>
<summary>`vulkaninfo` log</summary>
<p>

```
ben@pc:~$ docker run -e DISPLAY --volume /tmp/.X11-unix:/tmp/.X11-unix --rm --gpus all -it benchbot/backend:base /bin/bash -c 'vulkaninfo'
WARNING: [Loader Message] Code 0 : loader_icd_scan: Can not find 'ICD' object in ICD JSON file /usr/share/vulkan/icd.d/nvidia_layers.json.  Skipping ICD JSON
error: XDG_RUNTIME_DIR not set in the environment.
==========
VULKANINFO
==========

Vulkan Instance Version: 1.2.162

Instance Extensions: count = 17
===============================
	VK_EXT_acquire_xlib_display            : extension revision 1
	VK_EXT_debug_report                    : extension revision 9
	VK_EXT_debug_utils                     : extension revision 2
	VK_EXT_direct_mode_display             : extension revision 1
	VK_EXT_display_surface_counter         : extension revision 1
	VK_KHR_device_group_creation           : extension revision 1
	VK_KHR_display                         : extension revision 23
	VK_KHR_external_fence_capabilities     : extension revision 1
	VK_KHR_external_memory_capabilities    : extension revision 1
	VK_KHR_external_semaphore_capabilities : extension revision 1
	VK_KHR_get_display_properties2         : extension revision 1
	VK_KHR_get_physical_device_properties2 : extension revision 2
	VK_KHR_get_surface_capabilities2       : extension revision 1
	VK_KHR_surface                         : extension revision 25
	VK_KHR_surface_protected_capabilities  : extension revision 1
	VK_KHR_xcb_surface                     : extension revision 6
	VK_KHR_xlib_surface                    : extension revision 6

Layers: count = 6
=================
VK_LAYER_KHRONOS_validation (Khronos Validation Layer) Vulkan version 1.2.162, layer version 1:
	Layer Extensions: count = 3
		VK_EXT_debug_report        : extension revision 9
		VK_EXT_debug_utils         : extension revision 1
		VK_EXT_validation_features : extension revision 2
	Devices: count = 1
		GPU id = 0 (GeForce GTX 1080)
		Layer-Device Extensions: count = 3
			VK_EXT_debug_marker     : extension revision 4
			VK_EXT_tooling_info     : extension revision 1
			VK_EXT_validation_cache : extension revision 1

VK_LAYER_LUNARG_api_dump (LunarG API dump layer) Vulkan version 1.2.162, layer version 2:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id = 0 (GeForce GTX 1080)
		Layer-Device Extensions: count = 1
			VK_EXT_tooling_info : extension revision 1

VK_LAYER_LUNARG_device_simulation (LunarG device simulation layer) Vulkan version 1.2.162, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id = 0 (GeForce GTX 1080)
		Layer-Device Extensions: count = 1
			VK_EXT_tooling_info : extension revision 1

VK_LAYER_LUNARG_gfxreconstruct (GFXReconstruct Capture Layer Version 0.9.5-unknown) Vulkan version 1.2.162, layer version 36869:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id = 0 (GeForce GTX 1080)
[... 1309 lines cut ...]
	vulkanMemoryModelDeviceScope                  = true
	vulkanMemoryModelAvailabilityVisibilityChains = true

VkPhysicalDeviceYcbcrImageArraysFeaturesEXT:
--------------------------------------------
	ycbcrImageArrays = true
```

</p>
</details>

To me the key difference there is these lines:
```
No protocol specified
XCB failed to connect to the X server due to error:1.
```

which suggests to me you're having a trouble with creating X windows from that machine. If we do the following:

```
docker run -e DISPLAY --volume /tmp/.X11-unix:/tmp/.X11-unix --rm --gpus all -it benchbot/backend:base /bin/bash -c 'sudo apt install x11-apps && xeyes'
```
do you get some eyes showing up? If not I'd suggest having a look into some of these to understand what's happening with the connection between your X server in the Docker container and the X windowing client on your end:
- https://unix.stackexchange.com/questions/209746/how-to-resolve-no-protocol-specified-for-su-user
- https://unix.stackexchange.com/questions/118811/why-cant-i-run-gui-apps-from-root-no-protocol-specified

tyou1, 2021-04-27:
@btalb Thanks for the info! It seems that creating x windows (`xeyes, xclock…`)  works fine on the machine;

Both commands to debug vulkaninfo and vkcube are working now, for me the key is the line ` xhost +local:host `:

[image]

However, after that the command `benchbot_run --robot carter --env miniroom:1 --task semantic_slam:passive:ground_truth` still doesn't work, unfortunately. (file is attached)

[benchbot1.log](https://github.com/qcr/benchbot/files/6383048/benchbot1.log)

I am not sure that commenting out this line works for me (?), since the error still exits;
https://github.com/qcr/benchbot/blob/783e9ca27866ae64c891262aa5f46a7a53ee37c7/bin/benchbot_run#L347

Is there anything else I should also pay attention to while I fix this problem (as non-root user)? 

Thanks!

btalb, 2021-04-27:
The above line simply disables those insecure `xhost` permissions, after we enable them earlier to start the Docker containers. To me, it shouldn't have any effect on the success of things and shouldn't need to be commented out.

I've had a look through the error log, and it's still showing symptoms of Vulkan issues:
```
[2021.04.27-08.09.00:640][  0]LogRHI: Warning: Failed to find entry point for vkDestroySurfaceKHR
[2021.04.27-08.09.00:640][  0]LogRHI: Warning: Failed to find entry point for vkGetPhysicalDeviceSurfaceSupportKHR
[2021.04.27-08.09.00:640][  0]LogRHI: Warning: Failed to find entry point for vkGetPhysicalDeviceSurfaceCapabilitiesKHR
[2021.04.27-08.09.00:640][  0]LogRHI: Warning: Failed to find entry point for vkGetPhysicalDeviceSurfaceFormatsKHR
[2021.04.27-No protocol specified
08.09.00:640][  0]LogRHI: Warning: Failed to find entry point for vkGetPhysicalDeviceSurfacePresentModesKHR
[2021.04.27-08.09.00:645][  0]LogLinux: Warning: MessageBox: Failed to find all required Vulkan entry points! Try updating your driver.: No Vulkan entry points found!: 
```

The final line is the issue, and it is interesting there is that `[2021.04.27-No protocol specified` error popping up again. 

Can you try the following for me (all in the same terminal so we can ensure consistent state):

1. Make sure `vkcube` runs successfully in the terminal like you did above
2. Change these two lines by getting rid of the output suppression `> /dev/null` (but keep them both uncommented):
    https://github.com/qcr/benchbot/blob/3d41cab9be053470e420f16549fbd8d5304556c8/bin/benchbot_run#L310
    https://github.com/qcr/benchbot/blob/3d41cab9be053470e420f16549fbd8d5304556c8/bin/benchbot_run#L347
3. Log the results
4. If it still fails, comment out both the lines above and try again (ensure `vkcube` still works in the terminal)
5. If it still fails, log the results of the `vulkaninfo` command to share with me

Also can you let me know what state your BenchBot repository is in after these changes by showing me the results of the following from the directory where it is cloned:
```
git status
```
```
git diff
```

Thanks @tyou1 !

tyou1, 2021-04-28:
Thanks! @btalb, unfortunately it still fails:

1. This is the crash log  when getting rid of the output suppression `> /dev/null` in these 2 lines:

https://github.com/qcr/benchbot/blob/3d41cab9be053470e420f16549fbd8d5304556c8/bin/benchbot_run#L310

https://github.com/qcr/benchbot/blob/3d41cab9be053470e420f16549fbd8d5304556c8/bin/benchbot_run#L347

[benchbot2.log](https://github.com/qcr/benchbot/files/6393315/benchbot2.log)

and the log when commenting out both the lines above:

[benchbot3.log](https://github.com/qcr/benchbot/files/6393344/benchbot3.log)

2. The `vulkaninfo` runs the following output after the command `xhost +local:host`:

<details><summary>vulkaninfo</summary>
<p>

```
tyou@benchbot:~$ xhost +local:host
non-network local connections being added to access control list

tyou@benchbot:~$ docker run -e DISPLAY --volume /tmp/.X11-unix:/tmp/.X11-unix --rm --gpus all -it benchbot/backend:base /bin/bash -c 'vulkaninfo'

WARNING: [Loader Message] Code 0 : loader_icd_scan: Can not find 'ICD' object in ICD JSON file /usr/share/vulkan/icd.d/nvidia_layers.json.  Skipping ICD JSON
error: XDG_RUNTIME_DIR not set in the environment.
==========
VULKANINFO
==========

Vulkan Instance Version: 1.2.162

Instance Extensions: count = 17
===============================
	VK_EXT_acquire_xlib_display            : extension revision 1
	VK_EXT_debug_report                    : extension revision 9
	VK_EXT_debug_utils                     : extension revision 2
	VK_EXT_direct_mode_display             : extension revision 1
	VK_EXT_display_surface_counter         : extension revision 1
	VK_KHR_device_group_creation           : extension revision 1
	VK_KHR_display                         : extension revision 23
	VK_KHR_external_fence_capabilities     : extension revision 1
	VK_KHR_external_memory_capabilities    : extension revision 1
	VK_KHR_external_semaphore_capabilities : extension revision 1
	VK_KHR_get_display_properties2         : extension revision 1
	VK_KHR_get_physical_device_properties2 : extension revision 2
	VK_KHR_get_surface_capabilities2       : extension revision 1
	VK_KHR_surface                         : extension revision 25
	VK_KHR_surface_protected_capabilities  : extension revision 1
	VK_KHR_xcb_surface                     : extension revision 6
	VK_KHR_xlib_surface                    : extension revision 6

Layers: count = 6
=================
VK_LAYER_KHRONOS_validation (Khronos Validation Layer) Vulkan version 1.2.162, layer version 1:
	Layer Extensions: count = 3
		VK_EXT_debug_report        : extension revision 9
		VK_EXT_debug_utils         : extension revision 1
		VK_EXT_validation_features : extension revision 2
	Devices: count = 1
		GPU id = 0 (NVIDIA GeForce RTX 3080)
		Layer-Device Extensions: count = 3
			VK_EXT_debug_marker     : extension revision 4
			VK_EXT_tooling_info     : extension revision 1
			VK_EXT_validation_cache : extension revision 1

VK_LAYER_LUNARG_api_dump (LunarG API dump layer) Vulkan version 1.2.162, layer version 2:
	Layer Extensions: count = 0
	Devices: count = 1
[... 1418 lines cut ...]
@@ -307,7 +307,7 @@ if [ -z "$simulator_required" ]; then
 fi
 
 # Declare reusable parts to ensure our containers run with consistent settings
-xhost +local:root > /dev/null
+# xhost +local:root > /dev/null
 ros_master_host="benchbot_roscore"
 docker_run="docker run -t --gpus all \
             --env DISPLAY \
@@ -344,7 +344,7 @@ ${cmd// /$'\t'} --ip "$URL_SUPERVISOR" -d $DOCKER_TAG_BACKEND /bin/bash -c \
 echo -e "\n${colour_blue}Starting container for BenchBot Debugging:${colour_nc}"
 cmd="${docker_run//'$name'/benchbot_debug}"
 ${cmd// /$'\t'} --ip "$URL_DEBUG" -it -d $DOCKER_TAG_BACKEND /bin/bash
-xhost -local:root > /dev/null
+# xhost -local:root > /dev/null 
 
 # Print the output of the Supervisor, watching for failures
 header_block "BENCHBOT IS RUNNING (Ctrl^C to exit) ..." ${colour_green}
tyou@benchbot:~/benchbot$ 
```

</p>
</details>

Thanks!

btalb, 2021-04-29:
Thanks for the logs @tyou1. I've combed through, but aren't finding too much of use in terms of pointing us to the cause.

I've got the following thoughts as to how to dig deeper:

1. Try a fresh install of BenchBot just for sanity's sake:
    ```
    ./install --uninstall
    ./install
    ```
2. I sifted through your `vulkaninfo` log and compared it to a working machine. The only noticeable difference (besides your GPU having a lot more capabilities than ours :stuck_out_tongue:), is this warning which appears in your log:
    ```
    WARNING: [Loader Message] Code 0 : loader_icd_scan: Can not find 'ICD' object in ICD JSON file /usr/share/vulkan/icd.d/nvidia_layers.json.  Skipping ICD JSON
    ```
    I am not sure whether that warning is a hint to something more sinister, but it could be worth digging into. For your records, here is a full dump of `vulkaninfo` on our working machine: [working.log](https://github.com/qcr/benchbot/files/6396850/working.log)

3. Dig deep into the internals, and try running the simulator manually from a Docker container that we validate has `vkcube` working:
   a. Start a new container like before, but make sure the addons directory is passed through as a volume mount (note the path should match where the addons directory exists on your machine):
      ```
      docker run -e DISPLAY --volume /tmp/.X11-unix:/tmp/.X11-unix --volume /home/benchbot/benchbot/addons/benchbot_addons:/addons:ro --rm --gpus all -it benchbot/backend:base /bin/bash
      ```
   b. Get the `vkcube` example running in this container, don't proceed until it is working within the container.
   c. Move to the directory where all of the simulator data is stored:
      ```
      cd /addons/benchbot-addons/envs_isaac_develop/environments/.sim_package
      ```
   d. Run this pretty gross command, which should get an Isaac simulator window open with `miniroom:1`:
      ```
      /IsaacSimProject.sh "/Game/AI_vol3_03_base/Maps/AI_vol3_scene_03" -isaac_sim_config_json='/benchbot/isaac_sdk/apps/carter/carter_sim/bridge_config/carter_full.json' -windowed -ResX=960 -ResY=540 -vulkan -game
      ```

If all else fails, you can also try [our OpenGL version of the simulator](https://github.com/qcr/benchbot/wiki/FAQs#i-need-to-get-around-the-vulkan-related-crashes-to-use-benchbot-remotely), which is used by selecting `carter_remote` as the robot instead of `carter`. 

Let me know how you go.

tyou1, 2021-05-08:
Sorry about the delay @btalb ,

Thanks for the info! I looked into the `vulkaninfo` and tried the following on the machine, unfortunately the vkcube doesn't seem to be working in the container and I am still looking into this.

Meanwhile, [the OpenGL version of the simulator](https://github.com/qcr/benchbot/wiki/FAQs#i-need-to-get-around-the-vulkan-related-crashes-to-use-benchbot-remotely) looks like a good option to me 😃

(This issue can be closed from me, thanks for all the help on this! @btalb )