qcr/benchbot #21: Benchbt Installation error 24/36

ernestoMM94, 2021-04-10:
Error: Building benchbot backend return a non zero error code: 1

btalb, 2021-04-12:
Thanks for reporting @ernestoMM94 . I'm including the images sent via email with this issue to hopefully help any other people encountering the same issue.

Please let me know if this is not okay.

[image]
[image]
[image]

btalb, 2021-04-12:
As to the issue, from what I can see there seems to have been some sort of internet issues during the installation process.

Is the machine behind a proxy like in issue #16 ?

If not, the only evidence I have from those logs is that there was disrupted internet connectivity which will likely resolve itself if install is run again.

Can you run `benchbot_install` again, but this time also log all output to a file. Post the log file here if install fails again:

```
benchbot_install 2>&1  | tee benchbot.log
```

ernestoMM94, 2021-04-19:
Yes your right it was a internet connection issue. I was able to solve it

closed 2021-04-19 by btalb