qcr/benchbot #23: Ground Truth Mini Room

ernestoMM94, 2021-04-18:
I try running the script : benchbot_run --robot carter --env miniroom:1 --task semantic_slam:active:ground_truth

But don't get in image?

btalb, 2021-04-18:
Can you provide us with logs & more information about what occurs please @ernestoMM94 ? Something like:
```
benchbot_run --robot carter --env miniroom:1 --task semantic_slam:active:ground_truth | tee benchbot.log
```

How are you connecting to this machine? Directly? Remote? SSH? VNC?

Also, it sounds similar to #18. There may be something in that conversation that helps with the issue.

Lastly, as a fallback there is an [OpenGL-based headless mode](https://github.com/qcr/benchbot/wiki/FAQs#i-need-to-get-around-the-vulkan-related-crashes-to-use-benchbot-remotely) which should work in most remote scenarios.

Let us know how you go.

ernestoMM94, 2021-04-19:
Thank you I was able to get it to work

closed 2021-04-19 by btalb