qcr/benchbot #27: Compile the CUDA layers for PointNet++:

ernestoMM94, 2021-04-23:
~/votenet/pointnet2$ python3 setup.py install

giving me an error on installation is there another way I caninstall the cuda layers manually?

btalb, 2021-04-27:
We can offer assistance in fixing the errors if you post the errors here for us @ernestoMM94.

A good practice when reporting issues is to include an error log created by piping the output through `tee`. For example:

```
python3 setup.py install 2>&1 | tee error_log.txt
```

The other option is to run the example in containerised mode, which should work more reliably:

```
benchbot_submit --example semantic_slam_attempt --example-containerised
```

The downside is that the CUDA layers have to be compiled everytime you run the example due to the way Docker handles GPU integration during the build step.

ernestoMM94, 2021-04-29:
[image]

ernestoMM94, 2021-04-29:
I got this error when using --> python3 setup.py install 2>&1 | tee error_log.txt

btalb, 2021-04-29:
You haven't changed into the `pointnet2` directory there @ernestoMM94.

The full instructions for getting the `semantic_slam_attempt` example working with VoteNet are available [in the example](https://github.com/benchbot-addons/examples_ssu/blob/4a74f4cf83ec6c02b494961c6e1932da75d51f67/examples/.semantic_slam_attempt/semantic_slam_attempt#L45):

> Please install votenet & make sure it is available at the above location. Full
> install instructions can be found on the 'votenet' Github page:
>     https://github.com/facebookresearch/votenet
> General install instructions:
>     1. Install pytorch (>=1.5.0 currently fails to build CUDA layers):
>         pip install 'torch<1.5.0' torchvision
>     2. Install tensorflow:
>         pip install tensorflow
>     3. Compile CUDA layers for PointNet++:
>         cd pointnet2 && python setup.py install
>     4. Install votenet dependencies:
>         pip install matplotlib opencv-python plyfile \\
>                 'trimesh>=2.35.39,<2.35.40' 'networkx>=2.2,<2.3'
>     5. Download pre-trained model & extract to '../votenet/demo_files':
>         wget "https://drive.google.com/uc?export=download&id=1oem0w5y5pjo2whBhAqTtuaYuyBu1OG8l" \\
>                 -O demo_files.zip && unzip demo_files.zip && rm demo_files.zip

closed 2021-05-09 by btalb