qcr/benchbot #28: Error installing "BenchBot evaluation"

hellovuong, 2021-04-26:
Hello, Thank you for this exciting challenge. I encountered an issue when It was trying to install `Benchbot evaluation`, the error is `requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://pypi.org/simple/benchbot-addons/`. FUll log is shown in the screenshot. Hope that will hear from you soon
[image]

btalb, 2021-04-27:
Thanks for getting involved in the challenge @hellovuong!

Interesting issue. It seems to be trying to install `benchbot-addons` from PyPI rather than using the locally installed version. Can we check a few things related to your Python installation:
1. Are you using an environment management tool (i.e. Python tools like `venv`, `virtualenv`, etc.; or Conda)? Can you show the output of:
   ```
   which pip3
   ```
2. Can I see what BenchBot packages pip believes are currently installed:
   ```
   pip list | grep benchbot
   ```
   For example, this is what the command shows on our test machine:
   ```
   ben@pc:~$ pip list | grep benchbot
   benchbot-addons         2.1.0                /home/ben/repos/benchbot/addons
   benchbot-api            2.1.0                /home/ben/repos/benchbot/api   
   benchbot-eval           2.1.0                /home/ben/repos/benchbot/eval 
   ```

hellovuong, 2021-04-27:
Thank you for the answer!
The output of `which pip3` is

> /usr/bin/pip3

The output of `pip list | grep benchbot` is

> benchbot-addons (2.1.1)

The screenshot may also help you

[image]

btalb, 2021-04-27:
Hmm that's odd. I would expect `benchbot_api` to at least be on that list based on the first screenshot you supplied.

Let's get a better idea of your system state. Check what `python3` can find and where by running the following:
```
ben@pc:~$ python3 -c 'import benchbot_addons; print(benchbot_addons.__file__); import benchbot_api; print(benchbot_api.__file__); import benchbot_eval; print(benchbot_eval.__file__)'           
/home/ben/repos/benchbot/addons/benchbot_addons/__init__.py
/home/ben/repos/benchbot/api/benchbot_api/__init__.py
/home/ben/repos/benchbot/eval/benchbot_eval/__init__.py
```

Then lets see if there's any odd inconsistency with `python3` paths:
```
ben@pc:~$ which python; which python3; which pip; which pip3;
/home/ben/.python-venv/bin/python
/home/ben/.python-venv/bin/python3
/home/ben/.python-venv/bin/pip
/home/ben/.python-venv/bin/pip3
```

hellovuong, 2021-04-27:
Hi,
The result of first is 
`/home/vuong/Code/rvc/benchbot/addons/benchbot_addons/__init__.py
/home/vuong/Code/rvc/benchbot/api/benchbot_api/__init__.py
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'benchbot_eval'
`

The result of second is
`/usr/bin/python
/usr/bin/python3
/usr/bin/pip
/usr/bin/pip3`
[image]

It seems to me that `benchebot_api` was already installed, somehow `pip list | grep benchbot` didn't list it out.

Also, it is worth pointing out that, during the installing of `benchbot_api` and `benchbot_addons`. It always gets the error `fatal: A branch named 'master' already exists.` when the installer run `git checkout -b master remotes/origin/master`. My solution for that error is that I manually clone `benchbot_api` and `benchbot_addons` to their correct folder in benchbot root folder. Then install it manually by run:
 `git checkout master`
 `python3 -m pip install --upgrade -e .`
After running it, everything seems fine as you see in my first screenshot.
Is it maybe the error come from here?

btalb, 2021-04-29:
Thanks for the information @hellovuong; it led me to the following:

- I was able to reproduce the `fatal: A branch named 'master' already exists` issue on our test machine
- This error was then subtly causing the `benchbot-addons` package to not install properly
- Then `benchbot-eval` would look on PyPI for `benchbot-addons` as it couldn't find it locally

I've published a new version (v2.1.2) that should address these issues.

To start afresh please do the following:

- Uninstall everything using BenchBot:
    ```
    ./install --uninstall
    ```
- Ensure all BenchBot Python packages no longer exist; this should now be empty:
    ```
    pip list | grep benchbot
    ```
- Re-run the BenchBot `install` script:
    ```
    ./install
    ```

Let me know how this goes.

hellovuong, 2021-04-29:
Hi @btalb. Thank you so much for the efforts. 
- I uninstall everything, clone `v2.1.2` then re-run the BenchBot `install` script, It still encountered the same problem.
- Then I uninstalled everything again and switch to the virtual environment, re-install it, and everything is installed properly now.
Thank you again for your support. I will close the issue now.
Cheer,

closed 2021-04-29 by hellovuong