qcr/benchbot #29: "cloudstor.aarnet.edu.au" ERROR 503: Service Temporarily Unavailable and Missing simulated environments

hellovuong, 2021-04-29:
Hi again,
At the `PART 5: INSTALLING BENCHBOT ADD-ONS`
When the installer tried to connect to `cloudstor.aarnet.edu.au`, it return back with the `ERROR 503: Service Temporarily Unavailable.`
Full log:

```Configuring locations for local add-on content:

Created '/home/vuong/Code/rvc/benchbot/addons/benchbot_addons/.local/my_addons/{batches,environments,evaluation_methods,examples,formats,ground_truths,robots,tasks}'.

Installing add-ons based on the request string 'benchbot-addons/ssu':

Installing addon 'benchbot-addons/ssu' in '/home/vuong/Code/rvc/benchbot/addons/benchbot_addons':
	Found install path './benchbot-addons/ssu'.
	No action - latest already installed.
Installing addon 'benchbot-addons/tasks_ssu' in '/home/vuong/Code/rvc/benchbot/addons/benchbot_addons':
	Found install path './benchbot-addons/tasks_ssu'.
	No action - latest already installed.
Installing addon 'benchbot-addons/formats_object_map' in '/home/vuong/Code/rvc/benchbot/addons/benchbot_addons':
	Found install path './benchbot-addons/formats_object_map'.
	No action - latest already installed.
Installing addon 'benchbot-addons/envs_isaac_develop' in '/home/vuong/Code/rvc/benchbot/addons/benchbot_addons':
	Found install path './benchbot-addons/envs_isaac_develop'.
	No action - latest already installed.
	Found remote content to install to 'environments': https://cloudstor.aarnet.edu.au/plus/s/ZqgsznfVNyxnbxV/download
	Remote content is new. Fetching ...
--2021-04-29 13:46:57--  https://cloudstor.aarnet.edu.au/plus/s/ZqgsznfVNyxnbxV/download
Resolving cloudstor.aarnet.edu.au (cloudstor.aarnet.edu.au)... 202.158.207.20
Connecting to cloudstor.aarnet.edu.au (cloudstor.aarnet.edu.au)|202.158.207.20|:443... connected.
HTTP request sent, awaiting response... 503 Service Temporarily Unavailable
2021-04-29 13:46:59 ERROR 503: Service Temporarily Unavailable.

	Fetching of remote content FAILED!!!
Installing addon 'benchbot-addons/robots_isaac' in '/home/vuong/Code/rvc/benchbot/addons/benchbot_addons':
	Found install path './benchbot-addons/robots_isaac'.
	No action - latest already installed.
Installing addon 'benchbot-addons/envs_isaac_challenge' in '/home/vuong/Code/rvc/benchbot/addons/benchbot_addons':
	Found install path './benchbot-addons/envs_isaac_challenge'.
	No action - latest already installed.
	Found remote content to install to 'environments': https://cloudstor.aarnet.edu.au/plus/s/aSjCoQxLjVNUuwR/download
	Remote content is new. Fetching ...
--2021-04-29 13:47:00--  https://cloudstor.aarnet.edu.au/plus/s/aSjCoQxLjVNUuwR/download
Resolving cloudstor.aarnet.edu.au (cloudstor.aarnet.edu.au)... 202.158.207.20
Connecting to cloudstor.aarnet.edu.au (cloudstor.aarnet.edu.au)|202.158.207.20|:443... connected.
HTTP request sent, awaiting response... 503 Service Temporarily Unavailable
2021-04-29 13:47:02 ERROR 503: Service Temporarily Unavailable.

	Fetching of remote content FAILED!!!
```
I think that issue led to there is no `simulated environment` on my computer. 
`benchbot_run --list-environments`
```
The following environments are supported in your BenchBot installation:
    
	NONE!

```
`benchbot_run --list-robots`
```
The following robot targets are supported in your BenchBot installation:
    
	carter
	carter_remote

```
What should I do in this case?
Thank you for your time and waiting for your answers!

hellovuong, 2021-04-29:
It is turn out that the cloud is under maintenance when I was trying to download zip files. It is fine now.

closed 2021-04-29 by hellovuong

btalb, 2021-04-29:
Sorry @hellovuong. CloudStor, the service we use was down yesterday as you noticed. 

For future reference, you can check the status of CloudStor here: https://status.aarnet.edu.au/