qcr/benchbot #31: BUG“Could not find a BenchBot supervisor”

rex131, 2021-07-11:
I found something wrong when I was trying to run the example of  "benchbot_submit --example hello_passive". i have finished the second step and I have gatten the pic1. but when i went to third step, it told me "Could not find a Benchbot supervisor"(pic2). how can i solve this problem? it has confused me for a long time. here are the details:

   File ".hello_passive/hello_passive", line 11, in <module>
    BenchBot(agent=GuidedAgent()).run()
  File "/home/turtlepc-04/.local/lib/python3.6/site-packages/benchbot_api/benchbot.py", line 84, in __init__
    self.start()
  File "/home/turtlepc-04/.local/lib/python3.6/site-packages/benchbot_api/benchbot.py", line 368, in start
    "Are you sure it is available?" % self.supervisor_address)
requests.exceptions.ConnectionError: Could not find a BenchBot supervisor @ 'http://benchbot_supervisor:10000/'. Are you sure it is available?
[image]
[image]

btalb, 2021-07-11:
Thanks for reporting @rex131 .

I'll need a bit more information to understand what is happening on your system.

Can you post the output of the active `benchbot_run` command:

```
benchbot_run --robot carter --env miniroom:1 --task semantic_slam:passive:ground_truth 2&>1 | tee benchbot_run.log
```

And the output of your `benchbot_submit` command:

```
benchbot_submit --example hello_passive 2&>1 | tee benchbot_submit.log
```

And lastly some local network information (while `benchbot_run` is running):
```
ip a
```
```
curl benchbot_supervisor:10000
```

This will help me get an understanding of where the issue could be coming from.

rex131, 2021-07-12:
the first one is just a picture(pic1)
[image]

the scond "benchbot_submit" command
nothing happened
turtlepc-04@TurtlePC-04:~$ benchbot_submit --example hello_passive 2&>1 | tee benchbot_submit.log
turtlepc-04@TurtlePC-04:~$ 

the third " ip a":

ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp7s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 54:05:db:b6:85:86 brd ff:ff:ff:ff:ff:ff
3: wlp0s20f3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 68:54:5a:76:6e:4d brd ff:ff:ff:ff:ff:ff
    inet 10.21.49.65/17 brd 10.21.127.255 scope global dynamic noprefixroute wlp0s20f3
       valid_lft 42532sec preferred_lft 42532sec
    inet6 2001:da8:201d:1109::37d/128 scope global dynamic noprefixroute 
       valid_lft 42533sec preferred_lft 26333sec
    inet6 fe80::6915:e63:429e:8f12/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
4: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 02:42:dc:1c:30:24 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
    inet6 fe80::42:dcff:fe1c:3024/64 scope link 
       valid_lft forever preferred_lft forever
116: br-47bf8759793e: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default 
    link/ether 02:42:d7:3e:0c:90 brd ff:ff:ff:ff:ff:ff
    inet 172.20.0.254/24 brd 172.20.0.255 scope global br-47bf8759793e
       valid_lft forever preferred_lft forever
    inet6 fe80::42:d7ff:fe3e:c90/64 scope link 
       valid_lft forever preferred_lft forever
118: veth7ca0a46@if117: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-47bf8759793e state UP group default 
    link/ether 7e:8b:0c:33:e6:2a brd ff:ff:ff:ff:ff:ff link-netnsid 0
    inet6 fe80::7c8b:cff:fe33:e62a/64 scope link 
       valid_lft forever preferred_lft forever
120: veth9f3bd4a@if119: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-47bf8759793e state UP group default 
    link/ether 5a:87:98:c3:f5:b5 brd ff:ff:ff:ff:ff:ff link-netnsid 1
    inet6 fe80::5887:98ff:fec3:f5b5/64 scope link 
       valid_lft forever preferred_lft forever
122: vethc6b482f@if121: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-47bf8759793e state UP group default 
    link/ether ca:e9:bb:0d:fc:99 brd ff:ff:ff:ff:ff:ff link-netnsid 2
    inet6 fe80::c8e9:bbff:fe0d:fc99/64 scope link 
       valid_lft forever preferred_lft forever
124: veth63187a8@if123: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-47bf8759793e state UP group default 
    link/ether 1e:58:5e:e3:d4:18 brd ff:ff:ff:ff:ff:ff link-netnsid 3
    inet6 fe80::1c58:5eff:fee3:d418/64 scope link 
       valid_lft forever preferred_lft forever

the last "curl benchbot_supervisor:10000"

turtlepc-04@TurtlePC-04:~$ curl benchbot_supervisor:10000
curl: (5) Could not resolve proxy: http

btalb, 2021-07-12:
Thanks for the information @rex131 .

The last command is telling me that the issues are being caused by the local network configuration on your machine. 

Are you running a proxy? If yes, what limitations does it have? It appears to be blocking your local http traffic which will stop BenchBot's process intercommunication.

rex131, 2021-07-16:
i have successfully solved this bug by turning off the vpn. thank you for your time

closed 2021-07-17 by btalb