qcr/benchbot #34: question about data saving

rex131, 2021-07-31:
hi, I am trying to save the data of "image_rgb", "image_ depth", "laser". but i have tried many ways, it doesn't work. what's more, the terminal didn't show anything is wrong.

david2611, 2021-08-02:
Hi @rex131. Glad you are trying to use BenchBot for your data collecting needs 🙂.

When you say "tried many ways" that unfortunately doesn't give me much to go on to provide direct assistance so apologies if the following does not apply to your situation.

I will just let you know how I would be collecting data if I was in your position and you can correct me on your needs from there.

1. Install the data_collect add-on for BenchBot `benchbot_install -a benchbot-addons/data_collect`
2. Run a `data_collect` task in your desired environment. For example: `benchbot_run -r carter -e miniroom:1 -t data_collect:passive:ground_truth`
3. Submit a data_collect example solution `benchbot_submit -E data_collect_passive -a '-s <save_root>'` which will save the data after each step in the given root folder. **Note** this will store more information than you currently request. To trim down what is saved you will need to go into the example code and change things as desired.
4. Check that the data has been saved as appropriate and use as needed.

I hope that this sorts out your problem. Please let me know if it does and if not, please provide more specifics about what it is you are currently doing.

If there is a particular terminal output that you would be expecting from what you have already been doing, please let me know what it is you have been doing and what you would have expected from the terminal.

btalb, 2022-03-31:
Closing due to inactivity.