qcr/benchbot #35: How can I make 3D map using json result files?

kjwoo31, 2021-09-14:
I want to make 3D semantic map just like the below images. Can you explan me how I can make them? Is there any addons or code to visualize them?

I've done the environment setting and evaluation, but got stuck in the part to visualize the result. I want to see what's the problem in 3D map view. Thanks for a great software. :+1: :D

[image]
[image]

btalb, 2021-09-17:
Thanks for the feedback @kjwoo31 ! :smiley: 

We haven't seen the second image before, where did you get that one from? It looks nice.

Unfortunately we don't have any packaged tools for visualising object maps with BenchBot. The first image was produced by one of our researchers with MATLAB when they were developing the Semantic SLAM Attempt example.

We agree it would be a nice addition, but not sure when we'll be able to get around to it. I've added it as a future enhancement for now.

kjwoo31, 2021-09-17:
Thank you for responding to my question.

I found the image from issue #14, and this paper. (Gopinathan, M., Truong, G., & Abu-Khalaf, J. (2021). Indoor Semantic Scene Understanding using Multi-modality Fusion. arXiv preprint arXiv:2108.07616.) 

I'll be looking forward to use the visualization tool in benchbot. 😄

rpartsey, 2022-07-24:
Hi @kjwoo31, 
If you are still interested in 3D semantic map visualisation tools, you may check the examples implemented by me [here](https://github.com/rpartsey/ssu_map3d_visualization) :wink:. 

Example visualisations:
| Ground-truth  | Ground-truth vs built using [VoteNet](https://github.com/facebookresearch/votenet) |
| ------------- | ------------- |
| [image] | [image]  |

Btw, @btalb, if you think that such tools are useful and should be a part of BenchBot Software Stack, I'd be happy to submit a PR :slightly_smiling_face:.

btalb, 2022-07-27:
Thanks @rpartsey .

These tools look great. The code in your `vis_tools/` would be a nice addition to the [Benchbot API](https://github.com/qcr/benchbot_api) if you're happy to contribute. 

Maybe as a new `extras.py` file in here: https://github.com/qcr/benchbot_api/tree/master/benchbot_api?

Happy to look at a PR and try to merge it in.

closed 2023-07-20 by kjwoo31