qcr/benchbot #36: Isaac 2019.2 not supported by Nvidia anymore

vincentcartillier, 2021-11-16:
Hi, 
It seems the ISAAC 2019.2 SDK is not supported anymore by Nvidia.
There is no "Archive" folder on their download page.
(cf https://forums.developer.nvidia.com/t/downloading-isaac-sdk-2020-2/195227)

Is there any workaround? 
Can Benchbot compile with the new ISAAC 2021.1 SDK?

btalb, 2021-11-18:
Thanks for bringing this to our attention @vincentcartillier.

Getting BenchBot to work with the latest version of ISAAC will require some engineering work that we hadn't allocated time for. We didn't expect NVIDIA to proceed this way. We will need to dig in to the latest versions of ISAAC to understand: 1) how much work that will involve, 2) if we have the cycles available to get it done.

Sorry for the non-answer at the moment, but we will get back to you once we know more.

vincentcartillier, 2021-11-18:
Thanks for your reply.

I have started to look into compiling benchbot with Isaac-2021. 

**Environment**: I am running a VM from GCP with Ubuntu18 and a Tesla P4.

Following are some notes to what I have tried and how far I reached..
**Notes**:

- I have started by adding the isaac-2021.1 tar.gz file under `isaac/` and modify the checksum condition in the installation script (`bin/benchbot_install:L379`):
`chk_isaac_check='find $PATH_ISAAC_SRCS -type f -name "isaac*"'`
 
- There seem to be discrepancies with the `.isaac_patches/` from the `benchbot_simulator`. I am commenting out this part for the moment... (`docker/backend.Dockerfile:59`)
 `#.isaac_patches/apply_patches`

- When building ROS, Bazel seems to have a hard time finding packages:
[image]

I will continue looking and I'll post updates here if I manage to get it working. In the meantime any suggestions are welcome!

PS: The installation script is very well made ! :)

btalb, 2021-11-21:
Thanks for digging in for us @vincentcartillier .

Here's my notes on what would need work for this to work with the latest Isaac:

- The checks you've identified in the install scripts would definitely have to be updated
- Post 2019.2 ISAAC dropped support for the Unreal Engine version of ISAAC Sim which is what we use for our environments. This is the big step where most of the uncertainty lies. We would have to:
    - port the existing environments ([here](https://github.com/benchbot-addons/envs_isaac_develop) and [here](https://github.com/benchbot-addons/envs_isaac_challenge)) to [Omniverse](https://developer.nvidia.com/nvidia-omniverse-platform), which looks achievable based on the Omniverse documentation
    - port our short list of BenchBot specific additions (only one I can think of is we use Unreal to tell us if the robot has collided)
    - probably mark [`benchbot_simulator`](https://github.com/qcr/benchbot_simulator) deprecated and rename it to `benchbot_simulator_unreal` or something similar
    - make new simulator bindings for Omniverse environments to ROS (something like `benchbot_simulator_omniverse`), which should be much easier these days as the ISAAC SDK sits in the middle and has become much more mature
    - add [support for a new simulator target to the ecosystem](https://github.com/qcr/benchbot_addons#environment-add-ons) `sim_omniverse`
    - implement support in both the `benchbot_install` and `benchbot_run` scripts

I provide this exhaustive list just for context. It's certainly not a "these things are why it's too hard, so we won't do it" situation. Because they are all doable, it just requires time. It's more a "this is the roadmap of tasks we would need to complete to add this support in the future".

Thanks for the nice comments about the install script, and for digging into BenchBot for us; it's the community engagement that makes projects like this worthwhile!

btalb, 2022-02-20:
Please see the latest release @vincentcartillier .

We now have official support for the latest Isaac Sim (2021.2), powered by Omniverse. Apologies about the delay.

We're aware there may be some bugs with a migration of this size, so please let us know if you encounter any.

closed 2022-02-20 by btalb

floodsung, 2022-03-22:
> Hi, It seems the ISAAC 2019.2 SDK is not supported anymore by Nvidia. There is no "Archive" folder on their download page. (cf https://forums.developer.nvidia.com/t/downloading-isaac-sdk-2020-2/195227)
> 
> Is there any workaround? Can Benchbot compile with the new ISAAC 2021.1 SDK?

I find this archive https://github.com/NvPhysX/UnrealEngine/tree/IsaacSim_1.2

btalb, 2022-03-25:
That will get Unreal Engine at the version that was used in Isaac Sim 1.2 @floodsung , but doesn't include the extra file which was required to pull in the appropriate dependencies unfortunately.

That was contained in the download page which is unfortunately gone.