qcr/benchbot #38: Simulator crashs after fresh install

gmuraleekrishna, 2022-03-01:
I am trying to setup your simulator on my machine. After a successful install, I ran `benchbot_run --robot carter_omni --env miniroom:1 --task semantic_slam:active:ground_truth` but the command crashes moments after waiting for robot controller. The log is attached 
[here](https://github.com/qcr/benchbot/files/8159040/fault_log.txt)

Specs:
OS: Ubuntu 20.04 LTS
GPU: NVIDIA RTX A6000
CUDA: v11.6

gmuraleekrishna, 2022-03-08:
@btalb Any updates on this please?

btalb, 2022-03-08:
Sorry about the delay @gmuraleekrishna .

We've been heavily impacted by the QLD floods and unable to access our machines at QUT.

Yesterday I was able to get someone to reproduce what you're experiencing, so we'll look into resolving and releasing an update. 

Can I ask how you're using your machine? Are you sitting at your machine and running BenchBot directly on it? SSHing in? Or remote desktop?

gmuraleekrishna, 2022-03-09:
@btalb Sorry to hear about the QLD floods. Hope everyone is safe.

It is installed on a remote machine. The issue exist for both ssh and rdp sessions.

btalb, 2022-03-15:
@gmuraleekrishna I believe we've found the cause of your issue.

Previously we could disable the `xhost +local:root` permissions immediately after starting `benchbot_run`. The new Omniverse-based simulator seems to require these permissions enabled the entire time BenchBot is running.

You can try one of these two options if you want to get up and running immediately:
- Comment out this line in `benchbot_run`: https://github.com/qcr/benchbot/blob/f473fac5741b983a561cc6aa8d62dc796dc15e06/bin/benchbot_run#L363 so permissions are always open
- Checkout the develop branch (`git checkout develop`) to get the fix in b57a170

Otherwise, we should be releasing an update in the next few days.

btalb, 2022-03-25:
Fix is now live on master. Re-running `benchbot_install` will update you to the latest version.

Note: if using a remote machine, you still must ensure the rendering is done on that machine not your client (i.e. set `DISPLAY` env variable, or use correctly configured remote desktop, etc.).

We may look at making this easier to configure in a future update.

closed 2022-03-25 by btalb

ShuangHigh, 2022-03-29:
I ran into the same problem using the latest version of the code. This bug does not seem to have been fixed. Cloud you please help me solve it?

ShuangHigh, 2022-03-29:
> I ran into the same problem using the latest version of the code. This bug does not seem to have been fixed. Cloud you please help me solve it?

ShuangHigh, 2022-03-29:
> the command crashes moments after waiting for robot controller

btalb, 2022-03-29:
Sure @ShuangHigh; could you please post some details to help me understand the problem:

- log from the `benchbot_run` command:
   ```
   benchbot_run ...<rest_of_your_command> 2&>1 | tee benchbot_run.log
   ```
- details about how you're accessing your machine: are you sitting at the machine using it directly? SSHing in? Using remote desktop?
- log from a `benchbot_install` (it won't reinstall what's already installed):
   ```
   benchbot_install 2&>1 | tee benchbot_install.log
   ```

ShuangHigh, 2022-04-02:
> Sure @ShuangHigh; could you please post some details to help me understand the problem:
>
> * log from the `benchbot_run` command:
>   ```
>   benchbot_run ...<rest_of_your_command> 2&>1 | tee [...]
@btalb Thanks for your help!
When I ran  “benchbot_run --robot carter_omni --env miniroom:2 --task semantic_slam:active:ground_truth”，  but the command crashes moments after waiting for robot controller. The log appears as follows：
[log.txt](https://github.com/qcr/benchbot/files/8401995/log.txt)

OS： Ubuntu18.04
GPU:  Quadro P2000
CUDA Version: 11.6

And I sitting at the machine using it directly.
Thanks for your help!