qcr/benchbot #44: Optimise controller gains when moving to poses

btalb, 2022-03-25:
We have gone with conservative controller gains for now, but they could be more aggressive.

Well-tuned gains would mean the robot would move between poses quicker, while still not suffering from overshoot or instabilities.

ivbelkin, 2022-03-28:
Hello! It looks like I face a problem related to this issue. 

When I try to run hello_passive example on the office:1 environment robot stucks (or moves extremely slow) after ~7 route pose. 

I start the simulation with the command:
benchbot_run --robot carter_omni --env office:1 --task semantic_slam:passive:ground_truth
and example:
benchbot_submit --example hello_passive

btalb, 2022-03-28:
Thanks for reporting @ivbelkin .

Did it finish the run (but extremely slowly), or get stuck and not continue from pose 7?

We will be improving the controller gains within the next couple of days which should make a big difference to speed. 

Sorry about this. We had the opposite problem before with instabilities causing crashes, but have gone back the other way a little too far.

david2611, 2022-03-28:
@ivbelkin thanks for the info. I have just checked your problem and can confirm that it is an unrelated issue but one that will be easily fixed in our next update.

The issue is due to a starting pose that is not correct under the omniverse system which throws off all subsequent trajectory points.

If you want to run office:1 before the next update just go to 
`/path/to/benchbott/addons/benchbot_addons/benchbot-addons/envs_bear_challenge_sim_omni/environments/office_1.yaml`
and update the start pose on line 13 to 
`start_pose: [0.781, 0.0, 0.0, 0.625, -1.0, -9.5, 0.3]`

I can warn you that the full trajectory will take a while due to the original issue posted here. Mine took 36 mins. We will keep working to improve this but hopefully this quick fix can sort out your issues with office:1 :+1: 

If it doesn't, please open a new issue and we will get onto it

ivbelkin, 2022-03-28:
> @ivbelkin thanks for the info. I have just checked your problem and can confirm that it is an unrelated issue but one that will be easily fixed in our next update.
>
> The issue is due to a starting [...]

Thank you, it fixes the problem for me too

btalb, 2022-04-01:
These values should be better with the release of [v2.2.2](https://github.com/qcr/benchbot/releases/tag/v2.2.2), but we still aren't quite satisfied.

We will release an update in a few days which should result in quicker traversal times through environments.