qcr/benchbot #52: Installation failure at PPA check

jingnanshi, 2022-04-11:
Hi,

Thanks for the project! I am encountering some small issues when installing benchbot. Specifically, the script thinks I'm using an invalid PPA when I am in fact using the graphics-drivers PPA. The error message is the following:
```
The installed NVIDIA driver was detected to have come from somewhere other
  than the NVIDIA / Ubuntu graphics-drivers PPAs. The driver was sourced from:
    'http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu/pool/main/n/nvidia-graphics-drivers-510/nvidia-driver-510_510.60.02-0ubuntu0.20.04.1_amd64.deb'

  Our Docker container only supports official drivers, & GPUs only work in 
  containers if the driver version exactly matches the version of the host.
  Please use a driver from either of the following locations (the NVIDIA PPA is
  preferred):
    http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64
    http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu
```
My system is Ubuntu 20.04. Any help will be appreciated. Thanks!

btalb, 2022-04-14:
Thanks for reporting @jingnanshi .

This is an odd one. I haven't been able to reproduce it locally. This is the check we use which seems to return the expected result for your PPA location:

```
user@pc:~$ [[ "http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu/pool/main/n/nvidia-graphics-drivers-510/nvidia-driver-510_510.60.02-0ubuntu0.20.04.1_amd64.deb" =~ .*(developer.download.nvidia.com/|ppa.launchpad.net/graphics-drivers/ppa/ubuntu).* ]]; echo $?
0
```

Can you please the results of the following to help me get to the bottom of this:
```
apt list --installed | grep nvidia-driver-
```
```
apt download --print-uris nvidia-driver-*
```

If you're looking for a short-term fix that should get you into a working state I'd recommend upgrading to the drivers provided on NVIDIA's official PPAs:
https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=20.04&target_type=deb_network

The official drivers from there generally offer the best experience on Linux.

Let me know how you go.

jingnanshi, 2022-04-17:
Hi @btalb, thanks for the feedback.

Here are the results of running the two commands:
The first command returns:
``` nvidia-driver-510/focal,now 510.60.02-0ubuntu0.20.04.1 amd64 [installed] ```
The second command returns:
```
'http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu/pool/main/n/nvidia-graphics-drivers-390/nvidia-driver-390_390.147-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-390_390.147-0ubuntu0.20.04.1_amd64.deb 393348 SHA256:2aa31a0d3df157fb22329ef8c58b3aafcf082469b38be25effd3d6ab8f3c0f3c
'http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu/pool/main/n/nvidia-graphics-drivers-510/nvidia-driver-495_510.60.02-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-495_510.60.02-0ubuntu0.20.04.1_amd64.deb 6388 SHA256:0e9b14c883b97880ec2f90b3bad396f6c551b177c874928fc2eeca343eb0ff2d
'http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu/pool/main/n/nvidia-graphics-drivers-510/nvidia-driver-510_510.60.02-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-510_510.60.02-0ubuntu0.20.04.1_amd64.deb 454524 SHA256:089ad6b01578e99312d3cb769e2c0165e2b96e23a1811a9a6835e0ed4d883868
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-430/nvidia-driver-418_430.50-0ubuntu3_amd64.deb' nvidia-driver-418_430.50-0ubuntu3_amd64.deb 6932 SHA256:83a49f1382c3a61d20a2bb1080b965b82a45affd54d8fcea3a9ce200092cfe9d
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-440/nvidia-driver-430_440.100-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-430_440.100-0ubuntu0.20.04.1_amd64.deb 7340 SHA512:9103e878d85798b9472b7d9ef644089c1ccfe1eeb4e759a67d8c2f7ad1fe28646bfc9ed42feabcc98d71a45c63822b7385bebdb62964ab55562b7a04638ddc0d
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-455/nvidia-driver-435_455.45.01-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-435_455.45.01-0ubuntu0.20.04.1_amd64.deb 6004 SHA512:a39c9100463423ca374f3b7c206c58e3dcccf47f3014cb882590bfe1c2a7c98cf8c71fd9e90c8274d5b55bd7084bd6dfd31ace15e398ea56b3c9699d0819198e
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-450/nvidia-driver-440_450.119.03-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-440_450.119.03-0ubuntu0.20.04.1_amd64.deb 6804 SHA512:2e227865f535aabd6526f2964932815f5fd2309c8f97fcc7d21b8729e4509f96a342015750614988ec8ab871b44488ffcaa68efbca84c42edd794fc126a3c659
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-460/nvidia-driver-450_460.91.03-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-450_460.91.03-0ubuntu0.20.04.1_amd64.deb 7624 SHA512:3d438ca5b4db9b72ac6f2fa36facbe5e614c8a11175f0b8e507a8074c71e674f7b4c58b51f564b9be1d8120457fe975cd29500a9659071bb39e69ed4c9bdaa55
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-450-server/nvidia-driver-450-server_450.172.01-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-450-server_450.172.01-0ubuntu0.20.04.1_amd64.deb 431440 SHA512:e45e45b2f6cd6141439e336af372c657759a274cf3409bbfdcc393de12245552e4c64db7f8ce35e04dda1c16d98f32a8fe5a1f7f79e01b8aa8b51d8d4263c24d
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-460/nvidia-driver-455_460.91.03-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-455_460.91.03-0ubuntu0.20.04.1_amd64.deb 7624 SHA512:c558028ac61e7d8b3ade61b8f144ce981f31674af7f7f589c8b7c74308ec7ad90e353d20c73a11c6909496d963b906f6ed4c7669f9e765b2935376a354a6a7db
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-470/nvidia-driver-460_470.103.01-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-460_470.103.01-0ubuntu0.20.04.1_amd64.deb 7156 SHA512:446b8d609d53ab9aec48a004f6743061896b16c29c0e245c0796b86a25083edd363c5285ceb6c370153004fdaf9643be8e061ec8003a32b35e9c461104791fec
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-470/nvidia-driver-465_470.103.01-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-465_470.103.01-0ubuntu0.20.04.1_amd64.deb 7156 SHA512:011a9917e4267e11ec0646a10b668d4030f3222c094447afa39b9c3ff486a941d1c9581dc3d24079964a84a4fc106c7373ebe9f87c7e4913c1c8937f361b4615
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-470/nvidia-driver-470_470.103.01-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-470_470.103.01-0ubuntu0.20.04.1_amd64.deb 446128 SHA512:d880852af0d91fabb8fa3432dfd69537ee654efa3f721548cf30248f5c7bbcd30de3187803eb6f0224fc3bda4ca635b7f09707e05abce6403009e688a8370b59
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-470-server/nvidia-driver-470-server_470.103.01-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-470-server_470.103.01-0ubuntu0.20.04.1_amd64.deb 445316 SHA512:dcae3768cd417fc11f803d94210eea30a6b5e45316bcc49ee8b6cc4652845b7cb4c993be105d9287f4eedaf390c4d2839483c46cf49f6e594488ce356f6680eb
'http://us.archive.ubuntu.com/ubuntu/pool/restricted/n/nvidia-graphics-drivers-510-server/nvidia-driver-510-server_510.47.03-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-510-server_510.47.03-0ubuntu0.20.04.1_amd64.deb 453296 SHA512:92babd404809656bca462813a54a9c85b0fb9a9b5ad3d7ef6136c51af5e35432c3dc9a4c0ace95b3f08bcb80cc65d9cef79adfec3341d1dae0da90aa2f6b010c
'http://us.archive.ubuntu.com/ubuntu/pool/multiverse/n/nvidia-graphics-drivers-418-server/nvidia-driver-418-server_418.226.00-0ubuntu0.20.04.2_amd64.deb' nvidia-driver-418-server_418.226.00-0ubuntu0.20.04.2_amd64.deb 404220 SHA512:c849a514a3172ea6e43b7f99c35c77300ed51842f42f824e7c40a7198dec2e470488dbb43a8d41746cdd2ef6eff745f4ef002dae2001ac0ff59b3e443762d2b1
[... cut ...] 261b8bb817ed799f6e5bede2e72e7ef6df484b45133b302dc00c76d5e8
'http://us.archive.ubuntu.com/ubuntu/pool/multiverse/n/nvidia-graphics-drivers-470-server/nvidia-driver-460-server_470.103.01-0ubuntu0.20.04.1_amd64.deb' nvidia-driver-460-server_470.103.01-0ubuntu0.20.04.1_amd64.deb 6048 SHA512:49efe3d80c3a2a218ed12c3952c31ccbcebf8662ac8dcb64eb75a4066762d9f9ff8f874db6ad37cec1ac7b687460c75cb4d17aca93f106e9bb08729929f5d0b7
 ```

I will try with the nvidia's official PPA's later and report back any issues.

btalb, 2022-04-26:
@jingnanshi any update on how this went?

closed 2022-05-20 by btalb