qcr/benchbot #63: A Startup problem

TongChiYip, 2022-05-18:
I've tried my best to install benchbot software stack via your script `./install`, all the docker images have been built well (at least I think so), and when I execute this command, no error occurred:
`u@pc:~$ benchbot_run --robot carter_omni --env miniroom:1 --task semantic_slam:active:ground_truth`
Then `benchbot_run` script call out a window like below which seems that the renderer didn't go well:
[image]
I think it might be relative to the GPU I'm using which is a GeForce 1080Ti. When I was building the simulator image, at pre-baking step, `gpu.foundation.plugin` showed out lots of errors as below:
[image]
Is it impossible that a GeForce 1080 to run this stuff or just I've made some mistakes unconsiously? Looking forward to you reply. Thanks a lot.

btalb, 2022-05-20:
Unfortunately, the [Isaac Sim system requirements](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/requirements.html) section does seem to suggest you need a 2070 plus. I believe it's for the RTX capabilities, which only came in the 20 series cards.

I can't say 100% it's why you're having the issue @TongChiYip, but that would be my guess at this stage.

TongChiYip, 2022-05-23:
> > Unfortunately, the [Isaac Sim system requirements](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/requirements.html) section does seem to suggest you need a 2070 plus. I believe it's for the RTX capabilities, which only came in the 20 series cards.
> > I can't say 100% it's why you're having the issue @TongChiYip, but that would be my guess at this stage.
> 
> Thanks for your reply a lot. I,m going to use an RTX 3090 environment to check out whether this fault will occur again. If so, I will let you know.

I've changed to an RTX 3090 environment, and the simulator works well. It seems that problem was exactly because of inappropriate devices. Thanks a lot. Meanwhile, I found that when I try to execute commands like:
`u@pc:~$ benchbot_submit --containerised --example hello_active`
to run the examples, another problem showed up hinting that I'm requesting a non-existing Dockerfile. I think there might be something wrong under the directory `benchbot/addons/` but I don't exactly know how to fix it. Would you mind help me out? Thanks!

btalb, 2022-05-24:
Good to hear you've progressed @TongChiYip .

I've had reports of the examples not working in `--containerised` mode and will look into it (see issue #65). For now though, you should be able to proceed using the examples in `--native` mode (e.g. `benchbot_submit --example hello_active`).

I'll mark this issue as resolved for now, but feel free to post further comments if you have further issues.