qcr/benchbot #65: Examples don't work in containerised mode

btalb, 2022-05-24:
Appears as though examples don't work in `--containerised` mode.

Will investigate and post a fix.

For now, `--native` mode should work (or running without the flag altogether).

david2611, 2022-05-24:
This appears to be an issue in documentation. The following from the main benchbot page does not work
`benchbot_submit --containerised --example hello_active`
however this neglects updated functionality for containerised mode and so should be updated to the following
`benchbot_submit --example-containerised hello_active` 

However, when testing this I did find that the `--containerised` option does not seem to work for examples unless being run from within the folder of the Dockerfile.

Example:
`u@pc: benchbot/addons/benchbot_addons/benchbot-addons/examples_base/examples/.hello_benchbot$ benchbot_submit -c ./Dockerfile` 

works fine but

`u@pc: benchbot/addons/benchbot_addons/benchbot-addons/examples_base/examples$ benchbot_submit -c .hello_benchbot/Dockerfile` 

does not

TongChiYip, 2022-05-24:
> 

> This appears to be an issue in documentation. The following from the main benchbot page does not work `benchbot_submit --containerised --example hello_active` however this neglects updated [...]

Thanks for your testing. I've run command:
`u@pc: benchbot/addons/benchbot_addons/benchbot-addons/examples_base/examples/.hello_benchbot$ benchbot_submit -c ./Dockerfile`
exactly under the folder as above, and shell is able to execute the `docker build` command properly, building a temporate image. But then there shows that "non-network local connections being added to access control list, waiting for Docker network to become available" with no further information, which seems is a Docker-related problem. How can I fix it?

david2611, 2022-05-24:
@TongChiYip is that behaviour happening even when you have a `benchbot_run` command going? 

If not it will hang there, looking for a Docker network until one is created via the `benchbot_run` command.

TongChiYip, 2022-05-24:
@david2611 Thanks for your reply. Command
`u@pc:~$ benchbot_run --robot carter_omni --env miniroom:1 --task semantic_slam:active:ground_truth`
runs well on my system. Does it mean that I have to run a `benchbot_run` command at the same time as I run the `benchbot_submit` command? How should I run that?

david2611, 2022-05-24:
Standard pipeline for operating with benchbot is as follows:
1. Setup the experiment you are running with a `benchbot_run` command in a given terminal. This is where you define what task you are performing, what robot you are using, and the environment you are operating within
2. Submit your solution to the experiment using `benchbot_submit` in a separate terminal while `benchbot_run` is still going. This enables your solution to interact with the environment as you attempt to complete the experiment you are running. 
3. Evaluate your solution with a `benchbot_eval` command to see how well you did after your solution has completed. This will not require `benchbot_run` to be running and must happen after your solution has completed (`benchbot_submit` has ended).

More info can be found on the main [benchbot](https://github.com/qcr/benchbot) page and in our [Tutorial](https://github.com/qcr/benchbot/wiki/Tutorial:-Performing-Semantic-SLAM-with-Votenet).

In your situation what I think you are wanting to do is the following:
1. In one terminal run `u@pc:~$ benchbot_run --robot carter_omni --env miniroom:1 --task semantic_slam:active:ground_truth`
2. In a second terminal run `u@pc: benchbot/addons/benchbot_addons/benchbot-addons/examples_base/examples/.hello_benchbot$ benchbot_submit -c ./Dockerfile`

Now the `hello_benchbot` example is an example that is meant to have the robot drive backwards and forwards again to test everything is working ... but in miniroom it **will** crash in hilarious ways :) so if you want to have an example that shows you benchbot more thoroughly I might recommend running the `hello_active` example instead

closed 2022-05-27 by btalb