qcr/benchbot #68: KeyError when using votenet to do the detections.

TongChiYip, 2022-05-26:
Sorry for bothering you again. I was trying to build my first agent according to your [tutorial](https://github.com/qcr/benchbot/wiki/Tutorial:-Performing-Semantic-SLAM-with-Votenet), a `KeyError` occurred showing that there is no key called 'camera' in the dictionary `observations['poses']`:
`Traceback (most recent call last):
    File "my_semantic_slam/my_semantic_slam.py", line 24, in <module>
        Benchbot(agent=MyAgent()).run()
    File "~/benchbot/api/benchbot_api/benchbot.py", line 326, in run
        scene_fn()
    File "~/benchbot/api/benchbot_api/benchbot.py", line 321, in scene_fn
        action, action_args = self.agent.pick_action(
    File "my_semantic_slam/my_semantic_slam.py", line 12, in pick_action
        results = votenet_detection(self._votenet, observations)
    File "~/benchbot/my_semantic_slam/votenet_benchbot.py", line 206, in votenet_detection
        camera_pose = observations['poses']['camera']
KeyError: 'camera'`
I think it might be caused by some API mistakes, while there is actually **no mention** for a key 'camera' according to the README file in [benchbot_api repository](https://github.com/qcr/benchbot_api#observation-channels).

btalb, 2022-05-27:
That's a good pickup @TongChiYip ; thanks for that.

This is actually a bug in our updated controller code. The pose `'camera'` should exist in that object. 

I'll make a fix and push out a new release later today.

btalb, 2022-05-27:
Above commit should fix this. Let me know if updating BenchBot doesn't resolve the issue.

closed 2022-05-27 by btalb

TongChiYip, 2022-05-27:
Thanks @btalb . Is a complete reinstallation needed? Or we just need to rebuild the Docker images related?

btalb, 2022-05-27:
If you just re-run `benchbot_install` it will go through all the steps, skipping whatever is already installed and reusing cached Docker layers.

TongChiYip, 2022-05-27:
Due to the firewall set by our government, a re-run of `benchbot_install` might be annoying, which makes me installed `benchbot` in a "semi-manual way" previously. If a simple re-building for related Docker images is all what I need, that would save my time.

btalb, 2022-05-27:
Ah I see, no worries.

In that case, yes updating the Docker images should be all you need (mainly Git pulling the repo in `/benchbot/benchbot_robot_controller/` directory, then committing the ).

You may also want to pull the latest API in the `benchbot/api` directory outside of Docker as well.

TongChiYip, 2022-05-27:
So the updates are only related to Git repos [benchbot_api](https://github.com/qcr/benchbot_api) and [benchbot_robot_controller](https://github.com/qcr/benchbot_robot_controller)?

btalb, 2022-05-27:
That's correct.

TongChiYip, 2022-05-27:
I've rebuilt the related `backend` image and `simulator` image. But when I try `benchbot_run`, another error occurred showing that the BenchBot Robot Controller container exited unexpectedly due to file `/benchbot/ros_ws/src/benchbot_robot_controller/nodes/benchbot_robot_controller` is not executable. This error haven't happened in previous versions. Could this be caused by the catkin compilation? Or just my mistakes? I'm not sure.

btalb, 2022-05-30:
That I would believe be an error somewhere in the manual process you've taken @TongChiYip.

That file's existed for a long time, and not been changed in any recent updates. If you're re-building with Catkin manually, you should just have to source `/benchbot/ros_ws/develop/setup.bash`, re-run `catkin_make`, and then Docker commit the results.

Sorry I can't offer anything more concrete, supporting manual building is a bit outside of our scope.