qcr/benchbot #70: Working with remote machine?

GokulEpiphany, 2022-06-02:
Is there a proven way to work with benchbot installed on a remote instance (AWS/azure)? I have tried everything I could think of, including things suggested previously. I have previously worked with Omniverse using websocket live streaming. Tried port forwarding, tightvnc, tigervnc. Install only works headless but fails when running benchbot_run with internal server error during starting of robot controller. Most errors seem to come from glx when installing with headless=False.

GokulEpiphany, 2022-06-02:
If possible, could you please spawn a remote instance with supported GPU and check for compatibility issues by doing a fresh install?

btalb, 2022-06-03:
Sorry to hear about that experience @GokulEpiphany .

At the moment, BenchBot expects there to be hardware-accelerated screen to render to. There would be a bit of work to unwind this assumption on our end.

Something like [VirtualGL](https://virtualgl.org/About/Background) is probably the most user-friendly way to achieve this.

Once a hardware-accelerated display exists, and it is set as the terminal's target with `export DISPLAY=<address_of_display>`, then BenchBot should work as expected. A lot of BenchBot's development work was done from home during COVID lockdowns with this method.

If you're having trouble confirming a hardware-accelerated display is being used, I found tools like `vkinfo`, `glxgears`, and `glxinfo` useful.

GokulEpiphany, 2022-06-03:
For now, working headless with benchbot..thanks for the suggestion.. closing this..

closed 2022-06-03 by GokulEpiphany