qcr/benchbot #72: Action input in Active Dead Reckoning task

antabangun, 2022-06-04:
Hello again!

I am attempting the active actuation with noisy dead-reckoning task (challenge 3 and 6). 
I encounter an issue on my end with the robot seemingly moving randomly when given the action command.

To illustrate this, I am giving the robot constant 45.0 move_angle input, and this is a gif showing what happened on my BenchBot simulator
[https://drive.google.com/file/d/10srrhtgLjS_8Mtp7U3QTT5VzCo6BFsFO/view?usp=sharing](url)
(I apologize for the rather large file size)

This is what happens when I give the robot a constant 1.0 move_distance
[https://drive.google.com/file/d/1wymVz4w0ACH58XegeoaX_-Vgf71imKbK/view?usp=sharing](url)

Sometimes the robot doesn't even move at all, and it becomes almost impossible to move the robot around the scene. 
Best regards.

antabangun, 2022-06-04:
I had no issue moving the robot in Active with ground truth localization mode

btalb, 2022-06-05:
Thanks for reporting @antabangun . Certainly doesn't look like correct behaviour.

We will investigate on our end and see what we find.

david2611, 2023-01-13:
Apologies for the massive delay on fixing this issue.  We found that the error stemmed from multiple bugs in generating noisy odometry that we had to implement quickly when transitioning into Isaac Sim Omniverse.

These have now been addressed with the latest release of our `benchbot_robot_controller` ([v2.5.0](https://github.com/qcr/benchbot_robot_controller/releases/tag/v2.5.0)).

A fresh `benchbot_install` should now properly allow for active dead reckoning with noisy pose data and noisy movement (i.e. movement based on estimated change in pose rather than true change of pose)

closed 2023-01-13 by david2611