qcr/benchbot #77: Ask about the easy way to get started

yizhouzhao, 2022-07-07:
Thanks for your excellent work. I really want to investigate this fantastic project.

However, the hardware and software requirements seem overwhelming (especially when I only have an old Inter CPU with one RTX graphic card).  Is there any easy way to start this project and challenge? I can run Isaac-sim on my system.

Thanks for your help?

yizhouzhao, 2022-07-09:
Careful when I want to start?

My whole system gets crashed!!!!!!!!!!!!!!!!!!!!

Intel I7-8700, Nividia 3090, Driver version 510

david2611, 2022-07-10:
HI @yizhouzhao. 

With regards to your first issue, the hardware requirements we provide are based largely off of the requirements for Omniverse Isaac Sim recommendations.

With regards to the second, sorry to hear your system crashed. Could you please provide more info regarding what went wrong? Such as:

- At what stage the process crashed for you (installation, running an environment, submitting a solution, evaluation)
- What error messages you got as you attempted to utilise BenchBot?
- In what ways has your whole system crashed?

btalb, 2022-08-16:
Closing due to lack of activity. Feel free to re-open if needed.

closed 2022-08-16 by btalb