qcr/benchbot #78: Scene data visualization

rpartsey, 2022-07-08:
Thanks for the amazing work!

Could you, please, suggest what is the best way to visualize the scene and all the ground truth data like objects' bounding boxes, the agent's trajectory, etc.

The pictures in issues https://github.com/qcr/benchbot/issues/46 and https://github.com/qcr/benchbot/issues/69 look very beautiful. Being able to visualize data like that may speed up the research phase. 

Any help/pieces of advice is much appreciated.

david2611, 2022-07-10:
Hi @rpartsey thanks for the feedback! 

You are right in that it would be good to have an in-built tool for enabling visualizations. We don't currently have one as a part of the main BenchBot code base. I have added this as an enhancement for future consideration. 

In the meantime, all the visualizations you have seen in those previous issues has been done through utilising the [rviz](http://wiki.ros.org/rviz) tool within ROS alongside some custom-built tools for interpreting the GT file and turning it into box visualizations.

One of our users has actually provided their methodology for visualization in one of our previous issues which you can find [here](https://github.com/qcr/benchbot/issues/47#issuecomment-1100881140). 

Until we have provided some "official" benchbot tool for visualization it is up to you as to how you want to visualize the process, either just with benchbot observations or raw ROS data. Hope that this is at least useful as a starting point for you.

btalb, 2022-08-16:
Visualisation tools are now live in [BenchBot API v2.4.3](https://github.com/qcr/benchbot_api/releases/tag/v2.4.3).

closed 2022-08-16 by btalb