qcr/benchbot #79: ERROR: Building BenchBot "core" returned a non-zero error code: 1

StevenDavenport, 2022-08-04:
Hi, when running the benchbot install script, I get the error: 'Building BenchBot "core" returned a non-zero error code: 1'

I've checked docker can run without root access, it seems that it can. 

Read threads #59, #60 - Unsure where to go from this. 

Output:

################################################################################
######################## PART 3: BUILDING DOCKER IMAGES ########################
################################################################################

BUILDING BENCHBOT CORE DOCKER IMAGE:
unknown shorthand flag: 't' in -t
See 'docker --help'.

Usage:  docker [OPTIONS] COMMAND

A self-sufficient runtime for containers

Options:
      --config string      Location of client config files (default "/home/steven/.docker")
  -c, --context string     Name of the context to use to connect to the daemon (overrides
                           DOCKER_HOST env var and default context set with "docker context use")
  -D, --debug              Enable debug mode
  -H, --host list          Daemon socket(s) to connect to
  -l, --log-level string   Set the logging level ("debug"|"info"|"warn"|"error"|"fatal")
                           (default "info")
      --tls                Use TLS; implied by --tlsverify
      --tlscacert string   Trust certs signed only by this CA (default
                           "/home/steven/.docker/ca.pem")
      --tlscert string     Path to TLS certificate file (default "/home/steven/.docker/cert.pem")
      --tlskey string      Path to TLS key file (default "/home/steven/.docker/key.pem")
      --tlsverify          Use TLS and verify the remote
  -v, --version            Print version information and quit

Management Commands:
  builder     Manage builds
  config      Manage Docker configs
  container   Manage containers
  context     Manage contexts
  image       Manage images
  manifest    Manage Docker image manifests and manifest lists
  network     Manage networks
  node        Manage Swarm nodes
  plugin      Manage plugins
  secret      Manage Docker secrets
  service     Manage services
  stack       Manage Docker stacks
  swarm       Manage Swarm
  system      Manage Docker
  trust       Manage trust on Docker images
  volume      Manage volumes

Commands:
  attach      Attach local standard input, output, and error streams to a running container
  build       Build an image from a Dockerfile
  commit      Create a new image from a container's changes
  cp          Copy files/folders between a container and the local filesystem
  create      Create a new container
  diff        Inspect changes to files or directories on a container's filesystem
  events      Get real time events from the server
  exec        Run a command in a running container
  export      Export a container's filesystem as a tar archive
  history     Show the history of an image
  images      List images
  import      Import the contents from a tarball to create a filesystem image
  info        Display system-wide information
  inspect     Return low-level information on Docker objects
  kill        Kill one or more running containers
  load        Load an image from a tar archive or STDIN
  login       Log in to a Docker registry
  logout      Log out from a Docker registry
  logs        Fetch the logs of a container
  pause       Pause all processes within one or more containers
  port        List port mappings or a specific mapping for the container
  ps          List containers
  pull        Pull an image or a repository from a registry
  push        Push an image or a repository to a registry
  rename      Rename a container
  restart     Restart one or more containers
  rm          Remove one or more containers
  rmi         Remove one or more images
  run         Run a command in a new container
  save        Save one or more images to a tar archive (streamed to STDOUT by default)
  search      Search the Docker Hub for images
  start       Start one or more stopped containers
  stats       Display a live stream of container(s) resource usage statistics
  stop        Stop one or more running containers
  tag         Create a tag TARGET_IMAGE that refers to SOURCE_IMAGE
  top         Display the running processes of a container
  unpause     Unpause all processes within one or more containers
  update      Update configuration of one or more containers
  version     Show the Docker version information
  wait        Block until one or more containers stop, then print their exit codes

Run 'docker COMMAND --help' for more information on a command.

To get more help with docker, check out our guides at https://docs.docker.com/go/guides/

ERROR: Building BenchBot "core" returned a non-zero error code: 1

Ensure that Docker has been installed correctly AND that you can run Docker 
WITHOUT root access (there is no need to ever run Docker with root). See 
https://docs.docker.com/install/linux/linux-postinstall/ for details on how to 
fix this. 

If the error is more generic, please contact us so that we can update our
pre-install host system checks.

btalb, 2022-08-05:
Thanks for reporting @StevenDavenport .

This is one I haven't seen before. Can I get you to share a full log so I can dig a bit deeper:
```
benchbot_install 2>&1 | tee benchbot.log
```

StevenDavenport, 2022-08-05:
I was under the impression benchbot would have to be installed before commands like this would work? Perhaps I've missed something.

With the command: "benchbot_install 2>&1 | tee benchbot.log"

I get the output: "benchbot_install: command not found"

StevenDavenport, 2022-08-09:
So the issue seems to be with the following line in the benchbot_install file.

docker buildx build -t "$DOCKER_TAG_CORE" -f "PATH_DOCKERFILE_CORE" \ etc..

As the error is "unknown shorthand 't' in -t" I changed the -t to --tag. This produced the same error. 

I don't know loads about docker so I'm really struggling to figure this one out. 

Any help greatly appreciated!!

StevenDavenport, 2022-08-09:
My bad, didn't mean to close..

So the issue seems to be with the following line in the benchbot_install file.

docker buildx build -t "$DOCKER_TAG_CORE" -f "PATH_DOCKERFILE_CORE" \ etc..

As the error is "unknown shorthand 't' in -t" I changed the -t to --tag. This produced the same error.

I don't know loads about docker so I'm really struggling to figure this one out.

Any help greatly appreciated!!

StevenDavenport, 2022-08-10:
The issue was with the command docker buildx. To fix this I re-installed docker and docker-desktop, ensuring that I had the docker buildx binaries.

btalb, 2022-08-10:
Sorry @StevenDavenport for not getting back to you sooner; glad you were able to resolve.

I was struggling to re-produce the error, and wasn't sure if it was to do with using 22.04 (you had a newer version of Docker than available on 20.04). 

Can I you provide me the result of `apt list --installed | grep docker` now you've got it installed. I'd like to see if there are installer checks I should update to catch this issue?

StevenDavenport, 2022-08-11:
No problem @btalb the results are:
```

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

docker-ce-cli/jammy,now 5:20.10.17~3-0~ubuntu-jammy amd64 [installed]
docker-ce-rootless-extras/jammy,now 5:20.10.17~3-0~ubuntu-jammy amd64 [installed,automatic]
docker-ce/jammy,now 5:20.10.17~3-0~ubuntu-jammy amd64 [installed]
docker-compose-plugin/jammy,now 2.6.0~ubuntu-jammy amd64 [installed]
docker-desktop/now 4.11.0-83626 amd64 [installed,local]
docker-scan-plugin/jammy,now 0.17.0~ubuntu-jammy amd64 [installed,automatic]
```

btalb, 2022-08-11:
Thanks. All seems similar to what's on my working systems.

I'll close this for now, but re-open if this pops up again in 22.04 installs.

closed 2022-08-11 by btalb