qcr/benchbot #80: Problems installing.

StevenDavenport, 2022-08-10:
I've encountered some errors while installing benchbot.

The installation process gets to the: 
`"BAKING SIMULATOR DATA INTO 'sim_omni' DOCKER IMAGE:"`

Then am presented with multiple errors as follows:

```
|---------------------------------------------------------------------------------------------|
| Driver Version: 515.65      | Graphics API: Vulkan
|======================================================|

| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA RTX A4000 Laptop GPU      | Yes: 0 |     | 8192    MB | 10de      | 0          |
|     |                                  |        |     |            | 24b7      | 994ad1f3.. |
|======================================================|
| OS: Linux eeaab2d3b310, Version: 5.15.0-46-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101003 (1.21.1.3)
| Processor: 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz | Cores: Unknown | Logical: 16
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 15193 | Free Memory: 12512
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
libGL error: MESA-LOADER: failed to retrieve device information
libGL error: Version 4 or later of flush extension not found
libGL error: failed to load driver: i915
libGL error: failed to open /dev/dri/card0: No such file or directory
libGL error: failed to load driver: iris
X Error of failed request:  GLXBadFBConfig
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  34 ()
  Serial number of failed request:  264
  Current serial number in output stream:  262
There was an error running python
```

btalb, 2022-08-10:
Sorry you're still having issues @StevenDavenport.

Can I get some more information on the system you're attempting to use BenchBot on? 

What I'm deducing from the logs:
- you've got a laptop with both an iGPU and dGPU (like a Dell 7650 w i7-11850H & NVIDIA RTX A4000)
- [NVIDIA Optimus](https://en.wikipedia.org/wiki/Nvidia_Optimus) would be running in some fashion to manage usage of the multi-GPU situation
- the BenchBot installer is getting confused at this step, choosing the iGPU when attempting the first Omniverse run

We can dig further into trying to get this setup running, but I'd just like to flag up front. I don't think that GPU will work with Omniverse based on [this](https://docs.omniverse.nvidia.com/prod_materials-and-rendering/prod_materials-and-rendering/manual.html#:~:text=Supported%20GPUs%20and%20APIs,-The%20NVIDIA%20Omniverse&text=Omniverse%20RTX%20Renderer%20supports%20the,A4000%2C%20A5000%2C%20A6000%2C%20A40). 

The documentation isn't particularly clear, nor up to date. We've had BenchBot working with other 20 & 30 series GPUs, but I'm not sure it'll work with a laptop GPU.

Do you have access to a machine with an NVIDIA desktop GPU?

StevenDavenport, 2022-08-11:
Thank you for your help @btalb!

More information about my system in the below screenshot:
[image]

I unfortunately don't currently have access to a machine with an NVIDIA desktop GPU.

I'm very eager to get BenchBot running on this machine, but I understand if it's not possible. Cheers for your guidance.

btalb, 2022-08-11:
I can't offer too much in direct support without a platform to test on sorry @StevenDavenport , but I can offer some advice based on my experiences working with hybrid graphics in the past:
- To keep things simple when checking if things _will_ work, configure [NVIDIA Optimus to _only_ use the NVIDIA card](https://wiki.archlinux.org/title/NVIDIA_Optimus)
- Run commands like `glxgears` and `glxinfo` to confirm the NVIDIA GPU is your renderer (hint: whenever you see "Mesa" that's Intel)
- Check what Vulkan will do with `vkcube` and `vulkaninfo` (as that's what Omniverse will use)
- Once you're sure that's all Intel GPU free, then see if the installer will run

Keep in mind this could all still end up not working if NVIDIA explicitly block that GPU somewhere in Omniverse code. At best experience may not be great.

Sorry that's not the most inspiring answer :pensive:

StevenDavenport, 2022-08-12:
Ahh big shame, I would love to run BenchBot, I have some great RL ideas!

I'll give it a whirl, regardless I'll learn a bit about NVIDIA Optimus. 

Cheers for all the help and support!

closed 2022-08-16 by btalb