qcr/benchbot #81: ERROR: An exception occurred while executing "benchbot_run"

ffgao, 2022-08-11:
Hi,When I installed the program and executed "benchbot_run" command, the following error occurred.
`benchbot_run --robot carter_omni --env miniroom:1 --task semantic_slam:passive:ground_truth

################################################################################
################# CHECKING FOR BENCHBOT SOFTWARE STACK UPDATES #################
################################################################################

Checking BenchBot version ...				Up-to-date.
Checking BenchBot API version ...			Up-to-date.
Checking BenchBot Eval version ...			Up-to-date.
Checking BenchBot Supervisor version ...		Up-to-date.
Checking installed BenchBot add-ons are up-to-date ...	Up-to-date.

BenchBot is up-to-date.

################################################################################
###################### CLEANING UP ALL BENCHBOT REMNANTS #######################
################################################################################

Sending stop request to running controller:
{"stop_success":true}

Deleted the following containers:
Total reclaimed space: 0B

Finished cleaning! (use 'benchbot_run -k' for a full clean)

################################################################################
##################### STARTING THE BENCHBOT SOFTWARE STACK #####################
################################################################################

Running the BenchBot system with the following settings:

    Selected task:        semantic_slam:passive:ground_truth
    Task results format:  object_map
    Selected robot:       carter_omni
    Selected environment: miniroom:1
    Scene/s:              miniroom:1, starting @ pose [0.7, 0, 0, -0.7, 1.2, 1.5, 0.3]
                          (map_path = '.sim_data/miniroom_1.usd')
    Simulator required:   Yes (sim_omni)

Creating shared network 'benchbot_network':

Starting persistent container for ROS core:
Skipping (already running)

Starting persistent container for BenchBot Robot Controller (sim_omni):
Skipping (already running)

Starting container for BenchBot Supervisor:
7f1589564ca18af1020a65f1e8e92058aeb3f1d8713aa86eddefaa117818ded4

Starting container for BenchBot Debugging:
d1e6f2eb17d5aa82d63936cf39747199fedc66936cd17844f777473c0eafb263

################################################################################
################### BENCHBOT IS RUNNING (Ctrl^C to exit) ... ###################
################################################################################

Initialising supervisor...

Configuring the supervisor...
Starting a supervisor with the following configuration:

{'environments': [{'_file_path': '/benchbot/addons/benchbot_addons/benchbot-addons/envs_bear_develop_sim_omni/environments/miniroom_1.yaml',
                   'description': 'Mini room environment with all of the base '
                                  'objects in their normal place.\n'
                                  'All of the other mini room environments are '
                                  'based off this with any combination\n'
                                  'of objects, object positions, & lighting '
[... 112 lines cut ...]
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: [Errno Expecting value] <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>
: 0

################################################################################
###################### CLEANING UP ALL BENCHBOT REMNANTS #######################
################################################################################

Sending stop request to running controller:
{"stop_success":true}

Stopped the following containers:
d1e6f2eb17d5

Deleted the following containers:
Deleted Containers:
d1e6f2eb17d5aa82d63936cf39747199fedc66936cd17844f777473c0eafb263
7f1589564ca18af1020a65f1e8e92058aeb3f1d8713aa86eddefaa117818ded4

Total reclaimed space: 189.2kB

Finished cleaning! (use 'benchbot_run -k' for a full clean)`

btalb, 2022-08-11:
Thanks for reporting @ffgao .

Have you got more context of what was going on when this happened:
- had the simulator window popped up?
- was this the first run since you installed? Or had previous runs worked on your system?
- does this error occur every time you do `benchbot_run`?
- anything interesting about your system setup? Are you running remotely? SSH'ed in? Or sitting directly at the machine?

Struggling to reproduce this locally, so any extra information you may have is helpful.

ffgao, 2022-08-12:
Thank you for your help @btalb 
More information is below:
- no simulator window.
- first install and do this every time
- Running locally
- I installed isaac sim 2022 at host

Below is a running record:
[image]

And then it was wrong

btalb, 2022-08-16:
Hmmm.... sorry about the delay @ffgao , I'm struggling to immediately see what could be causing this (and can't reproduce locally).

I might need a bit deeper in terms of logging:
- Can you uncomment this line in your install so `benchbot_run` doesn't exit immediately after the error: https://github.com/qcr/benchbot/blob/12daa9335e391b10170d372cdfad24df12d0e320/bin/benchbot_run#L398
- then send me the entire output of `docker logs benchbot_robot_controller`

You may have to do a `benchbot_run -k` to clean up after this.

ffgao, 2022-08-17:
No **benchbot_robot_controller** in the list of container.@btalb
The following is my operation.
`docker ps
CONTAINER ID   IMAGE                         COMMAND                  CREATED          STATUS          PORTS                                                                                                                                                                                              NAMES
372aaf8feeb4   benchbot/backend:base         "/bin/bash"              16 minutes ago   Up 16 minutes                                                                                                                                                                                                      benchbot_debug
c555e44fb73c   benchbot/simulator:sim_omni   "/bin/bash -c 'sourc…"   16 minutes ago   Up 16 minutes   3007/tcp, 3009/tcp, 3080/tcp, 3085/tcp, 3333/tcp, 8011-8012/tcp, 8080/tcp, 8211/tcp, 8888/tcp, 8891/tcp, 8899/tcp, 47995-48012/tcp, 47995-48012/udp, 49000-49007/tcp, 49100/tcp, 49000-49007/udp   benchbot_robot
40f0281b23a5   benchbot/backend:base         "/bin/bash -c 'sourc…"   16 minutes ago   Up 16 minutes                                                                                                                                                                                                      benchbot_ros
`

david2611, 2022-10-11:
Apologies that we have been so slow in fixing problems.

I have also come across this when trying to verify a fix for #82. 

If you perform a `benchbot_install -b develop` operation and run again you should get more meaningful output from your `benchbot_robot` log. Would you be able to send through what you receive so I can confirm it is the same as what I have been getting?

david2611, 2022-10-17:
While current solution is not pretty I have put a hotfix in place that seems to solve this issue for me. A fresh `benchbot_install` on the main branch should sort this out.

Worse case scenario I have found with the hotfix is that the first `benchbot_run` seems to fail but all subsequent ones succeed.

I will work to fix this bug but will open a new issue regarding this.

Can you confirm that this fixes the issues you originally encountered?

david2611, 2023-01-09:
Local testing seemed to show this issue as fixed. Closing due to inactivity

closed 2023-01-09 by david2611