qcr/benchbot #82: Installation error

Jianhao-zheng, 2022-09-07:
Hello, When I run "./install"
I get the following error in BUILDING BENCHBOT BACKEND DOCKER IMAGE:

#0 4.108 HEAD is now at b7a85ed Merge branch 'develop'
#0 4.814 Collecting colorama (from -r requirements.txt (line 1))
#0 5.159   Downloading https://files.pythonhosted.org/packages/77/8b/7550e87b2d308a1b711725dfaddc19c695f8c5fa413c640b2be01662f4e6/colorama-0.4.5-py2.py3-none-any.whl
#0 5.239 Collecting flask (from -r requirements.txt (line 2))
#0 5.321   Downloading https://files.pythonhosted.org/packages/e8/6d/994208daa354f68fd89a34a8bafbeaab26fda84e7af1e35bdaed02b667e6/Flask-1.1.4-py2.py3-none-any.whl (94kB)
#0 5.452 Collecting gevent (from -r requirements.txt (line 3))
#0 6.101   Using cached https://files.pythonhosted.org/packages/c8/18/631398e45c109987f2d8e57f3adda161cc5ff2bd8738ca830c3a2dd41a85/gevent-21.12.0.tar.gz
#0 7.344 Collecting jsonpickle (from -r requirements.txt (line 4))
#0 7.516   Downloading https://files.pythonhosted.org/packages/c6/85/b4920d8087ef480eed4e7b6b0d46c90674e923e59b22e7929fd17aba5030/jsonpickle-2.2.0-py2.py3-none-any.whl
#0 7.605 Requirement already satisfied: numpy in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 5))
#0 7.606 Requirement already satisfied: pyyaml in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 6))
#0 7.629 Collecting requests (from -r requirements.txt (line 7))
#0 7.753   Using cached https://files.pythonhosted.org/packages/2d/61/08076519c80041bc0ffa1a8af0cbd3bf3e2b62af10435d269a9d0f40564d/requests-2.27.1-py2.py3-none-any.whl
#0 7.821 Collecting scipy (from -r requirements.txt (line 8))
#0 8.473   Downloading https://files.pythonhosted.org/packages/24/40/11b12af7f322c1e20446c037c47344d89bab4922b8859419d82cf56d796d/scipy-1.2.3-cp27-cp27mu-manylinux1_x86_64.whl (24.8MB)
#0 19.63 Collecting transforms3d (from -r requirements.txt (line 9))
#0 19.79   Downloading https://files.pythonhosted.org/packages/05/bb/6d9bc2d2cc82c7984e0c6af586e561e8ca2e9e641483e617bbd256e5a8da/transforms3d-0.4.tar.gz (1.4MB)
#0 20.76     Complete output from command python setup.py egg_info:
#0 20.76     Traceback (most recent call last):
#0 20.76       File "<string>", line 1, in <module>
#0 20.76       File "/tmp/pip-build-QOxthY/transforms3d/setup.py", line 8, in <module>
#0 20.76         import versioneer
#0 20.76       File "versioneer.py", line 370
#0 20.76         LONG_VERSION_PY: Dict[str, str] = {}
#0 20.76                        ^
#0 20.76     SyntaxError: invalid syntax
#0 20.76     
#0 20.76     ----------------------------------------
#0 20.82 Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-QOxthY/transforms3d/
------
error: failed to solve: executor failed running [/bin/bash -c git clone $BENCHBOT_CONTROLLER_GIT $BENCHBOT_CONTROLLER_PATH &&     pushd $BENCHBOT_CONTROLLER_PATH && git checkout $BENCHBOT_CONTROLLER_HASH &&     pip install -r requirements.txt && pushd $ROS_WS_PATH &&     pushd src && git clone https://github.com/eric-wieser/ros_numpy && popd &&     ln -sv $BENCHBOT_CONTROLLER_PATH src/ && source devel/setup.bash && catkin_make]: exit code: 1

ERROR: Building BenchBot backend returned a non-zero error code: 1

Ensure that Docker has been installed correctly AND that you can run Docker 
WITHOUT root access (there is no need to ever run Docker with root). See 
https://docs.docker.com/install/linux/linux-postinstall/ for details on how to 
fix this. 

If the error is more generic, please contact us so that we can update our
pre-install host system checks.

__________________________________________________________________________________________________
It seems that the problem is from the pip installation of transforms3d. I tried to install it in my own conda environment with pip and it works. Do you have any ideas on how to fix this?

For further information, I attached log file here: [benchbot.log](https://github.com/qcr/benchbot/files/9506963/benchbot.log).

Thank you very much! :)

SophieLeichtle, 2022-09-13:
This seems related to this
https://github.com/matthew-brett/transforms3d/issues/49
is there any way to work around this or can benchbot just not be installed until this is fixed?

david2611, 2022-10-17:
Thanks to @Jianhao-zheng  for submitting the report and @SophieLeichtle for the information. Apologies for the delay in getting a fixed version public. #81 caused larger issues that prevented testing to confirm that installation created and ran correctly.

The latest version of benchbot should now install without this issue. 

Please let us know if you continue to have issues with installing benchbot.

david2611, 2023-01-09:
Local testing seemed to fix this issue. Closing due to inactivity

closed 2023-01-09 by david2611