qcr/benchbot #87: install error

zhangzscn, 2023-01-31:
./install 

################################################################################
###################### CHECKING BENCHBOT SCRIPTS VERSION #######################
################################################################################

Fetching latest hash for Benchbot scripts ... 
		12daa9335e391b10170d372cdfad24df12d0e320.

BenchBot scripts are up-to-date.

################################################################################
######################## PART 1: EXAMINING SYSTEM STATE ########################
################################################################################

Core host system checks:
	Ubuntu version >= 18.04:                                  Passed (22.04)

Running Nvidia related system checks:
	NVIDIA GPU available:                     Found card of type '10de:2684'
	NVIDIA driver is running:                                          Found
	NVIDIA driver version valid:                           Valid (525.60.13)
	NVIDIA driver from a standard PPA:                          PPA is valid
	CUDA drivers installed:                                    Drivers found
	CUDA drivers version valid:                          Valid (525.60.13-1)
	CUDA drivers from the NVIDIA PPA:                           PPA is valid
	CUDA is installed:                                            CUDA found
	CUDA version valid:                                         Valid (12.0)
	CUDA is from the NVIDIA PPA:                                PPA is valid

Running Docker related system checks:
	Docker is available:                                               Found
	Docker version valid:                                   Valid (20.10.23)
	NVIDIA Container Toolkit installed:                       Found (1.11.0)
	Docker runs without root:                                         Passed

Running checks of filesystem used for Docker:
	/var/lib/docker on ext4 filesystem:                 Yes (/dev/nvme0n1p2)
	/var/lib/docker supports suid:                                   Enabled
	/var/lib/docker drive space check:              Sufficient space (1583G)

Miscellaneous requirements:
	Pip python package manager available:                     Found (22.2.2)
	Tkinter for Python installed:                                      Found
	PIL (with ImageTk) for Python install                              Found

Manual installation steps for Omniverse-powered Isaac Sim:
	License accepted for Omniverse:                                      Yes
	Access to nvcr.io Docker registry:                                   Yes

All requirements & dependencies fulfilled. Docker containers for the BenchBot
software stack will now be built (which may take anywhere from a few seconds
to many hours). Would you like to proceed (y/N)? y

Proceeding with install ... 

################################################################################
################ PART 2: FETCHING LATEST BENCHBOT VERSION INFO #################
################################################################################

Fetching latest hash for BenchBot Simulator 'sim_omni' ... 
		23355ffecf560529fdaeee179d81bfa90aad6b34.
Fetching latest hash for BenchBot Robot Controller ... 
		5ba97fbaceaa5d4f4c7ee04db9c7595d6df1a8dc.
Fetching latest hash for BenchBot Supervisor ... 
		e5e50c46373d24176eb6e99dcc93661086571a68.
Fetching latest hash for BenchBot API ... 
		03ae4f9eef40a750d87bc112df837ee062097228.
Fetching latest hash for BenchBot ROS Messages ... 
		29153dd20b7e611088318cc4b3ef5f1770f44e87.
[... 26 lines cut ...]
#0 8.832 Reading package lists...
#0 9.271 Building dependency tree...
#0 9.330 Reading state information...
#0 9.361 Some packages could not be installed. This may mean that you have
#0 9.361 requested an impossible situation or if you are using the unstable
#0 9.361 distribution that some required packages have not yet been created
#0 9.361 or been moved out of Incoming.
#0 9.361 The following information may help to resolve the situation:
#0 9.361 
#0 9.361 The following packages have unmet dependencies:
#0 9.387  libnvidia-gl-525 : Depends: libnvidia-common-525 (= 525.60.13-0ubuntu1) but 525.85.12-0ubuntu1 is to be installed
#0 9.387  nvidia-dkms-525 : Depends: nvidia-kernel-common-525 (= 525.60.13-0ubuntu1) but 525.85.12-0ubuntu1 is to be installed
#0 9.392 E: Unable to correct problems, you have held broken packages.
------
ERROR: failed to solve: executor failed running [/bin/bash -c CUDA_NAME="cuda-$(echo "${CUDA_VERSION}" |     sed 's/\([0-9]*\)\.\([0-9]*\).*/\1\.\2/; s/\./-/')" &&     NVIDIA_NAME="nvidia-driver-$(echo "${NVIDIA_DRIVER_VERSION}" |     sed 's/\(^[0-9]*\).*/\1/')" &&     NVIDIA_DEPS="$(apt depends "$NVIDIA_NAME=$NVIDIA_DRIVER_VERSION" 2>/dev/null |     grep '^ *Depends:' | sed 's/.*Depends: \([^ ]*\) (.\?= \([^)]*\))/\1 \2/' |     while read d; do read a b <<< "$d"; v=$(apt policy "$a" 2>/dev/null |     grep "$b" | grep -vE "(Installed|Candidate)" | sed "s/.*\($b[^ ]*\).*/\1/");     echo "$a=$v"; done)" &&     CUDA_DRIVERS_DEPS="$(apt depends "cuda-drivers=$CUDA_DRIVERS_VERSION" 2>/dev/null |     grep '^ *Depends:' | sed 's/.*Depends: \([^ ]*\) (.\?= \([^)]*\))/\1 \2/' |     while read d; do read a b <<< "$d"; v=$(apt policy "$a" 2>/dev/null |     grep "$b" | grep -vE "(Installed|Candidate)" | sed "s/.*\($b[^ ]*\).*/\1/");     echo "$a=$v"; done)" &&     CUDA_DEPS="$(apt depends "$CUDA_NAME=$CUDA_VERSION" 2>/dev/null |     grep '^ *Depends:' | sed 's/.*Depends: \([^ ]*\) (.\?= \([^)]*\))/\1 \2/' |     while read d; do read a b <<< "$d"; v=$(apt policy "$a" 2>/dev/null |     grep "$b" | grep -vE "(Installed|Candidate)" | sed "s/.*\($b[^ ]*\).*/\1/");     echo "$a=$v"; done)" &&     TARGETS="$(echo "$NVIDIA_DEPS $NVIDIA_NAME=$NVIDIA_DRIVER_VERSION"     "$CUDA_DRIVERS_DEPS cuda-drivers=$CUDA_DRIVERS_VERSION"     "$CUDA_DEPS $CUDA_NAME=$CUDA_VERSION" |     tr '\n' ' ')" &&     DEBIAN_FRONTEND=noninteractive apt install -yq $TARGETS]: exit code: 100

ERROR: Building BenchBot "core" returned a non-zero error code: 1

Ensure that Docker has been installed correctly AND that you can run Docker 
WITHOUT root access (there is no need to ever run Docker with root). See 
https://docs.docker.com/install/linux/linux-postinstall/ for details on how to 
fix this. 

If the error is more generic, please contact us so that we can update our
pre-install host system checks.

zhangzscn, 2023-02-01:
Are there any plans to update the code to support the latest hardware and software？

The configuration in use：
ubuntu 22.04
i9-13900k
rtx 4090

btalb, 2023-02-07:
Thanks for reporting @zhangzscn .

There's no reason the code shouldn't support those hardware and software specs. @david2611 what NVIDIA driver is your latest BenchBot install using? Is it using 525?

Your logs seem to suggest you've edited the [`core` Dockerfile](https://github.com/qcr/benchbot/blob/master/docker/core.Dockerfile)? Layers 5,6,7,8 don't match what's in the current  Dockerfile.

Can you past the output of an `./install` with a clean `benchbot` directory please? 

1. Run the following in your `benchbot` directory:
    ```
    ( git reset --hard origin/master && ./install 2>&1 ) | tee benchbot.log
    ```
2. Post the `benchbot.log` file here

david2611, 2023-04-12:
525 is the current version that is known to work and is recommended.

david2611, 2023-04-12:
With the latest update for BenchBot using the latest Omniverse version, more up-to-date hardware should now work

david2611, 2023-07-21:
There has been an update to the install procedure that fixes some cuda driver compatability issues that others had encountered. If a fresh install fixes issues do let us know. Issue will be closed due to inactivity otherwise

closed 2023-07-21 by zhangzscn