qcr/benchbot #88: install error

ThePassedWind, 2023-02-06:
./install
################################################################################
###################### CHECKING BENCHBOT SCRIPTS VERSION #######################
################################################################################

Fetching latest hash for Benchbot scripts ... 
fatal: not a git repository (or any of the parent directories): .git
		12daa9335e391b10170d372cdfad24df12d0e320.

BenchBot scripts are outdated. Updating & restarting install script ...

fatal: not a git repository (or any of the parent directories): .git

Done.

################################################################################
######################## PART 1: EXAMINING SYSTEM STATE ########################
################################################################################

Core host system checks:
	Ubuntu version >= 18.04:                                  Passed (18.04)

Running Nvidia related system checks:
	NVIDIA GPU available:                     Found card of type '10de:1e81'
	NVIDIA driver is running:                                          Found
	NVIDIA driver version valid:                           Valid (525.85.12)
	NVIDIA driver from a standard PPA:                          PPA is valid
	CUDA drivers installed:                                    Drivers found
	CUDA drivers version valid:                          Valid (525.85.12-1)
	CUDA drivers from the NVIDIA PPA:                           PPA is valid
	CUDA is installed:                                            CUDA found
	CUDA version valid:                                         Valid (12.0)
	CUDA is from the NVIDIA PPA:                                PPA is valid

Running Docker related system checks:
	Docker is available:                                               Found
	Docker version valid:                                     Valid (23.0.0)
	NVIDIA Container Toolkit installed:                       Found (1.11.0)
	Docker runs without root:                                         Passed

Running checks of filesystem used for Docker:
	/var/lib/docker on ext4 filesystem:                 Yes (/dev/nvme0n1p2)
	/var/lib/docker supports suid:                                   Enabled
	/var/lib/docker drive space check:               Sufficient space (203G)

Miscellaneous requirements:
	Pip python package manager available:                     Found (21.3.1)
	Tkinter for Python installed:                                      Found
	PIL (with ImageTk) for Python install                              Found

Manual installation steps for Omniverse-powered Isaac Sim:
	License accepted for Omniverse:                                      Yes
	Access to nvcr.io Docker registry:                                   Yes

All requirements & dependencies fulfilled. Docker containers for the BenchBot
software stack will now be built (which may take anywhere from a few seconds
to many hours). Would you like to proceed (y/N)? y

Proceeding with install ... 

################################################################################
################ PART 2: FETCHING LATEST BENCHBOT VERSION INFO #################
################################################################################

Fetching latest hash for BenchBot Simulator 'sim_omni' ... 
		23355ffecf560529fdaeee179d81bfa90aad6b34.
Fetching latest hash for BenchBot Robot Controller ... 
		5ba97fbaceaa5d4f4c7ee04db9c7595d6df1a8dc.
Fetching latest hash for BenchBot Supervisor ... 
		e5e50c46373d24176eb6e99dcc93661086571a68.
[... 39 lines cut ...]
#0 16.72 Get:8 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic/main amd64 vdpau-driver-all amd64 1.3-0ubuntu0~gpu18.04.2 [8392 B]
#0 18.86 Get:9 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 keyboard-configuration all 1.178ubuntu2.9 [369 kB]
#0 20.70 Get:10 http://archive.ubuntu.com/ubuntu bionic/main amd64 libargon2-0 amd64 0~20161029-1.1 [19.9 kB]
...
...
...
**#0 304.0 Err:243 http://archive.ubuntu.com/ubuntu bionic/main amd64 xfonts-base all 1:1.0.4+nmu1
#0 304.0   Connection failed [IP: 185.125.190.36 80]**
#0 351.5 Get:244 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  libnpp-12-0 12.0.1.104-1 [95.4 MB]
#0 403.1 Get:245 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  libnvjitlink-12-0 12.0.140-1 [14.6 MB]
#0 410.5 Get:246 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  libnvjpeg-12-0 12.0.1.102-1 [1761 kB]
#0 411.4 Get:247 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  cuda-libraries-12-0 12.0.1-1 [2528 B]
#0 411.4 Get:248 http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  libnvidia-common-525 525.85.12-0ubuntu1 [10.5 kB]

**ERROR**: failed to solve: process "/bin/bash -c CUDA_NAME=\"cuda-$(echo \"${CUDA_VERSION}\" |     sed 's/\\([0-9]*\\)\\.\\([0-9]*\\).*/\\1\\.\\2/; s/\\./-/')\" &&     NVIDIA_NAME=\"nvidia-driver-$(echo \"${NVIDIA_DRIVER_VERSION}\" |     sed 's/\\(^[0-9]*\\).*/\\1/')\" &&     NVIDIA_DEPS=\"$(apt depends \"$NVIDIA_NAME=$NVIDIA_DRIVER_VERSION\" 2>/dev/null |     grep '^ *Depends:' | sed 's/.*Depends: \\([^ ]*\\) (.\\?= \\([^)]*\\))/\\1 \\2/' |     while read d; do read a b <<< \"$d\"; v=$(apt policy \"$a\" 2>/dev/null |     grep \"$b\" | grep -vE \"(Installed|Candidate)\" | sed \"s/.*\\($b[^ ]*\\).*/\\1/\");     echo \"$a=$v\"; done)\" &&     CUDA_DRIVERS_DEPS=\"$(apt depends \"cuda-drivers=$CUDA_DRIVERS_VERSION\" 2>/dev/null |     grep '^ *Depends:' | sed 's/.*Depends: \\([^ ]*\\) (.\\?= \\([^)]*\\))/\\1 \\2/' |     while read d; do read a b <<< \"$d\"; v=$(apt policy \"$a\" 2>/dev/null |     grep \"$b\" | grep -vE \"(Installed|Candidate)\" | sed \"s/.*\\($b[^ ]*\\).*/\\1/\");     echo \"$a=$v\"; done)\" &&     CUDA_DEPS=\"$(apt depends \"$CUDA_NAME=$CUDA_VERSION\" 2>/dev/null |     grep '^ *Depends:' | sed 's/.*Depends: \\([^ ]*\\) (.\\?= \\([^)]*\\))/\\1 \\2/' |     while read d; do read a b <<< \"$d\"; v=$(apt policy \"$a\" 2>/dev/null |     grep \"$b\" | grep -vE \"(Installed|Candidate)\" | sed \"s/.*\\($b[^ ]*\\).*/\\1/\");     echo \"$a=$v\"; done)\" &&     TARGETS=\"$(echo \"$NVIDIA_DEPS $NVIDIA_NAME=$NVIDIA_DRIVER_VERSION\"     \"$CUDA_DRIVERS_DEPS cuda-drivers=$CUDA_DRIVERS_VERSION\"     \"$CUDA_DEPS $CUDA_NAME=$CUDA_VERSION\" |     tr '\\n' ' ')\" &&     DEBIAN_FRONTEND=noninteractive apt install -yq $TARGETS" did not complete successfully: exit code: 100

**ERROR**: Building BenchBot "core" returned a non-zero error code: 1

Ensure that Docker has been installed correctly AND that you can run Docker 
WITHOUT root access (there is no need to ever run Docker with root). See 
https://docs.docker.com/install/linux/linux-postinstall/ for details on how to 
fix this. 

If the error is more generic, please contact us so that we can update our
pre-install host system checks.

ThePassedWind, 2023-02-07:
Except for the problem that I have proposed, another question is "gnutls_handshake() failed: Error in the pull function."
Maybe the reasons are the same - it is not stable to connect to the website 'https://github.com/qcr/xxx' in your install program(command './install'), even though my VPN is good enough to visit any github website.

The detailed errors in PART2-FETCHING LATEST BENCHBOT VERSION INFO:
1. fatal: unable to access 'https://github.com/qcr/benchbot_sim_omni/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
2. fatal: unable to access 'https://github.com/qcr/benchbot_robot_controller/': gnutls_handshake() failed: Error in the pull function.

The detailed errors in PART3-BUILDING DOCKER IMAGES
1. when running the command 'apt update && apt install -y python3 python3-pip &&     git clone https://github.com/qcr/benchbot_supervisor /benchbot/ben', TIMEOUT(install a doc which is just 40+MB) always happened.

btalb, 2023-02-07:
Thanks for reporting @ThePassedWind .

As you've noted they definitely all seem to be network errors.

Thoughts:
- the very start of the log seems odd `fatal: not a git repository (or any of the parent directories): .git`. Can you show me the output of `git status` in the same directory as the `install` script? If it shows any errors remove the entire benchbot directory and re-clone `git clone https://github.com/qcr/benchbot`
- the errors you're quoting in Part 2 don't appear in the initial log. Do they appear all the time, or intermittently?

The install script expects a stable internet connection and is not written to handle network failures. Is it possible to access a more stable network connection? The logs show network failures to GitHub and Ubuntu apt repositories, something that doesn't happen under normal conditions.

ThePassedWind, 2023-02-08:
Thank you! After I tried to open SSL proxy, my problems were solved! 
However, when **BUILDING BENCHBOT SIMULATOR 'sim_omni' DOCKER IMAGE in PART3**, some problems occurred:

BUILDING BENCHBOT SIMULATOR 'sim_omni' DOCKER IMAGE:
[+] Building 30.8s (8/15)                                                                                                               
 => [internal] load build definition from Dockerfile                                                                               0.0s
 => => transferring dockerfile: 3.56kB                                                                                             0.0s
 => [internal] load .dockerignore                                                                                                  0.0s
 => => transferring context: 52B                                                                                                   0.0s
 => [internal] load metadata for nvcr.io/nvidia/isaac-sim:2021.2.1                                                                 0.7s
 => [ 1/12] FROM nvcr.io/nvidia/isaac-sim:2021.2.1@sha256:4a9a1cfadc276c72dc15b64fde4b08c1030ca92005f480605950e54c324c0708         0.0s
 => CACHED [ 2/12] RUN apt-key adv --fetch-keys     https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/3b  0.0s
 => CACHED [ 3/12] RUN sed -i 's/$@/"\0"/' python.sh                                                                               0.0s
 => CACHED [ 4/12] RUN sed -i 's/sleep/# \0/' start_nucleus.sh                                                                     0.0s
 => ERROR [ 5/12] RUN apt update && apt install -y git                                                                            30.1s
------                                                                                                                                  
 > [ 5/12] RUN apt update && apt install -y git:                                                                                        
#0 0.263                                                                                                                                
#0 0.263 WARNING: apt does not have a stable CLI interface. Use with caution in scripts.                                                
#0 0.263                                                                                                                                
#0 0.848 Get:3 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]                                                    
#0 1.086 Get:4 http://archive.ubuntu.com/ubuntu bionic InRelease [242 kB]
**_#0 1.181 Get:1 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  InRelease [1581 B]
#0 1.259 Err:1 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64  InRelease
#0 1.259   The following signatures couldn't be verified because the public key is not available: NO_PUBKEY A4B469963BF863CC_**
#0 1.540 Ign:2 https://developer.download.nvidia.cn/compute/machine-learning/repos/ubuntu1804/x86_64  InRelease
#0 1.626 Get:5 https://developer.download.nvidia.cn/compute/machine-learning/repos/ubuntu1804/x86_64  Release [564 B]
#0 1.662 Get:6 https://developer.download.nvidia.cn/compute/machine-learning/repos/ubuntu1804/x86_64  Release.gpg [833 B]
#0 1.803 Get:7 https://developer.download.nvidia.cn/compute/machine-learning/repos/ubuntu1804/x86_64  Packages [73.8 kB]
#0 1.948 Get:8 http://security.ubuntu.com/ubuntu bionic-security/universe i386 Packages [1317 kB]
#0 2.570 Get:9 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
#0 2.902 Get:10 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [83.3 kB]
#0 3.240 Get:11 http://archive.ubuntu.com/ubuntu bionic/universe amd64 Packages [11.3 MB]
#0 3.482 Get:12 http://security.ubuntu.com/ubuntu bionic-security/restricted i386 Packages [38.8 kB]
#0 3.505 Get:13 http://security.ubuntu.com/ubuntu bionic-security/main i386 Packages [1627 kB]
#0 6.862 Get:14 http://security.ubuntu.com/ubuntu bionic-security/restricted amd64 Packages [1389 kB]
#0 7.064 Get:15 http://security.ubuntu.com/ubuntu bionic-security/multiverse i386 Packages [6602 B]
#0 7.069 Get:16 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [23.8 kB]
#0 7.083 Get:17 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [3161 kB]
#0 10.29 Get:18 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [1578 kB]
#0 14.67 Get:19 http://archive.ubuntu.com/ubuntu bionic/multiverse amd64 Packages [186 kB]
#0 14.75 Get:20 http://archive.ubuntu.com/ubuntu bionic/universe i386 Packages [11.3 MB]
#0 20.70 Get:21 http://archive.ubuntu.com/ubuntu bionic/main amd64 Packages [1344 kB]
#0 21.41 Get:22 http://archive.ubuntu.com/ubuntu bionic/main i386 Packages [1328 kB]
#0 22.11 Get:23 http://archive.ubuntu.com/ubuntu bionic/restricted amd64 Packages [13.5 kB]
#0 22.12 Get:24 http://archive.ubuntu.com/ubuntu bionic/multiverse i386 Packages [177 kB]
#0 22.21 Get:25 http://archive.ubuntu.com/ubuntu bionic/restricted i386 Packages [13.5 kB]
#0 22.22 Get:26 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [30.8 kB]
#0 22.23 Get:27 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [3581 kB]
#0 24.12 Get:28 http://archive.ubuntu.com/ubuntu bionic-updates/restricted amd64 Packages [1430 kB]
#0 24.88 Get:29 http://archive.ubuntu.com/ubuntu bionic-updates/main i386 Packages [1999 kB]
[... cut ...]
--------------------------------------------------------------------------------------------------------------------------------------------------

I tried all methods in the Internet but finally I failed to solve this problem! **Could you help me find the reason?**
I have added "deb https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu1804/x86_64 /" 
into the document ''/etc/apt/sources.list.d/cuda.list"
Then, I added the public key to apt by the command "apt-key adv --recv-keys --keyserver keyserver.ubuntu.com A4B469963BF863CC" and some commands like this.
Eventually, it didn't work.

btalb, 2023-02-09:
Hmmm. I've verified locally that those keys are definitely still valid:
```
ben@ben-meshify:~$ sudo apt update
[sudo] password for ben: 
Sorry, try again.
[sudo] password for ben: 
Get:1 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  InRelease [1,581 B]
Get:2 http://dl.google.com/linux/chrome/deb stable InRelease [1,811 B]         
Hit:3 https://download.docker.com/linux/ubuntu focal InRelease                 
Get:4 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Packages [871 kB]
Hit:5 https://pkg.cloudflare.com/cloudflared focal InRelease                   
Hit:6 http://archive.ubuntu.com/ubuntu focal InRelease                         
Hit:7 http://repository.spotify.com stable InRelease                           
Hit:8 http://ppa.launchpad.net/inkscape.dev/stable/ubuntu focal InRelease      
Get:9 http://dl.google.com/linux/chrome/deb stable/main amd64 Packages [1,061 B]
Ign:10 http://linux.dropbox.com/ubuntu disco InRelease                         
Get:11 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]       
Hit:12 http://packages.ros.org/ros/ubuntu focal InRelease                      
Hit:13 http://linux.dropbox.com/ubuntu disco Release                           
Hit:14 http://ppa.launchpad.net/jonathonf/vim/ubuntu focal InRelease           
Get:16 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]     
Hit:17 https://librealsense.intel.com/Debian/apt-repo focal InRelease          
Get:18 http://archive.ubuntu.com/ubuntu focal-security InRelease [114 kB]      
Hit:19 https://apt.kitware.com/ubuntu focal InRelease     
Get:20 http://archive.ubuntu.com/ubuntu focal-updates/main i386 Packages [785 kB]
Get:21 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [2,374 kB]
Get:22 http://archive.ubuntu.com/ubuntu focal-updates/main Translation-en [408 kB]
Get:23 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 DEP-11 Metadata [274 kB]
Get:24 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 c-n-f Metadata [16.3 kB]
Get:25 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1,595 kB]
Get:26 http://archive.ubuntu.com/ubuntu focal-updates/restricted Translation-en [224 kB]
Get:27 http://archive.ubuntu.com/ubuntu focal-updates/universe i386 Packages [710 kB]
Get:28 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1,024 kB]
Get:29 http://archive.ubuntu.com/ubuntu focal-updates/universe Translation-en [237 kB]
Get:30 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 DEP-11 Metadata [408 kB]
Get:31 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 c-n-f Metadata [23.5 kB]
Get:32 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 DEP-11 Metadata [940 B]
Get:33 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 DEP-11 Metadata [7,964 B]
Get:34 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 DEP-11 Metadata [30.5 kB]
Get:35 http://archive.ubuntu.com/ubuntu focal-security/main amd64 Packages [1,993 kB]
Get:36 http://archive.ubuntu.com/ubuntu focal-security/main i386 Packages [556 kB]
Get:37 http://archive.ubuntu.com/ubuntu focal-security/main Translation-en [326 kB]
Get:38 http://archive.ubuntu.com/ubuntu focal-security/main amd64 DEP-11 Metadata [59.9 kB]
Get:39 http://archive.ubuntu.com/ubuntu focal-security/main amd64 c-n-f Metadata [12.2 kB]
Get:40 http://archive.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [1,495 kB]
Get:41 http://archive.ubuntu.com/ubuntu focal-security/restricted Translation-en [211 kB]
Get:42 http://archive.ubuntu.com/ubuntu focal-security/universe amd64 Packages [794 kB]
Get:43 http://archive.ubuntu.com/ubuntu focal-security/universe i386 Packages [579 kB]
Get:44 http://archive.ubuntu.com/ubuntu focal-security/universe Translation-en [154 kB]
Get:45 http://archive.ubuntu.com/ubuntu focal-security/universe amd64 DEP-11 Metadata [94.0 kB]
Get:46 http://archive.ubuntu.com/ubuntu focal-security/multiverse amd64 DEP-11 Metadata [940 B]
Fetched 15.6 MB in 11s (1,429 kB/s)                                            
Reading package lists... Done
Building dependency tree       
Reading state information... Done
15 packages can be upgraded. Run 'apt list --upgradable' to see them.
ben@ben-meshify:~$ sudo apt-key adv --fetch-keys \
>     https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/3bf863cc.pub
Executing: /tmp/apt-key-gpghome.o83BhFCYM1/gpg.1.sh --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/3bf863cc.pub
gpg: requesting key from 'https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/3bf863cc.pub'
gpg: key A4B469963BF863CC: "cudatools <cudatools@nvidia.com>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
```

Best option might be to try and bust those cached layers. This should word:
```
docker builder prune
```

Next time you run you shouldn't see `CACHED` on any of the layers. If you do, another option (but will delete _all_ your layers) is:

```
docker system prune -a
```

ThePassedWind, 2023-02-10:
Thank you very much!!!
I tried your method and it worked!
Just input the command 'docker builder prune'.
Installing again and finally solving my problem!

closed 2023-02-10 by btalb