qcr/benchbot #90: Get started 1st step

colin-de, 2023-03-22:
hi, after the installation, I was stuck at the first step of Getting started

_Run a simulator with the BenchBot software stack by selecting an available robot, environment, and task definition:
`_benchbot_run --robot carter_omni --env miniroom:1 --task semantic_slam:active:ground_truth`

And I got:
```Supervisor is now available @ 'http://0.0.0.0:10000' ...

Waiting until a robot controller is found @ 'http://benchbot_robot:10000' ... 
	Found
Sending environment data & robot config to controller ... 
	Ready
Starting the robot controller ... 
	Ready
```
and no simulation environment afterwards. Could you please help me with that?

david2611, 2023-03-22:
This can happen sometimes on your first running of benchbot. Cancel your previous command then run a clean up with `benchbot_run -k` and then try running again.

david2611, 2023-03-27:
Pinging this issue again. If the problem has been solved let me know, if not this issue will be closed in 3 days

david2611, 2023-04-05:
Closed due to inactivity.

closed 2023-04-05 by david2611