qcr/benchbot #91: Error in BUILDING BENCHBOT CORE DOCKER IMAGE:

colin-de, 2023-03-23:
```
[+] Building 56.1s (8/12)                                                       
 => [internal] load build definition from Dockerfile                       0.0s
 => => transferring dockerfile: 2.90kB                                     0.0s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 52B                                           0.0s
 => [internal] load metadata for docker.io/benchbot/core:base              0.0s
 => [1/9] FROM docker.io/benchbot/core:base                                0.0s
 => CACHED [2/9] WORKDIR /benchbot                                         0.0s
 => CACHED [3/9] RUN apt update && apt install -y python3 python3-pip &&   0.0s
 => CACHED [4/9] RUN mkdir -p /benchbot                                    0.0s
 => ERROR [5/9] RUN apt update && apt install -y curl &&     echo "deb h  56.1s
------                                                                          
 > [5/9] RUN apt update && apt install -y curl &&     echo "deb http://packages.ros.org/ros/ubuntu bionic main" >     /etc/apt/sources.list.d/ros-latest.list &&     curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc |     apt-key add - && apt update &&     apt install -y ros-melodic-ros-base python-rosdep     python-rosinstall python-rosinstall-generator python-wstool     python-catkin-tools python-pip build-essential     ros-melodic-tf2-ros ros-melodic-tf:

#0 56.03 Fetched 130 MB in 46s (2806 kB/s)
#0 56.03 E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/b/boost-defaults/libboost-log-dev_1.65.1.0ubuntu1_amd64.deb  Connection timed out [IP: 185.125.190.39 80]
#0 56.03 E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
------
Dockerfile:26
--------------------
  25 |     # Install ROS Melodic
  26 | >>> RUN apt update && apt install -y curl && \
  27 | >>>     echo "deb http://packages.ros.org/ros/ubuntu bionic main" > \
  28 | >>>     /etc/apt/sources.list.d/ros-latest.list && \
  29 | >>>     curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | \
  30 | >>>     apt-key add - && apt update && \
  31 | >>>     apt install -y ros-melodic-ros-base python-rosdep \
  32 | >>>     python-rosinstall python-rosinstall-generator python-wstool \
  33 | >>>     python-catkin-tools python-pip build-essential \
  34 | >>>     ros-melodic-tf2-ros ros-melodic-tf
  35 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c apt update && apt install -y curl &&     echo \"deb http://packages.ros.org/ros/ubuntu bionic main\" >     /etc/apt/sources.list.d/ros-latest.list &&     curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc |     apt-key add - && apt update &&     apt install -y ros-melodic-ros-base python-rosdep     python-rosinstall python-rosinstall-generator python-wstool     python-catkin-tools python-pip build-essential     ros-melodic-tf2-ros ros-melodic-tf" did not complete successfully: exit code: 100

ERROR: Building BenchBot backend returned a non-zero error code: 1

Ensure that Docker has been installed correctly AND that you can run Docker 
WITHOUT root access (there is no need to ever run Docker with root). See 
https://docs.docker.com/install/linux/linux-postinstall/ for details on how to 
fix this. 

If the error is more generic, please contact us so that we can update our
pre-install host system checks.
```
I appreciate any help from this error:)

david2611, 2023-03-27:
It looks to me like a connection issue. Are you experiencing any internet issues or are operating from a country with restricted internet access?

david2611, 2023-04-05:
Pinging this issue again. If no further information provided will assume issue resolved and close the issue in 3 working days

david2611, 2023-04-10:
Closed due to inactivity

closed 2023-04-10 by david2611