qcr/benchbot #92: Benchbot Robot Controller Crash

tan007, 2023-04-04:
BenchBot start OK until ...
Supervisor is now available @ 'http://0.0.0.0:10000' ...

Waiting until a robot controller is found @ 'http://benchbot_robot:10000' ... 
	Found
Sending environment data & robot config to controller ... 
	Ready
Starting the robot controller ... 
=======================================

Then, the Python Issac Sim Console gives out this error message

2023-04-04 14:45:57  [Warning] [omni.kit.widget.live.cache_state_menu] Unable to detect Omniverse Cache Server. Consider installing it for better IO performance.

and crashes. 
Any idea to fix this..

david2611, 2023-04-05:
Hi @tan007. Thanks for lodging the issue. 
First off, are you using the most recent version of BenchBot and Omniverse? Can you tell me the version number of Omniverse when it appears in the pop up window?

What is it that is crashing? Just Omniverse or the benchbot_run command and the benchbot stack in general?

To try and get a better idea of what's going on, could you uncomment [this line](https://github.com/qcr/benchbot/blob/master/bin/benchbot_run#L398) in your benchbot_run script and then after the crash has occurred run `docker logs benchbot_robot` and share the output of that here please? 

Note that you will want to recomment the previous line when we are done debugging and will need to run `benchbot_run -k` to clean up the docker containers otherwise.

tan007, 2023-04-05:
Thanks for your reply. I am using the latest version of Benchbot( git default branch). The pop-up window (Issac sim python 2022.2.1), crashes and totally disappears. There is the yellow warning message " Unable to detect Omniverse Cache Server. Consider installing it for better IO performance. " at the bottom of the window. 

The log file output is here , after uncommenting the "sleep infinity" 
==================================================
Robot controller is now available @ 'http://0.0.0.0:10000' ...
172.20.0.102 - - [2023-04-04 14:45:43] "GET // HTTP/1.1" 200 152 0.000419
172.20.0.102 - - [2023-04-04 14:45:44] "POST //configure HTTP/1.1" 200 137 0.026483
Preparing the requested controller ... Done
172.20.0.102 - - [2023-04-04 14:46:00] "GET //prepare HTTP/1.1" 200 133 15.813365
Starting the requested run in running controller ... [INFO] [1680619580.108798]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw
publishing and latching message for 3.0 seconds

COMMAND TO START INSTANCE HAD NON-ZERO RETURN CODE (56)
Done
172.20.0.102 - - [2023-04-04 14:46:25] "GET //start HTTP/1.1" 200 132 25.774749
172.20.0.254 - - [2023-04-04 14:52:37] "GET /stop HTTP/1.1" 200 130 0.000308
172.20.0.254 - - [2023-04-04 15:13:54] "GET /stop HTTP/1.1" 200 130 0.000265
172.20.0.102 - - [2023-04-04 15:13:57] "GET // HTTP/1.1" 200 152 0.000250
[INFO] [1680621258.986191]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw

Controller isn't running, so nothing to stop.
172.20.0.102 - - [2023-04-04 15:14:18] "POST //configure HTTP/1.1" 200 137 20.009326
Preparing the requested controller ... Done
172.20.0.102 - - [2023-04-04 15:14:34] "GET //prepare HTTP/1.1" 200 133 15.356052
Starting the requested run in running controller ... [INFO] [1680621294.359775]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw
publishing and latching message for 3.0 seconds

COMMAND TO START INSTANCE HAD NON-ZERO RETURN CODE (56)
Done
172.20.0.102 - - [2023-04-04 15:14:59] "GET //start HTTP/1.1" 200 132 25.524091
172.20.0.254 - - [2023-04-05 01:34:15] "GET /stop HTTP/1.1" 200 130 0.000291
172.20.0.254 - - [2023-04-05 01:34:25] "GET /stop HTTP/1.1" 200 130 0.000263
172.20.0.102 - - [2023-04-05 01:34:28] "GET // HTTP/1.1" 200 152 0.000249
[INFO] [1680658489.851839]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw

Controller isn't running, so nothing to stop.
172.20.0.102 - - [2023-04-05 01:34:49] "POST //configure HTTP/1.1" 200 137 20.010748
Preparing the requested controller ... Done
172.20.0.102 - - [2023-04-05 01:35:05] "GET //prepare HTTP/1.1" 200 133 15.491744
Starting the requested run in running controller ... [INFO] [1680658525.361782]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw
publishing and latching message for 3.0 seconds

COMMAND TO START INSTANCE HAD NON-ZERO RETURN CODE (56)
Done
172.20.0.102 - - [2023-04-05 01:35:30] "GET //start HTTP/1.1" 200 132 25.523971
172.20.0.254 - - [2023-04-05 01:57:37] "GET /stop HTTP/1.1" 200 130 0.000438
172.20.0.254 - - [2023-04-05 01:57:48] "GET /stop HTTP/1.1" 200 130 0.000363
172.20.0.102 - - [2023-04-05 01:57:52] "GET // HTTP/1.1" 200 152 0.000361
[INFO] [1680659893.557523]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw

Controller isn't running, so nothing to stop.
172.20.0.102 - - [2023-04-05 01:58:13] "POST //configure HTTP/1.1" 200 137 20.008390
Preparing the requested controller ... Done
172.20.0.102 - - [2023-04-05 01:58:29] "GET //prepare HTTP/1.1" 200 133 15.475024
Starting the requested run in running controller ... [INFO] [1680659929.052392]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw
publishing and latching message for 3.0 seconds

COMMAND TO START INSTANCE HAD NON-ZERO RETURN CODE (52)
Done
172.20.0.102 - - [2023-04-05 01:59:10] "GET //start HTTP/1.1" 200 132 41.816659
172.20.0.254 - - [2023-04-05 02:00:25] "GET /stop HTTP/1.1" 200 130 0.000310
172.20.0.254 - - [2023-04-05 02:00:33] "GET /stop HTTP/1.1" 200 130 0.000255
172.20.0.102 - - [2023-04-05 02:00:37] "GET // HTTP/1.1" 200 152 0.000717
[INFO] [1680660058.302753]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw

Controller isn't running, so nothing to stop.
172.20.0.102 - - [2023-04-05 02:00:58] "POST //configure HTTP/1.1" 200 137 20.014455
Preparing the requested controller ... Done
172.20.0.102 - - [2023-04-05 02:01:13] "GET //prepare HTTP/1.1" 200 133 15.453577
Starting the requested run in running controller ... [INFO] [1680660093.774128]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw
publishing and latching message for 3.0 seconds

COMMAND TO START INSTANCE HAD NON-ZERO RETURN CODE (56)
Done
172.20.0.102 - - [2023-04-05 02:01:39] "GET //start HTTP/1.1" 200 132 25.525646
172.20.0.254 - - [2023-04-05 02:05:53] "GET /stop HTTP/1.1" 200 130 0.000313
172.20.0.254 - - [2023-04-05 02:06:59] "GET /stop HTTP/1.1" 200 130 0.000263
172.20.0.102 - - [2023-04-05 02:07:02] "GET // HTTP/1.1" 200 152 0.000253
[INFO] [1680660444.006010]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw

Controller isn't running, so nothing to stop.
172.20.0.102 - - [2023-04-05 02:07:24] "POST //configure HTTP/1.1" 200 137 20.008280
Preparing the requested controller ... Done
[... cut ...] :07:39] "GET //prepare HTTP/1.1" 200 133 15.393178
Starting the requested run in running controller ... [INFO] [1680660479.419428]: FAILURE: timeout exceeded while waiting for message on topic /camera/depth/image_raw
publishing and latching message for 3.0 seconds

COMMAND TO START INSTANCE HAD NON-ZERO RETURN CODE (56)
Done
172.20.0.102 - - [2023-04-05 02:08:04] "GET //start HTTP/1.1" 200 132 25.527548
=============================================================
Any idea to fix this ... Thanks

david2611, 2023-04-05:
Hmmm ok checking another couple of things.Is this your first time installing BenchBot or has it worked for you previously?

Am I correct in assuming after everything breaks and before you start a fresh attempt you have been running a `benchbot_run -k` command? 

Using the same setup as previously, instead of the docker logs command can I get you to print the output of the following?
`docker exec benchbot_robot /bin/bash -c "cat /tmp/benchbot_logs/0"`

That should give us the full terminal output from Isaac Sim. It is unusual that Isaac Sim itself is breaking.

tan007, 2023-04-05:
Yes. This is my first time installing BenchBot. 
The log printout is the following:
No python_sh_path provided. I SUPPOSE I SHALL BE Guessing ... 
Guessed: ./python.sh
Found existing installation: click 7.1.2
Uninstalling click-7.1.2:
  Successfully uninstalled click-7.1.2
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Requirement already satisfied: click in ./kit/python/lib/python3.7/site-packages (8.1.3)
Requirement already satisfied: importlib-metadata in ./kit/extscore/omni.kit.pip_archive/pip_prebundle (from click) (5.0.0)
Requirement already satisfied: zipp>=0.5 in ./kit/extscore/omni.kit.pip_archive/pip_prebundle (from importlib-metadata->click) (3.10.0)
Requirement already satisfied: typing-extensions>=3.6.4 in ./kit/extscore/omni.kit.pip_archive/pip_prebundle (from importlib-metadata->click) (3.7.4.3)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: You are using pip version 21.2.1+nv1; however, version 23.0.1 is available.
You should consider upgrading via the '/isaac-sim/kit/python/bin/python3 -m pip install --upgrade pip' command.
Found existing installation: typing-extensions 3.7.4.3
Uninstalling typing-extensions-3.7.4.3:
  Successfully uninstalled typing-extensions-3.7.4.3
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Requirement already satisfied: typing-extensions in ./extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1 (4.4.0)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: You are using pip version 21.2.1+nv1; however, version 23.0.1 is available.
You should consider upgrading via the '/isaac-sim/kit/python/bin/python3 -m pip install --upgrade pip' command.
HACK FIX FOR BROKEN PACKAGES
STARTING RUN.PY IN BENCHBOT_SIM_OMNI
inside run.py __main__
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0STARTING INSTANCE!!
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.isaac.kit.app_framework'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230405_180052.log
2023-04-05 18:00:52 [15ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.sensors.nv.lidar' or '/isaac-sim/exts/omni.sensors.nv.lidar/config'
2023-04-05 18:00:52 [15ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.sensors.nv.radar' or '/isaac-sim/exts/omni.sensors.nv.radar/config'
[0.167s] [ext: omni.stats-0.0.0] startup
[0.176s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.180s] [ext: omni.assets.plugins-0.0.0] startup
[0.181s] [ext: omni.gpu_foundation-0.0.0] startup
[0.184s] [ext: carb.windowing.plugins-1.0.0] startup
[0.197s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 530.30.02     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24564   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 6daf179d.. |
|=============================================================================================|
| OS: Linux benchbot_robot, Version: 5.15.0-69-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: 13th Gen Intel(R) Core(TM) i9-13900 | Cores: Unknown | Logical: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 128561 | Free Memory: 118100
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[0.973s] [ext: omni.kit.pipapi-0.0.0] startup
[0.978s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.979s] [ext: omni.kit.loop-isaac-1.0.0] startup
[... cut ...]
  File "/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 424, in update
    self._app.update()
RuntimeError: Operation not permitted
2023-04-05 18:01:23 [31,025ms] [Warning] [carb.audio.context] 1 contexts were leaked
./python.sh: line 41:   211 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python

======================================
Do I need to install any  pre-requisite package before installing BenchBot ? 

Thanks again for your help...

david2611, 2023-04-10:
Sorry for the delay in response.

The `benchbot_install` script should handle any packages that are required prior to installation. The Isaac Sim python environment is also separate to whatever you have on your own system. 

This might be a hardware or driver issue. Can you tell me what hardware (GPU in particular) you are using and give me the output of `nvidia-smi`?

When you installed benchbot initially did you pass all of the driver checks or did you need to install drivers through benchbot?

btalb, 2023-04-10:
Thanks for raising this @tan007 .

Can you get us a couple of things so we can dig a bit further please:
- BenchBot install logs as @david2611 mentioned:
    ```
    benchot_install  2>&1 | tee benchbot.log
    ```
- further details on the Vulkan crash (the `2023-04-05 18:01:22 [29,776ms] [Error] [carb.graphics-vulkan.plugin] GPU crash is detected. Crash dump is written into: /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230405_180052.nv-gpudmp` feels like the suspicious part of this all to me). With the `sleep infinity` line uncommented as @david2611  mentioned before, can you post the result of this after the crash:
    ```
    docker exec benchbot_robot /bin/bash -c "tail -n +1 /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.2/*.nv-gpudmp" 2>&1 | tee vulkan.log
    ```
    
When done, please post the `benchbot.log` & `vulkan.log` files back here for us.

tan007, 2023-04-12:
Thanks for your replay Ben,
The benchbot.log is the following.
=================================================================
################################################################################
###################### CHECKING BENCHBOT SCRIPTS VERSION #######################
################################################################################

Fetching latest hash for Benchbot scripts ... 
		f7573674da2905b12d62345721595de7f26a777e.

BenchBot scripts are up-to-date.

################################################################################
######################## PART 1: EXAMINING SYSTEM STATE ########################
################################################################################

Core host system checks:
	Ubuntu version >= 20.04:                                  Passed (20.04)

Running Nvidia related system checks:
	NVIDIA GPU available:                     Found card of type '10de:2684'
	NVIDIA driver is running:                                          Found
	NVIDIA driver version valid:                           Valid (530.30.02)
	NVIDIA driver from a standard PPA:                          PPA is valid
	CUDA drivers installed:                                    Drivers found
	CUDA drivers version valid:                          Valid (530.30.02-1)
	CUDA drivers from the NVIDIA PPA:                           PPA is valid
	CUDA is installed:                                            CUDA found
	CUDA version valid:                                         Valid (12.0)
	CUDA is from the NVIDIA PPA:                                PPA is valid

Running Docker related system checks:
	Docker is available:                                               Found
	Docker version valid:                                     Valid (23.0.3)
	NVIDIA Container Toolkit installed:                       Found (1.13.0)
	Docker runs without root:                                         Passed

Running checks of filesystem used for Docker:
	/var/lib/docker on ext4 filesystem:                 Yes (/dev/nvme0n1p2)
	/var/lib/docker supports suid:                                   Enabled
	/var/lib/docker drive space check:              Sufficient space (1658G)

Miscellaneous requirements:
	Pip python package manager available:                     Found (23.0.1)
	Tkinter for Python installed:                                      Found
	PIL (with ImageTk) for Python install                              Found

Manual installation steps for Omniverse-powered Isaac Sim:
	License accepted for Omniverse:                                      Yes
	Access to nvcr.io Docker registry:                                   Yes

All requirements & dependencies fulfilled. Docker containers for the BenchBot
software stack will now be built (which may take anywhere from a few seconds
to many hours). Would you like to proceed (y/N)? y

Proceeding with install ... 

################################################################################
################ PART 2: FETCHING LATEST BENCHBOT VERSION INFO #################
################################################################################

Fetching latest hash for BenchBot Simulator 'sim_omni' ... 
		c979337d0e0f2d0ff7113d79d126ca62d79dcc5a.
Fetching latest hash for BenchBot Robot Controller ... 
		142365e774cb9bdbcf6681ef4147f93154a4a7ba.
Fetching latest hash for BenchBot Supervisor ... 
		e5e50c46373d24176eb6e99dcc93661086571a68.
Fetching latest hash for BenchBot API ... 
		b8422ccb152332125b4ecb9ed253338fb495dd2d.
Fetching latest hash for BenchBot ROS Messages ... 
[... 801 lines cut ...]
| NVIDIA-SMI 530.30.02              Driver Version: 530.30.02    CUDA Version: 12.1     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                  Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf            Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4090         On | 00000000:01:00.0 Off |                  Off |
| 30%   34C    P8               38W / 450W|     12MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1202      G   /usr/lib/xorg/Xorg                            4MiB |
|    0   N/A  N/A      2063      G   /usr/lib/xorg/Xorg                            4MiB |
+---------------------------------------------------------------------------------------+

I cannot find the ppa link to install cuda-12.0 and nvidia-driver-525
The current ppa link installs cuda-12.1 and nvidia-driver-530.  
Could the latest cuda-driver and nvidia-driver cause the probem ?

Looking forward to receiving your reply.. Thx.

david2611, 2023-04-12:
We have been advised that the 530 drivers might be causing issues within Omniverse. Were your 530 drivers installed through the BenchBot install process or did you have those installed prior? The latest stable 525 drivers are what is recommended.

A dirty fix may be to uninstall your 530 drivers then run the benchbot_install script again.

tan007, 2023-04-12:
Thanks for your replay David,
I installed 525 Nividia-drivers at the beginning, but the Cuda-driver installation process re-installed the Nvidia-driver to 530.
I found that the installation script actually installs the latest 12.1 Cuda-driver and it also installs Nvidia-driver 530. 
The previous 12.0.1 or 12.0.0 Cuda-driver  works with  Nvidia-driver 525. 
Any quick fix to make the installation script of Benchbot to install the Cuda-12.0 or 12.01 driver, not the latest Cuda-12.1 driver?

Looking forward to your suggestion....Thanks

david2611, 2023-04-13:
The driver may not be the issue here. 
I have just re-installed benchbot on a test machine and have updated CUDA (which seems to update NVIDIA drivers even when you set it to the old version) and once installation completed, BenchBot operated as normal.

System state output from test machine
```
################################################################################
######################## PART 1: EXAMINING SYSTEM STATE ########################
################################################################################

Core host system checks:
	Ubuntu version >= 20.04:                                  Passed (20.04)

Running Nvidia related system checks:
	NVIDIA GPU available:                     Found card of type '10de:2230'
	NVIDIA driver is running:                                          Found
	NVIDIA driver version valid:                           Valid (530.30.02)
	NVIDIA driver from a standard PPA:                          PPA is valid
	CUDA drivers installed:                                    Drivers found
	CUDA drivers version valid:                          Valid (530.30.02-1)
	CUDA drivers from the NVIDIA PPA:                           PPA is valid
	CUDA is installed:                                            CUDA found
	CUDA version valid:                                         Valid (12.0)
	CUDA is from the NVIDIA PPA:                                PPA is valid

Running Docker related system checks:
	Docker is available:                                               Found
	Docker version valid:                                     Valid (23.0.3)
	NVIDIA Container Toolkit installed:                       Found (1.13.0)
	Docker runs without root:                                         Passed

Running checks of filesystem used for Docker:
	/var/lib/docker on ext4 filesystem:                 Yes (/dev/nvme0n1p2)
	/var/lib/docker supports suid:                                   Enabled
	/var/lib/docker drive space check:               Sufficient space (402G)

Miscellaneous requirements:
	Pip python package manager available:                     Found (20.0.2)
	Tkinter for Python installed:                                      Found
	PIL (with ImageTk) for Python install                              Found

Manual installation steps for Omniverse-powered Isaac Sim:
	License accepted for Omniverse:                                      Yes
	Access to nvcr.io Docker registry:                                   Yes
```

Does this match what you currently get when you run `benchbot_install`

tan007, 2023-04-13:
Thanks for your reply. David. However, I am still stuck in the installation process, even though I have the same 
system state as yours. 

################################################################################
######################## PART 1: EXAMINING SYSTEM STATE ########################
################################################################################

Core host system checks:
	Ubuntu version >= 20.04:                                  Passed (20.04)

Running Nvidia related system checks:
	NVIDIA GPU available:                     Found card of type '10de:2684'
	NVIDIA driver is running:                                          Found
	NVIDIA driver version valid:                           Valid (530.30.02)
	NVIDIA driver from a standard PPA:                          PPA is valid
	CUDA drivers installed:                                    Drivers found
	CUDA drivers version valid:                          Valid (530.30.02-1)
	CUDA drivers from the NVIDIA PPA:                           PPA is valid
	CUDA is installed:                                            CUDA found
	CUDA version valid:                                         Valid (12.0)
	CUDA is from the NVIDIA PPA:                                PPA is valid

Running Docker related system checks:
	Docker is available:                                               Found
	Docker version valid:                                     Valid (23.0.3)
	NVIDIA Container Toolkit installed:                       Found (1.13.0)
	Docker runs without root:                                         Passed

Running checks of filesystem used for Docker:
	/var/lib/docker on ext4 filesystem:                 Yes (/dev/nvme0n1p2)
	/var/lib/docker supports suid:                                   Enabled
	/var/lib/docker drive space check:              Sufficient space (1721G)

Miscellaneous requirements:
	Pip python package manager available:                     Found (20.0.2)
	Tkinter for Python installed:                                      Found
	PIL (with ImageTk) for Python install                              Found

Manual installation steps for Omniverse-powered Isaac Sim:
	License accepted for Omniverse:                                      Yes
	Access to nvcr.io Docker registry:                                   Yes

I still got the following error messages from the output  
========================================================
[2.781s] [ext: omni.kit.widget.spinner-1.0.4] startup
[2.782s] [ext: omni.kit.browser.sample-1.2.5] startup
**2023-04-13 08:48:37 [2,778ms] [Error] [carb.scripting-python.plugin] FileNotFoundError: [Errno 2] No such file or directory: '/root/.nvidia-omniverse/config/omniverse.toml'

At:
  /isaac-sim/kit/extscore/omni.kit.pip_archive/pip_prebundle/toml/decoder.py(133): load
  /isaac-sim/extscache/omni.kit.browser.folder.core-1.7.3/omni/kit/browser/folder/core/models/folder_browser_model.py(108): __init__
  /isaac-sim/extscache/omni.kit.browser.folder.core-1.7.3/omni/kit/browser/folder/core/models/tree_folder_browser_model.py(16): __init__
  /isaac-sim/extscache/omni.kit.browser.sample-1.2.5/omni/kit/browser/sample/model.py(20): __init__
  /isaac-sim/extscache/omni.kit.browser.sample-1.2.5/omni/kit/browser/sample/extension.py(31): on_startup
  /isaac-sim/kit/kernel/py/omni/ext/_impl/_internal.py(147): _startup_ext
  /isaac-sim/kit/kernel/py/omni/ext/_impl/_internal.py(198): startup
  /isaac-sim/kit/kernel/py/omni/ext/_impl/_internal.py(285): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  /isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py(301): _start_app
  /isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py(186): __init__
  <string>(1): <module>

2023-04-13 08:48:37 [2,778ms] [Error] [omni.ext.plugin] [ext: omni.kit.browser.sample-1.2.5] Failed to startup python extension.**
[2.855s] [ext: omni.graph.nodes-1.48.3] startup
[2.869s] [ext: omni.kit.viewport.menubar.core-104.4.3] startup
[2.902s] [ext: omni.kit.numpy.common-0.1.0] startup
=======================================================
and  warning messages
2023-04-13 08:48:38 [3,142ms] [Warning] [omni.kit.viewport.menubar.core.menu_item.viewport_menu_item] ViewportMenuItem order argument is deprecated, use order_setting_path

2023-04-13 08:48:38 [3,384ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template

2023-04-13 08:48:39 [4,707ms] [Warning] [omni.ext.plugin] [ext: omni.isaac.sensor-5.6.3] Native plugin search dir contains several files: /isaac-sim/exts/omni.isaac.sensor/bin . It makes dir walk slow, replace with the exact path to the plugin instead of a wildcard.

2023-04-13 08:48:40 [5,362ms] [Warning] [omni.isaac.isaac_sensor] omni.isaac.isaac_sensor is deprecated and was renamed; use omni.isaac.sensor

2023-04-13 08:48:41 [6,148ms] [Warning] [omni.kit.widget.live.cache_state_menu] Unable to detect Omniverse Cache Server. Consider installing it for better IO performance.

The pop up window also crashes after showing the warning message "Unable to detect Omniverse Cache Server. Consider installing it for better IO performance."

By the way, I am using RTX 4090. Could hardware be the problem? 

Looking forward to receiving your reply ....

david2611, 2023-04-14:
OK so just confirming I understand rightly, the system was able to install previously but is now giving you errors when you try to reinstall?

Has anything about your system changed between the first and second installs of BenchBot? The reason I ask is that the install process can only complete when the pop-up window appears and does not crash immediately (caching a bunch of things to speed up subsequent times using Isaac Sim).

Can I confirm that in your benchbot folder you have a .cache folder that contains an isaac-sim folder?

tan007, 2023-04-14:
I was able to modify the benchbot_install script to use the the nvidia-driver-525  and cuda-12.0
################################################################################
######################## PART 1: EXAMINING SYSTEM STATE ########################
################################################################################

Core host system checks:
	Ubuntu version >= 20.04:                                  Passed (20.04)

Running Nvidia related system checks:
	NVIDIA GPU available:                     Found card of type '10de:2684'
	NVIDIA driver is running:                                          Found
	NVIDIA driver version valid:                          Valid (525.105.17)
	NVIDIA driver from a standard PPA:                          PPA is valid
	CUDA drivers installed:                                    Drivers found
	CUDA drivers version valid:                         Valid (525.105.17-1)
	CUDA drivers from the NVIDIA PPA:                           PPA is valid
	CUDA is installed:                                            CUDA found
	CUDA version valid:                                         Valid (12.0)
	CUDA is from the NVIDIA PPA:                                PPA is valid

Running Docker related system checks:
	Docker is available:                                               Found
	Docker version valid:                                     Valid (23.0.3)
	NVIDIA Container Toolkit installed:                       Found (1.13.0)
	Docker runs without root:                                         Passed

Running checks of filesystem used for Docker:
	/var/lib/docker on ext4 filesystem:                 Yes (/dev/nvme0n1p2)
	/var/lib/docker supports suid:                                   Enabled
	/var/lib/docker drive space check:              Sufficient space (1667G)

Miscellaneous requirements:
	Pip python package manager available:                     Found (20.0.2)
	Tkinter for Python installed:                                      Found
	PIL (with ImageTk) for Python install                              Found

Manual installation steps for Omniverse-powered Isaac Sim:
	License accepted for Omniverse:                                      Yes
	Access to nvcr.io Docker registry:                                   Yes

I do have the    .cache folder that contains an Isaac-sim folder in the benchbot folder. 

I suspect that the docker file does not install the Omniverse properly. I check that my PC configuration can run nvidia-onmiverse and Issac-smi by installing the Nvidia omniverse-launcher-linux.App.image . 

By the way, is it possible to launch all the benchbot applications without using docker to run Issac-sim and Omniverse? 
I can easily run them from the Nvidia omniverse-launcher-Linux.App.image. 

Thanks again for your help...

tan007, 2023-04-15:
I guess I also need to define somewhere in the installation script the path of  .nvidia-omniverse folder. It is in my home directory. 
Is that right?

david2611, 2023-04-16:
Sorry for the delay.

If you are able to install and run NVIDIA Omniverse outside of BenchBot that is a good sign that things should work in BenchBot.

Unfortunately, BenchBot uses docker for a lot of its moving parts (from earlier design decisions to get it to work in both simulations and on physical hardware). @btalb may know of a way of using the external app in some way shape or form but you would have to do a lot of adaptation to base code that would be more hassle than it is work. If you are able to run Omniverse on your machine you should be able to run it in the docker.

Did you reinstall BenchBot after you changed your drivers? Did it go through the entire install procedure.

It might be a good idea to do a `benchbot_install -u` to uninstall benchbot before going through the install script again (install script might be using cached images if not). Because you have passed all the install checks with the output you have shown above, BenchBot shouldn't touch your drivers at all.

Nuclear option for a "clean install" would be a `docker system prune -a` command but that will get rid of ALL docker images.

tan007, 2023-04-17:
Thanks for your reply. David
I was able to start Benchbot properly after creating a static link of .nvidia-omniverse from my home directory to the Benchbot folder. The Issac-Sim dashboard starts and renders graphics of a scene. I can launch the hello_benchbot example as well. 
However, it still complains about not being able to connect to the omniverse-cache server and the example crashes after I run it for a number of steps. 
[1] How should I start and set up Omniverse-launcher outside of Benchbot, so Benchbot can find the cache-server ?
[2] Should I also need to configure the docker to allow more memory for the Benchbot container ?

btalb, 2023-04-17:
That's great progress @tan007 .

RE the App.Image: that wasn't available when we initially created BenchBot, so it's all been designed around the Docker installation methods. That being said, if you can run it through the flatpack, than BenchBot should be able to work on your system.

The symbolic linking of `.nvidia-omniverse` into the BenchBot directory is not something we've needed in the past... I wonder if it's popped up in the upgrade to the latest Isaac Sim and we haven't detected it due to our existing installs having the directory from the past.

Omniverse Cache has never been used, and shouldn't be required to run BenchBot. That is for if we were hosting an instance / assets in the cloud and wanted to interactively work on a simulation across multiple machines. @david2611 can you confirm nothing has changed in this regard with the newest Isaac Sim?

You shouldn't need to make any configuration changes to the Docker or setup cache for BenchBot to run. Have you got any updated logs of what is happening now when it "crashes"?  Does the Isaac Sim window die? Is there anything in the logs @david2611 pointed you to above (e.g. `docker exec benchbot_robot /bin/bash -c "cat /tmp/benchbot_logs/0"`).

Thanks for persevering with this with us.

tan007, 2023-04-18:
Thanks for your reply. I attach the last few lines of the log file below. Apparently, there is a core dump error. Any idea how to fix that? Looking forward to receiving your reply. 
================================================================================
2023-04-18 06:30:32 [108,757ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /robot/left_wheel_link/cylinder is using obsolete 'customGeometry' attribute. To toggle custom geome
try for cylinders, use the physics settings option.
2023-04-18 06:30:32 [108,757ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /robot/right_wheel_link/cylinder is using obsolete 'customGeometry' attribute. To toggle custom geom
etry for cylinders, use the physics settings option.
2023-04-18 06:30:32 [108,757ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /robot/rear_wheel_link/cylinder is using obsolete 'customGeometry' attribute. To toggle custom geome
try for cylinders, use the physics settings option.
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/com_link)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Translation DOF is expected to be locked for D6 joint in articulation hierarchy, the resulting joint in articulation will hav
e all translation DOF locked.
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/left_wheel)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/right_wheel)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/rear_pivot)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/imu_joint)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/rear_pivot_link/rear_axle)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/left_wheel)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/right_wheel)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/rear_pivot)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/imu_joint)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/rear_pivot_link/rear_axle)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/right_wheel)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/rear_pivot)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/imu_joint)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/rear_pivot_link/rear_axle)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/rear_pivot)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/imu_joint)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/rear_pivot_link/rear_axle)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/imu_joint)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Per-joint collision setting is ignored for articulations. (/robot/chassis_link/imu_joint)
2023-04-18 06:30:33 [108,886ms] [Warning] [omni.physx.plugin] Translation DOF is expected to be lo./python.sh: line 41:   243 Illegal instruction     (core dumped) $python_exe "$@" $args
There was an error running python
=====================================================================

david2611, 2023-04-19:
Hi all, sorry for delayed replies.

So I will admit to being very confused on how you have installed BenchBot such that it works with your local install of Isaac Sim. You have done something in the install process itself to make it point to the local simulator itself? Without needing to use the IsaacSim docker container? You have installed everything using the AppImage omniverse-launcher?

So when you run `benchbot_run` it is running the simulation commands to the local version of IsaacSim? 

When you installed with the launcher you should have been able to install Omniverse Cache which would give you access to the localhost version of cache.

@btalb the .nvidia-omniverse folder is a by-product of the local install of Isaac Sim and is not included normally in the standard BenchBot install of Isaac Sim which is done entirely through the IsaacSim docker container.

If you are still having issue with the local Isaac Sim and wanted to try the docker container, one trick to check would be the following commands from the Isaac Sim Docker instructions.

```
xhost +
docker run --name isaac-sim --entrypoint bash -it --gpus all -e "ACCEPT_EULA=Y" --rm --network=host \
 -v ~/docker/isaac-sim/kit/cache/Kit:/isaac-sim/kit/cache/Kit:rw \
 -v ~/docker/isaac-sim/cache/ov:/root/.cache/ov:rw \
 -v ~/docker/isaac-sim/cache/pip:/root/.cache/pip:rw \
 -v ~/docker/isaac-sim/cache/glcache:/root/.cache/nvidia/GLCache:rw \
 -v ~/docker/isaac-sim/cache/computecache:/root/.nv/ComputeCache:rw \
 -v ~/docker/isaac-sim/logs:/root/.nvidia-omniverse/logs:rw \
 -v ~/docker/isaac-sim/data:/root/.local/share/ov/data:rw \
 -v ~/docker/isaac-sim/documents:/root/Documents:rw \
 nvcr.io/nvidia/isaac-sim:2022.2.1
```
Then within the interactive session
`./runapp.sh`
If that works for you fine, the docker image for Isaac Sim as used in BenchBot should also be working.

tan007, 2023-04-21:
I am running the containerized issac-sim , and kill all the Nvidia Omniverse launcher background jobs.
As part of the BenchBot containerized Issac-Sim installation process in my workstation (RTX 4090),  the  availability of .nvidia-omniverse folder is required.  So I create a static link to the same folder in my home directory and Benchbot can launch successfully.    Hope this helps to improve the Benchbot...
rp5@dt4090:~$ docker container ls
CONTAINER ID   IMAGE                         COMMAND                  CREATED          STATUS          PORTS                                                                                                                        NAMES
c7530aa2b7fb   benchbot/backend:base         "/bin/bash"              19 minutes ago   Up 19 minutes                                                                                                                                benchbot_debug
e39aff1e9e0e   benchbot/backend:base         "/bin/bash -c 'sourc…"   19 minutes ago   Up 19 minutes                                                                                                                                benchbot_supervisor
8762849e0896   benchbot/simulator:sim_omni   "/bin/bash -c 'sourc…"   19 minutes ago   Up 19 minutes   8011-8012/tcp, 8211/tcp, 8891/tcp, 8899/tcp, 47995-48012/tcp, 47995-48012/udp, 49000-49007/tcp, 49100/tcp, 49000-49007/udp   benchbot_robot
80935f09feed   benchbot/backend:base         "/bin/bash -c 'sourc…"   19 minutes ago   Up 19 minutes   

rp5@dt4090:~$ ps -u
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
rp5         2384  0.0  0.0 172652  6392 tty2     Ssl+ 09:21   0:00 /usr/lib/gdm3/gdm-x-session --run-script env GNOME_SHELL_SESSION_MODE=ubuntu /usr/bin/gnome-session --systemd --session=ubuntu
rp5         2435  0.0  0.0 199364 15164 tty2     Sl+  09:21   0:00 /usr/libexec/gnome-session-binary --systemd --systemd --session=ubuntu
rp5         3502  0.0  0.0  19644  5576 pts/0    Ss   09:21   0:00 bash
rp5         5305  0.0  0.0  18880  4244 pts/0    S+   09:25   0:00 bash /usr/local/bin/benchbot_run --robot carter_omni --env house:2 --task semantic_slam:passive:ground_truth
rp5         6947  0.0  0.0 1625148 24968 pts/0   Sl+  09:25   0:00 docker logs --follow benchbot_supervisor
rp5         9805  0.0  0.0  19644  5296 pts/1    Ss   09:27   0:00 bash
rp5       500940  0.0  0.0  19512  5036 pts/2    Ss   09:43   0:00 bash
rp5       595699  0.0  0.0  18784  3964 pts/1    S+   09:48   0:00 bash /usr/local/bin/benchbot_submit --example hello_passive
rp5       595786 77.9  0.1 3001096 154068 pts/1  Sl+  09:48   1:28 python3 .hello_passive/hello_passive

tan007, 2023-04-26:
I manage to fix the frequent crashing of Issac-sim on my workstation. After extensive troubleshooting, I find that the RAM clock frequency is set too high. After reducing the RAM clock frequency, Issac-sim runs well.

david2611, 2023-04-26:
Excellent work! Sorry for the limited communication on my end but happy to hear you reached a satisfactory conclusion. Are you satisfied to close the issue or is there something more that is still breaking in your use of BenchBot?

tan007, 2023-04-27:
Please close this ticket.

closed 2023-04-27 by david2611