qcr/benchbot #98: Executing benchbot_submit results in no scene change, even when it involves an SCD task.

yutojubako, 2023-04-27:
I executed `benchbot_run --robot carter_omni --env house:2:3 --task scd:active:ground_truth` and observed the following log.
```
benchbot_run --robot carter_omni --env house:2:3 --task scd:active:ground_truth

################################################################################
################# CHECKING FOR BENCHBOT SOFTWARE STACK UPDATES #################
################################################################################

Checking BenchBot version ...				Up-to-date.
Checking BenchBot API version ...			Up-to-date.
Checking BenchBot Eval version ...			Up-to-date.
Checking BenchBot Supervisor version ...		Up-to-date.
Checking installed BenchBot add-ons are up-to-date ...	Up-to-date.

BenchBot is up-to-date.

################################################################################
###################### CLEANING UP ALL BENCHBOT REMNANTS #######################
################################################################################

Sending stop request to running controller:
{"stop_success":true}

Deleted the following containers:
Total reclaimed space: 0B

Finished cleaning! (use 'benchbot_run -k' for a full clean)

################################################################################
##################### STARTING THE BENCHBOT SOFTWARE STACK #####################
################################################################################

Running the BenchBot system with the following settings:

    Selected task:        scd:active:ground_truth
    Task results format:  object_map_with_states
    Selected robot:       carter_omni
    Selected environment: house:2:3
    Scene/s:              house:2, starting @ pose [0.781, 0, 0, 0.625, 3.2, -0.5, 0.3]
                          (map_path = '.sim_data/house_2.usd')
                          house:3, starting @ pose [0.781, 0, 0, 0.625, -3.2, 0.5, 0.3]
                          (map_path = '.sim_data/house_3.usd')
    Simulator required:   Yes (sim_omni)

Creating shared network 'benchbot_network':

Starting persistent container for ROS core:
Skipping (already running)

Starting persistent container for BenchBot Robot Controller (sim_omni):
Skipping (already running)

Starting container for BenchBot Supervisor:
52c72ec1bb7548691594e17263b3a5fe0bd7afe5327a3d63eff7c00368215978

Starting container for BenchBot Debugging:
f9bd8c43fe35631638cb97188d17c6c29e07b878294c28ea611eb9c81a992143

################################################################################
################### BENCHBOT IS RUNNING (Ctrl^C to exit) ... ###################
################################################################################

Initialising supervisor...

Configuring the supervisor...
Starting a supervisor with the following configuration:

{'environments': [{'_file_path': '/benchbot/addons/benchbot_addons/benchbot-addons/envs_bear_develop_sim_omni/environments/house_2.yaml',
                   'description': 'Large house environment with object '
                                  'variations under day lighting\n',
[... 112 lines cut ...]
           json.dump(empty_results, f)

if __name__ == '__main__':
   # Start BenchBot, & initialise our agent
   bb = BenchBot()
   a = AiGymAgent()

   # Use our Agent to run through the first scene until completion
   observations, action_result = bb.reset()
   print("camera matrix: %s" % observations['image_depth_info'])

   # Ask BenchBot to move us to the next scene
   bb.next_scene()

   # Run through the second scene until completion
   observations, action_result = bb.reset()
   print("observations: %s" % observations['image_depth_info'])

   # When we are finished save our results
   # NOTE: you must save to the default RESULT_LOCATION as show below for your
   # code to work with the '-e' and '-r|--results-location' flags of the
   # 'benchbot_submit' script
   a.save_result(RESULT_LOCATION, bb.empty_results(), bb.results_functions())
```
Please contact me if you know anything about this.

david2611, 2023-05-03:
Sorry for the delay in this. Have just updated the benchbot robot controller to try and fix this problem. It has passed tests on my end. Can you just run a `benchbot_install` on your end and confirm this also fixes the problem for you?

david2611, 2023-05-09:
Pinging this issue again. Can I please receive confirmation that SCD is now working for you? I shall close this issue in 3 working days if I hear of no further issues on your end.

yutojubako, 2023-05-11:
Thank you for your reply. Sorry for the delay in replying.
By running `benchbot_install` again, we were able to get SCD to work properly in the environment of `apartment1:4` for now.
This problem has been successfully resolved, but we have not yet tried it in all other environments and will contact you again if we encounter this type of trouble.

closed 2023-05-11 by yutojubako