qcr/benchbot #101: Downloading the dataset

psykaunot, 2024-12-28:
Hello, I would like to download the dataset, but it looks like the following link does not work anymore:
https://open.qcr.ai/dataset/benchbot-bear-data/

HassanRbotics, 2025-01-04:
Inside the following links the google drive link:
https://github.com/benchbot-addons/envs_bear_develop_sim_omni/blob/master/.remote
https://github.com/benchbot-addons/envs_bear_challenge_sim_omni/blob/master/.remote

psykaunot, 2025-01-06:
Dear Mr Hassan, thank you for your help. I was able to access them and download them.

However, before closing the issue, I do have a small question. I am unable to open most of them. It works for rooms, but it doesn't work for apartments.

Since my workstation is powerful enough, I wanted to know if you had any idea what the issue could be. I downloaded the file several times.